webcamoid/webcamoid #185: Can't compile webcamoid

Willi-Smith, 2019-07-17:
After installation of all dependencies I'm unable to compile Webcamoid from the git on Debian Buster. I'm not familiar with the error so it can be a trivial mistake of mine. Thanks for help
[Second run compilation log](https://gist.github.com/Willi-Smith/b8b3a910b99f9ec1ab198c20d78e0fab)
```
/usr/bin/ld: /home/william/apps/webcamoid/StandAlone/../libAvKys/Lib/release/Qt5.11.3/gpp/x86_64/bin/libavkys.so: undefined reference to `double qbswap<double>(double)'
/usr/bin/ld: /home/william/apps/webcamoid/StandAlone/../libAvKys/Lib/release/Qt5.11.3/gpp/x86_64/bin/libavkys.so: undefined reference to `float qbswap<float>(float)'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:194: release/Qt5.11.3/gpp/x86_64/bin/webcamoid] Error 1
make[1]: Leaving directory '/home/william/apps/webcamoid/StandAlone'
make: *** [Makefile:89: sub-StandAlone-make_first-ordered] Error 2
```

hipersayanX, 2019-07-17:
Is a known bug, will be fixed soon.

hipersayanX, 2019-07-18:
This was fixed in the latest commit.

closed 2019-07-18 by hipersayanX