webcamoid/webcamoid #186: Virtual webcam

Joe23232, 2019-08-27:
Using Virtual webcam, am I able to play some video and services such as skype will be fed the video  instead?

Joe23232, 2019-08-28:
> Yes, you can play a video and the virtual camera will send the frames to Skype or any other program.

Hi, how do I get Opera (a browser) to use the virtual camera instead of using the built in?

[image]

I tried to change the setting but this is the only setting, any ideas?

Joe23232, 2019-08-31:
> I have updated the wiki
> <https://github.com/webcamoid/webcamoid/wiki/Virtual-camera-support>. The
> virtual camera works in GNU/Linux, Mac and Windows, when you create the
> virtual camera try restarting the target program (opera, skype or whatever).
>
> [image: Screenshot_20190828_111923]
> <https://user-images.githubusercontent.com/1249885/63868492-fedcd300-c98c-11e9-8c4b-38b8c2b091db.png>
> [image: Screenshot_20190828_112100]
> <https://user-images.githubusercontent.com/1249885/63868493-ff756980-c98c-11e9-82fc-93ce5952f66e.png>

On this PC, for some reason it just doesn't allow me to use any camera,
even the intergrated camera, something is a bit fucked I am not too sure
why nothing works :/

hipersayanX, 2019-08-31:
If the problem is with your system I can't give you any help with that :confused:

malikAr, 2019-09-02:
Hi,

I have compiled webcamoid code on mac successfully , but not able to find out how to use akvirtualcamera.plugin properly.

I have copied it /Library/CoreMediaIO/Plug-Ins/DAL

Now @hipersayanX  will you please help me to find out my camera plugin in skype or zoom.

Any help is thankful.

hipersayanX, 2019-09-02:
@malikAr don't install the plugin manually, just copying the file isn't enough, some other files must be created and some services must be activated. When testing Webcamoid you can pass the **--vcam** option with the path to the plugin to the CLI (you can add that option in Qt Creator), and Webcamoid will do everything for you. For example:

    webcamoid --vcam /path/to/AkVirtualCamera.plugin

closed 2019-10-01 by hipersayanX