webcamoid/webcamoid #188: Format file

IvanAP1791, 2019-09-10:
Goodnight.

I turn to you, in order to ask if the software can be configured in any way so that the images are automatically saved in * .jpg

Greetings.

hipersayanX, 2019-09-10:
No, but it can be done.

IvanAP1791, 2019-09-11:
and how could we do?

hipersayanX, 2019-09-11:
Writing the code to do so.

somewhereinusa, 2020-01-24:
Has this been done yet? Would also like to save to a default file of my choosing

yllekz, 2022-01-24:
Hello, this app should really be saving webcam captures as jpg and not png. Where is the option to change this?

hipersayanX, 2022-01-25:
@yllekz _Preferences>Image Capture>File Format_ in the daily build, 8.8 has not such option. I will keep PNG as default for quality and compatibility reasons.

hipersayanX, 2022-02-15:
Feature added in 9.0.0, closing.

closed 2022-02-15 by hipersayanX