webcamoid/webcamoid #190: Source config not persisting after system reboot in Pop_OS! (Ubuntu 19.04)

fredevery, 2019-09-17:
I am using webcamoid to dial in very specific brightness, contrast, color temp, exposure, etc. settings and so far I have come up with some amazing visuals with my Logitech C922. My only gripe is that I can't find a way to save/load my settings, so when I shutdown I seem to lose all my settings. Right now I'm taking a screenshot of my settings and dialing them back in whenever I need to, but this is tedious.

Is there a way to save/load source config settings?

My environment:
OS:
```
Distributor ID:	Ubuntu
Description:	Pop!_OS 19.04
Release:	19.04
Codename:	disco
```
Camera: Logitech C922

fredevery, 2019-09-17:
As an additional suggestion, it would be great if the configs were portable and could be used on other operating systems.

hipersayanX, 2019-09-17:
Webcamoid only save the format, the resolution and the frame rate for now. But sure, that feature can be added.

>As an additional suggestion, it would be great if the configs were portable and could be used on other operating systems.

That's kinda impossible, webcam attributes differ a lot from one system to another, from one camera to another.

fredevery, 2019-09-20:
@hipersayanX If you are willing to give some pointers of where to start I'd be happy to take a crack at it. Like if you can point to where you're reading the format, resolution, and frame rate and persisting them I can attempt to expand that to include all the configs. Thanks.

hipersayanX, 2019-09-21:
@fredevery Probably a good idea is to save that configs at [setImageControls](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/videocaptureelement.cpp#L252) and [setCameraControls](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/videocaptureelement.cpp#L276), and restore the configs at [setMedia](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/videocaptureelement.cpp#L461) and [reset](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/videocaptureelement.cpp#L578), all camera configs goes in the **VideoCapture** group, or you can create a new group for each camera if you consider it necessary (choose the option that looks more readable in the config file).
Here you can read the documentation for [QSettings](https://doc.qt.io/qt-5/qsettings.html).