webcamoid/webcamoid #191: Disable the Virtual Camera when program is closed

GlassGruber, 2019-09-21:
Hei there!  
First of all thanks for this, I'm using your program on windows 7 64bit and the Virtual Camera is working quite well! (though I had a few issues trying to get it working properly, restarting the target program for the camera was a good tip, also maybe starting the portable `.bat` file as admin helped but I'm not sure).

So just as title, is it possible to disable or make the camera not detectable when the program is closed? I guess it's rather a minor issue, but I noticed the camera is detected even when webcamoid is closed, showing the default banner.

Else is it possible to change the default webcamoid banner to a solid black color or custom image/source when the program is closed?

I'm using the 32bit portable version 8.6.1

Thank you for your time!

hipersayanX, 2019-09-23:
>is it possible to disable or make the camera not detectable when the program is closed?

Hi, disabling the camera is not possible, DirectShow filters are statically registered in the registry.

>Else is it possible to change the default webcamoid banner to a solid black color or custom image/source when the program is closed?

Yes, you can put a 640x480 and 32 or 24 bits bitmap image called **TestFrame.bmp** in _C:\Program Files\AkVirtualCamera.plugin\share_ and  _C:\Program Files (x86)\AkVirtualCamera.plugin\share_ (the folders where the filter is installed).

GlassGruber, 2019-09-23:
Nice thank you! 
As a small note, the _\\share\\_ folder is not present by default, you have to create the directory and place the created **TestFrame.bmp** file inside it.

closed 2019-10-01 by hipersayanX