webcamoid/webcamoid #192: Camera image cannot be displayed.

GoEZM, 2019-09-26:
In macOS 10.14.6, XCode 10.1, Qt 5.12.2.

I followed the instructions for "Build and Install on Mac", which is built and run in debug mode.
Webcamoid lists all camera devices, but does not display lens images.
When we select "Screen", we can get the picture normally.

What should I check to make the program action?
Thank you.

[image]

[image]

hipersayanX, 2019-09-26:
The most relevant message is this:

```
objc[4794]: Class FrameGrabber is implemented in both /Users/jack/Documents/webcamoid-8.6.1/libAvKys/Plugins/DesktopCapture/debug/Qt5.12.2/clangpp/x86_64/bin/submodules/DesktopCapture/libavfoundation.dylib (0x10a757048) and
                                                                                                           /Applications/webcamoid.app/Contents/Plugins/avkys/submodules/DesktopCapture/libavfoundation.dylib (0x10a818a48). One of the two will be used. Which one is undefined.
objc[4794]: Class DeviceObserver is implemented in both /Users/jack/Documents/webcamoid-8.6.1/libAvKys/Plugins/VideoCapture/debug/Qt5.12.2/clangpp/x86_64/bin/submodules/VideoCapture/libavfoundation.dylib (0x10a7c7750) and
                                                                                                           /Applications/webcamoid.app/Contents/Plugins/avkys/submodules/VideoCapture/libavfoundation.dylib (0x10a849d20). One of the two will be used. Which one is undefined.
```

Webcamoid is finding the plugins in both the install directory and the development folder, and it is conflicting because is trying to load the definition from both sides. Try adding theses options the command line:

```
--vcam /Users/jack/Documents/webcamoid/libAvKys/Plugins/VirtualCamera/src/cmio/VirtualCamera/AkVirtualCamera.plugin -q /Users/jack/Documents/webcamoid/libAvKys -r -p /Users/jack/Documents/webcamoid/libAvKys/Plugins
```

If that does not works try [resetting the configs](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh#mac). If that still does not works try moving _/Applications/webcamoid.app_ to another place and resetting the configs.

Also, compile from the git repository instead, I've recently fixed some bugs that are relevant to Mac there.

GoEZM, 2019-09-27:
Thank you for your help. I will try your suggestion.

I have downloaded the current version (8.7.0), but compiled it in macOS + Qt 5.12.2 and found an error:

src/devicecontrols.cpp:280:25: error: implicit instantiation of undefined template 'QVector<unsigned short>'
    for (auto &control: UvcControl::allSelectors(PROCESSING_UNIT)) {
                        ^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:2133:1: note: template is declared here
QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG(Q_DECLARE_SEQUENTIAL_CONTAINER_METATYPE_ITER)
^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:215:7: note: expanded from macro 'QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG'
    F(QVector) \
      ^
src/devicecontrols.cpp:304:25: error: implicit instantiation of undefined template 'QVector<unsigned short>'
    for (auto &control: UvcControl::allSelectors(CAMERA_TERMINAL)) {
                        ^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:2133:1: note: template is declared here
QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG(Q_DECLARE_SEQUENTIAL_CONTAINER_METATYPE_ITER)
^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:215:7: note: expanded from macro 'QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG'
    F(QVector) \
      ^
src/devicecontrols.cpp:326:25: error: implicit instantiation of undefined template 'QVector<unsigned short>'
    for (auto &control: UvcControl::allSelectors(PROCESSING_UNIT))
                        ^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:2133:1: note: template is declared here
QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG(Q_DECLARE_SEQUENTIAL_CONTAINER_METATYPE_ITER)
^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:215:7: note: expanded from macro 'QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG'
    F(QVector) \
      ^
src/devicecontrols.cpp:339:25: error: implicit instantiation of undefined template 'QVector<unsigned short>'
    for (auto &control: UvcControl::allSelectors(CAMERA_TERMINAL))
                        ^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:2133:1: note: template is declared here
QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG(Q_DECLARE_SEQUENTIAL_CONTAINER_METATYPE_ITER)
^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:215:7: note: expanded from macro 'QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG'
    F(QVector) \
      ^
src/devicecontrols.cpp:645:38: error: implicit instantiation of undefined template 'QVector<UvcControl>'
    static const QVector<UvcControl> controls {
                                     ^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:2133:1: note: template is declared here
QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG(Q_DECLARE_SEQUENTIAL_CONTAINER_METATYPE_ITER)
^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:215:7: note: expanded from macro 'QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG'
    F(QVector) \
      ^
src/devicecontrols.cpp:709:25: error: implicit instantiation of undefined template 'QVector<UvcControl>'
    for (auto &control: controls())
                        ^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:2133:1: note: template is declared here
QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG(Q_DECLARE_SEQUENTIAL_CONTAINER_METATYPE_ITER)
^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:215:7: note: expanded from macro 'QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG'
    F(QVector) \
      ^
src/devicecontrols.cpp:715:25: error: implicit instantiation of undefined template 'QVector<UvcControl>'
    for (auto &control: controls())
                        ^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:2133:1: note: template is declared here
QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG(Q_DECLARE_SEQUENTIAL_CONTAINER_METATYPE_ITER)
^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:215:7: note: expanded from macro 'QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG'
    F(QVector) \
      ^
src/devicecontrols.cpp:719:23: error: implicit instantiation of undefined template 'QVector<UvcControl>'
    return &controls().first();
                      ^
[... 18 lines cut ...]
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:2133:1: note: template is declared here
QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG(Q_DECLARE_SEQUENTIAL_CONTAINER_METATYPE_ITER)
^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:215:7: note: expanded from macro 'QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG'
    F(QVector) \
      ^
src/devicecontrols.cpp:726:25: error: implicit instantiation of undefined template 'QVector<UvcControl>'
    for (auto &control: controls())
                        ^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:2133:1: note: template is declared here
QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG(Q_DECLARE_SEQUENTIAL_CONTAINER_METATYPE_ITER)
^
/Users/test/Qt5.12.2/5.12.2/clang_64/lib/QtCore.framework/Headers/qmetatype.h:215:7: note: expanded from macro 'QT_FOR_EACH_AUTOMATIC_TEMPLATE_1ARG'
    F(QVector) \
      ^
11 errors generated.
make[4]: *** [debug/Qt5.12.2/clangpp/x86_64/obj/devicecontrols.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [sub-src-avfoundation-make_first-ordered] Error 2
make[2]: *** [sub-VideoCapture-make_first-ordered] Error 2
make[1]: *** [sub-Plugins-make_first-ordered] Error 2
make: *** [sub-libAvKys-make_first-ordered] Error 2
10:42:13: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project Webcamoid (kit: Desktop Qt 5.12.2 clang 64bit)
When executing step "Make"

hipersayanX, 2019-09-27:
Fixed, it may be working now.

GoEZM, 2019-10-01:
thank you for your help.
I have downloaded 8.7.0, and remove all config files, but the camera image still does not display.
I checked the log message:
CMIO_Unit_Input_Device.cpp:244:GetPropertyInfo CMIOUInputFromProcs::GetPropertyInfo() failed for id 102, Error: -67456
I already have an authorized camera to use:
[image]

But it still doesn't work. What programs do I need to check?

GoEZM, 2019-10-01:
17:42:15: Debugging starts
2019-10-01 17:42:23.632506+0800 webcamoid[2026:7707] QML debugging is enabled. Only use this in a safe environment.
2019-10-01 17:42:24.925375+0800 webcamoid[2026:7763] QML Debugger: Waiting for connection on port 49177...
2019-10-01 17:42:27.365938+0800 webcamoid[2026:7707] [AudioHAL_Client] HALC_ShellObject.cpp:212:HasProperty:  HALC_ShellObject::HasProperty: initial inner method call failed, Error: 560947818 (!obj)
2019-10-01 17:42:27.366391+0800 webcamoid[2026:7707] [AudioHAL_Client] HALC_ShellObject.cpp:212:HasProperty:  HALC_ShellObject::HasProperty: initial inner method call failed, Error: 560947818 (!obj)
2019-10-01 17:42:27.374898+0800 webcamoid[2026:7707] [AudioHAL_Client] HALC_ShellObject.cpp:212:HasProperty:  HALC_ShellObject::HasProperty: initial inner method call failed, Error: 560947818 (!obj)
2019-10-01 17:42:27.374940+0800 webcamoid[2026:7707] [AudioHAL_Client] HALC_ShellObject.cpp:212:HasProperty:  HALC_ShellObject::HasProperty: initial inner method call failed, Error: 560947818 (!obj)
[2019-10-01 17:42:28.206, 0x1002785c0]  IpcBridge::IpcBridge()
[2019-10-01 17:42:28.212, 0x1002785c0]  IpcBridge::connectService()
[2019-10-01 17:42:28.212, 0x1002785c0]  IpcBridge::registerPeer()
[2019-10-01 17:42:28.213, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:28.218, 0x1002785c0]  IpcBridge::setDriverPaths()
2019-10-01 17:42:30.093850+0800 webcamoid[2026:7760] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=2026
2019-10-01 17:42:30.093988+0800 webcamoid[2026:7760] SecTaskCopyDebugDescription: webcamoid[2026]/0#-1 LF=0
2019-10-01 17:42:30.662879+0800 webcamoid[2026:7707] WARNING: Cannot find style "Universal" - fallback: "/Users/jack/Qt5.12.2/5.12.2/clang_64/qml/QtQuick/Controls/Styles/Desktop"
2019-10-01 17:42:31.583689+0800 webcamoid[2026:7707] qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[2019-10-01 17:42:31.726, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:31.726, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:31.726, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:31.726, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:31.726, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:31.726, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:31.727, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:31.727, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:31.731, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:31.731, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:31.731, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:31.731, 0x1002785c0]  IpcBridge::listDevices()
2019-10-01 17:42:32.018666+0800 webcamoid[2026:7785] [] CMIO_Unit_ScopeElement.h:200:SafeGetElement Throwing err: -67454
2019-10-01 17:42:32.024894+0800 webcamoid[2026:7785] [] CMIOUnitFigBaseObjectImpl.c:246:CMIOUnitCreateFromDescription Invalid paramater
2019-10-01 17:42:32.080781+0800 webcamoid[2026:7785] [] CMIO_Unit_Input_Device.cpp:244:GetPropertyInfo CMIOUInputFromProcs::GetPropertyInfo() failed for id 102, Error: -67456
2019-10-01 17:42:32.421967+0800 webcamoid[2026:7707] MessageTracer: Falling back to default whitelist
2019-10-01 17:42:35.095721+0800 webcamoid[2026:7785] [] CMIO_DAL_PlugIn.cpp:202:ObjectSetPropertyData got an error from the plug-in routine, Error: -536870186
2019-10-01 17:42:36.606952+0800 webcamoid[2026:7785] [] CMIO_DAL_PlugIn.cpp:269:StreamCopyBufferQueue got an error from the plug-in routine, Error: 1852797029
2019-10-01 17:42:36.608471+0800 webcamoid[2026:7707] IMKClient Stall detected, *please Report* your user scenario attaching a spindump (or sysdiagnose) that captures the problem - (imkxpc_bundleIdentifierWithReply:) block performed very slowly (4.11 secs).
2019-10-01 17:42:36.691712+0800 webcamoid[2026:7785] [] CMIO_Unit_ScopeElement.h:200:SafeGetElement Throwing err: -67454
2019-10-01 17:42:36.694135+0800 webcamoid[2026:7785] [] CMIOUnitFigBaseObjectImpl.c:246:CMIOUnitCreateFromDescription Invalid paramater
2019-10-01 17:42:36.696859+0800 webcamoid[2026:7785] [] CMIO_Unit_Input_Device.cpp:244:GetPropertyInfo CMIOUInputFromProcs::GetPropertyInfo() failed for id 102, Error: -67456
[2019-10-01 17:42:36.767, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:36.767, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:36.767, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:36.767, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:36.767, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:36.767, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:36.768, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:36.768, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:36.771, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:36.771, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:36.771, 0x1002785c0]  IpcBridge::listDevices()
[2019-10-01 17:42:36.772, 0x1002785c0]  IpcBridge::listDevices()
2019-10-01 17:42:36.772945+0800 webcamoid[2026:7707] qrc:/VirtualCamera/share/qml/main.qml:25:1: QML GridLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
2019-10-01 17:42:37.136362+0800 webcamoid[2026:7707] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=2026
2019-10-01 17:42:37.136461+0800 webcamoid[2026:7707] SecTaskCopyDebugDescription: webcamoid[2026]/0#-1 LF=0

hipersayanX, 2019-10-01:
I can't give you any answer now, this is maybe something related to permissions but not sure, I must investigate what causing this bug.

hipersayanX, 2019-10-06:
Ok, I've made a lot of changes to the code, Webcamod will now request camera access permissions if it were not granted, also I've disabled the camera controls (I think that could be conflicting with AVFoundation), also Qt seems to have changed the internal structure of the resources and the virtual camera was not working properly due to this, so now the default frame is loaded directly from the file. Also, upgraded the build image to XCode 9.3 (OS X 10.13), so it may not have problems with newer versions of the system.
The changes has been already pushed to the master and can also be tested in the daily builds.
I hope this solve most of the problems with Mac.

GoEZM, 2019-10-07:
Thank you very much for your support.
I downloaded the latest version on github, but still can't display the webcam image.

"upgraded the build image to XCode 9.3 (OS X 10.13)"
--> Do you mean to use XCode 10.3? Because using 9.3 will cause errors.
    ( undefined [authorizationStatusForMediaType...... )

GoEZM, 2019-10-07:
In macOS 10.14.6 XCode 10.1 Qt 10.12.2

I traced your latest code:
1)
void CaptureAvFoundation::updateDevices()
{
    If (!CaptureAvFoundationPrivate::canUseCamera()) {  
        Return;
    }
.......
**canUseCamera() return true, So webcam is usable.**
2)
In deviceobserver.mm
-(void) captureOutput: (AVCaptureOutput *) captureOutput
         didOutputSampleBuffer: (CMSampleBufferRef) videoFrame
         fromConnection: (AVCaptureConnection *) connection

**The function has been triggered, and the frame packet is transmitted to 
Convertvideogeneric.cpp:**

void ConvertVideoGeneric::packetEnqueue(const AkPacket &packet)
{
    Auto fourcc = packet.caps().property("fourcc").toString();
    AkVideoPacket videoPacket(packet);

    If (fourcc == "JPEG") {
        videoPacket =
                AkVideoPacket::fromImage(QImage::fromData(packet.buffer()),
                                         Packet);
    } else {
        AkVideoCaps caps (fourccToFormat->value(fourcc,
                                               AkVideoCaps::Format_none),
                         Packet.caps().property("width").toInt(),
                         Packet.caps().property("height").toInt(),
                         Packet.caps().property("fps").toString());

        If (packet.caps().contains("align"))
            caps.setAlign(packet.caps().property("align").toInt());

        videoPacket.caps() = caps;
        videoPacket.buffer() = packet.buffer();
        videoPacket.copyMetadata(packet);
        videoPacket = videoPacket.convert(AkVideoCaps::Format_rgb24);
    }

    If (videoPacket)
        Emit this->frameReady(videoPacket);   **<--- This function will not be executed**
}

hipersayanX, 2019-10-07:
>Do you mean to use XCode 10.3?

Nope, exactly as it say [XCode 9.3](https://docs.travis-ci.com/user/reference/osx/#macos-version).

>Because using 9.3 will cause errors.
>( undefined [authorizationStatusForMediaType...... )

Did not checked that, I assumed the code was fine because Travis did not marked error :confused:, will upgrade to XCode 10.0 that is the next that don't give error.

>The function has been triggered, and the frame packet is transmitted to
>Convertvideogeneric.cpp:

That is the problem, I wrote the generic converter for Android but decided to enable it for other OS just in case, the generic converter only handle a very small number of formats, and that is the reason because it is returning an invalid packet, go to _Preferences > General Options > Video convert/Audio convert/Video playback/Video record_ and select ffmpeg, that must solve the problem.

GoEZM, 2019-10-14:
I have tried to add the "ffmpeg" library:
**Step 1. Install the required dependencies:**
brew install  p7zip  python3  ccache  pkg-config  ffmpeg
brew install  gstreamer  gst-plugins-base  pulseaudio  jack  libuvc

**Step 2:**
  Projects > Build & Run > Build Steps 
  In "Additional arguments" add:
  FFMPEGINCLUDES=/[ffmpeg path]/include FFMPEGLIBS=-L/[ffmpeg path]/lib FFMPEGLIBS+=-lavformat FFMPEGLIBS+=-lavcodec GSTREAMERINCLUDES=/[gstream path]/include GSTREAMERLIBS=-L/[gstream path]/lib GSTREAMERLIBS+=-lgstaudio GSTREAMERLIBS+=-lgstvideo LIBUSBINCLUDES=/[libusb path]/include LIBUSBLIBS=-L/[libusb path]/lib LIBUSBLIBS+=-lusb LIBUVCINCLUDES=/[libuvc path]/include LIBUVCLIBS=-L/[libuvc path]/lib LIBUVCLIBS+=-luvc 

**I also tried to use the commands:**
~/[Qt path]/qmake Webcamoid.pro FFMPEGINCLUDES=/[ffmpeg path]/include FFMPEGLIBS=-L/[ffmpeg path]/lib FFMPEGLIBS+=-lavformat FFMPEGLIBS+=-lavcodec GSTREAMERINCLUDES=/[gstream path]/include GSTREAMERLIBS=-L/[gstream path]/lib GSTREAMERLIBS+=-lgstaudio GSTREAMERLIBS+=-lgstvideo LIBUSBINCLUDES=/[libusb path]/include LIBUSBLIBS=-L/[libusb path]/lib LIBUSBLIBS+=-lusb LIBUVCINCLUDES=/[libuvc path]/include LIBUVCLIBS=-L/[libuvc path]/lib LIBUVCLIBS+=-luvc

**But there is no "ffmpeg" option to choose from!**
[image]

Am I missing any settings?

hipersayanX, 2019-10-14:
Did you added _/usr/local/bin_ to the **PATH** build environment variable? (step 14.) otherwise Qt Creator can't detect pkg-config.
Remove the additional arguments, these are not necessary since pkg-config is who locate the required libraries automatically.
Also when you recompile the project does this table appears at the beginning of the build?:

```
Checking for alsa... no
Checking for avfoundation... yes
Checking for cmio... yes
Checking for coreaudio... yes
Checking for dshow... no
Checking for ffmpeg... yes
Checking for ffmpeg_avcodec_sendrecv... yes
Checking for ffmpeg_avcodec_subtitledata... yes
Checking for ffmpeg_avformat_codecpar... yes
Checking for ffmpeg_avutil_extraoptions... yes
Checking for ffmpeg_avutil_sampleformat64... yes
Checking for ffmpeg_avresample... yes
Checking for ffmpeg_swresample... yes
Checking for mediafoundation... no
Checking for ndk_audio... no
Checking for ndk_camera... no
Checking for ndk_media... no
Checking for opensl... no
Checking for v4l2... no
Checking for wasapi... no
```

GoEZM, 2019-10-15:
**I have added /usr/local/bin to my PATH, but the "ffmpeg" function is off:**
Checking for alsa... no
Checking for avfoundation... yes
Checking for cmio... yes
Checking for coreaudio... yes
Checking for dshow... no
Checking for ffmpeg... no
Checking for gstreamer... no
Checking for jack... no
Checking for libuvc... no
Checking for mediafoundation... no
Checking for ndk_audio... no
Checking for ndk_camera... no
Checking for ndk_media... no
Checking for opensl... no
Checking for pulseaudio... no
Checking for v4l2... no
Checking for wasapi... no

**and pkg-config list:**
/usr/local/bin/pkg-config --list-all | grep FFmpeg
libpostproc                libpostproc - FFmpeg postprocessing library
libswscale                 libswscale - FFmpeg image rescaling library
libavdevice                libavdevice - FFmpeg device handling library
libavutil                  libavutil - FFmpeg utility library
libavformat                libavformat - FFmpeg container format library
libavfilter                libavfilter - FFmpeg audio/video filtering library
libavcodec                 libavcodec - FFmpeg codec library
libswresample              libswresample - FFmpeg audio resampling library

**How to turn on the "FFMpeg" function?**

hipersayanX, 2019-10-15:
It's say no because pkg-config don't get detected. Does your Qt Creator settings looks like this?
 
[image]

hipersayanX, 2019-10-25:
Where you able to solve your problem?

GoEZM, 2019-10-28:
Sorry for replying too late
I haven't found the cause of the problem set by Qt.
So I ignore the problem for a while and try to write a video processing program myself.
I will try to solve the Qt setting problem later.

GoEZM, 2019-12-19:
I have found the cause of the problem:
I downloaded  "dmg file" from the official website and installed Qt, which caused errors in the identification software.

if use the Qt installed by Homebrew, Qt Creator Can correctly identify all the packages in the build process.

hipersayanX, 2019-12-19:
Good! :+1:

closed 2019-12-19 by hipersayanX