webcamoid/webcamoid #194: Can't build on Mac: couldn't find /usr/local/Cellar/qt

SheerLin, 2019-10-03:
## Summary

Can not build the project on MAC. I stuck in the step 5 in this documentation:
https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Mac

When I tried to add qt version, I just could not find the directory /usr/local/Cellar/qt.
I have followed the step 1, 3, 4 and then got stuck.
Step 2 was skipped because I was also confused about how to install these dependencies.(But at least these dependencies are irrelevant to this problem?)

## Your Environment
* Webcamoid information: Checked out from last commit bfb3ae84aac968bc7778bf5198dfc632a9aa0adb
* Operating System information: MacOS Mojave 10.14

hipersayanX, 2019-10-03:
You must install the **qt5** package (```brew install qt5```).

SheerLin, 2019-10-03:
Thank you! After I install the qt5 package, this folder appears and this problem is solved.

closed 2019-10-03 by SheerLin