webcamoid/webcamoid #195: Webcamoid crashed right after build and run

SheerLin, 2019-10-04:
## Summary
Webcamoid crashed right after build and run.

I have also installed CamTwist on my mac and I noticed that in the log the camtwist is discovered:
2019-10-03 16:57:19.893 webcamoid[3970:26576] Driver name is 8 CamTwist

Is this the reason why webcamoid crashed?
How can I fix it without uninstalling camtwist?

## Logs
I have replaced my real path with "[path]" and pasted the logs below:

16:57:19: Starting [path]/webcamoid/StandAlone/debug/Qt5.13.1/clangpp/x86_64/bin/webcamoid --vcam [path]/webcamoid/libAvKys/Plugins/VirtualCamera/src/cmio/VirtualCamera -r -p [path]/webcamoid/libAvKys/Plugins -q [path]/webcamoid/libAvKys...
QML debugging is enabled. Only use this in a safe environment.
2019-10-03 16:57:19.892 webcamoid[3970:26576] alloc
2019-10-03 16:57:19.892 webcamoid[3970:26576] query 2
2019-10-03 16:57:19.892 webcamoid[3970:26576] release 2
2019-10-03 16:57:19.892 webcamoid[3970:26576] ** MyPlugInitializeWithObjectID
2019-10-03 16:57:19.892 webcamoid[3970:26576] init
2019-10-03 16:57:19.892 webcamoid[3970:26576] MyPlugObjectSetPropertyData
2019-10-03 16:57:19.892 webcamoid[3970:26576] MyPlugObjectSetPropertyData
2019-10-03 16:57:19.892 webcamoid[3970:26576] MyPlugObjectSetPropertyData
2019-10-03 16:57:19.893 webcamoid[3970:26576] Trying to get Surface
2019-10-03 16:57:19.893 webcamoid[3970:26576] Unable to lookup surface
2019-10-03 16:57:19.893 webcamoid[3970:26576] Surface is 0x0
2019-10-03 16:57:19.893 webcamoid[3970:26576] Driver name is 8 CamTwist
2019-10-03 16:57:19.893 webcamoid[3970:26576] <CMVideoFormatDescription 0x7fbebe00ec80 [0x7fff908ae8e0]> {
	mediaType:'vide' 
	mediaSubType:'BGRA' 
	mediaSpecific: {
		codecType: 'BGRA'		dimensions: 320 x 240 
	} 
	extensions: {<CFBasicHash 0x7fbebe0350f0 [0x7fff908ae8e0]>{type = immutable dict, count = 4,
entries =>
	1 : <CFString 0x7fff90c904d8 [0x7fff908ae8e0]>{contents = "CVFieldCount"} = <CFNumber 0x715f8037ae212b9d [0x7fff908ae8e0]>{value = +1, type = kCFNumberSInt64Type}
	2 : <CFString 0x1173fa8a0 [0x7fff908ae8e0]>{contents = "CVBytesPerRow"} = <CFNumber 0x715f8037ae242a9d [0x7fff908ae8e0]>{value = +1280, type = kCFNumberSInt64Type}
	4 : <CFString 0x7fff90b2df90 [0x7fff908ae8e0]>{contents = "com.apple.cmio.format_extension.video.only_has_i_frames"} = <CFBoolean 0x7fff908aec30 [0x7fff908ae8e0]>{value = true}
	5 : <CFString 0x7fff90b1a8d8 [0x7fff908ae8e0]>{contents = "FormatName"} = <CFString 0x1173fa8c0 [0x7fff908ae8e0]>{contents = "Component Video - CCIR-601 RGB"}
}
}
}
2019-10-03 16:57:19.893 webcamoid[3970:26576] Name CamTwist
2019-10-03 16:57:19.893 webcamoid[3970:26576] alloc
2019-10-03 16:57:19.893 webcamoid[3970:26576] query 2
2019-10-03 16:57:19.893 webcamoid[3970:26576] release 2
2019-10-03 16:57:19.893 webcamoid[3970:26576] ** MyPlugInitializeWithObjectID
2019-10-03 16:57:19.893 webcamoid[3970:26576] init
2019-10-03 16:57:19.893 webcamoid[3970:26576] MyPlugObjectSetPropertyData
2019-10-03 16:57:19.893 webcamoid[3970:26576] MyPlugObjectSetPropertyData
2019-10-03 16:57:19.893 webcamoid[3970:26576] MyPlugObjectSetPropertyData
2019-10-03 16:57:19.893 webcamoid[3970:26576] <CMVideoFormatDescription 0x7fbebe036d90 [0x7fff908ae8e0]> {
	mediaType:'vide' 
	mediaSubType:'2vuy' 
	mediaSpecific: {
		codecType: '2vuy'		dimensions: 320 x 240 
	} 
	extensions: {<CFBasicHash 0x7fbebe036df0 [0x7fff908ae8e0]>{type = immutable dict, count = 6,
entries =>
	0 : <CFString 0x7fff90c90658 [0x7fff908ae8e0]>{contents = "CVImageBufferYCbCrMatrix"} = <CFString 0x7fff90c90698 [0x7fff908ae8e0]>{contents = "ITU_R_601_4"}
	1 : <CFString 0x7fff90c904d8 [0x7fff908ae8e0]>{contents = "CVFieldCount"} = <CFNumber 0x715f8037ae212b9d [0x7fff908ae8e0]>{value = +1, type = kCFNumberSInt64Type}
	2 : <CFString 0x7fff90c907b8 [0x7fff908ae8e0]>{contents = "CVImageBufferTransferFunction"} = <CFString 0x7fff90c90678 [0x7fff908ae8e0]>{contents = "ITU_R_709_2"}
	4 : <CFString 0x7fff90b2df90 [0x7fff908ae8e0]>{contents = "com.apple.cmio.format_extension.video.only_has_i_frames"} = <CFBoolean 0x7fff908aec30 [0x7fff908ae8e0]>{value = true}
	5 : <CFString 0x7fff90c90738 [0x7fff908ae8e0]>{contents = "CVImageBufferColorPrimaries"} = <CFString 0x7fff90c90778 [0x7fff908ae8e0]>{contents = "SMPTE_C"}
	6 : <CFString 0x7fff90b1a8d8 [0x7fff908ae8e0]>{contents = "FormatName"} = <CFString 0x1173fa920 [0x7fff908ae8e0]>{contents = "Component Video - CCIR-601 uyvy"}
}
}
}
2019-10-03 16:57:19.893 webcamoid[3970:26576] Name CamTwist (2VUY)
2019-10-03 16:57:19.931 webcamoid[3970:26576] MyPlugObjectHasProperty
2019-10-03 16:57:19.932 webcamoid[3970:26576] enog
[... 109 lines cut ...]
2019-10-03 16:57:19.935 webcamoid[3970:26576] bolg
2019-10-03 16:57:19.935 webcamoid[3970:26576] 0
2019-10-03 16:57:19.935 webcamoid[3970:26576] MyPlugObjectHasProperty
2019-10-03 16:57:19.935 webcamoid[3970:26576] trfn
2019-10-03 16:57:19.935 webcamoid[3970:26576] bolg
2019-10-03 16:57:19.935 webcamoid[3970:26576] 0
2019-10-03 16:57:19.935 webcamoid[3970:26576] DN: CamTwist (2VUY)
2019-10-03 16:57:19.935 webcamoid[3970:26576] MyPlugObjectHasProperty
2019-10-03 16:57:19.935 webcamoid[3970:26576] dulp
2019-10-03 16:57:19.935 webcamoid[3970:26576] tpni
2019-10-03 16:57:19.935 webcamoid[3970:26576] 0
2019-10-03 16:57:19.935 webcamoid[3970:26576] MyPlugObjectGetPropertyDataSize
2019-10-03 16:57:19.935 webcamoid[3970:26576] sel dnwo
2019-10-03 16:57:19.935 webcamoid[3970:26576] scope bolg
2019-10-03 16:57:19.936 webcamoid[3970:26576] elt 0
[2019-10-03 16:57:19.997, 0x1181385c0]  IpcBridge::IpcBridge()
libc++abi.dylib: terminating with uncaught exception of type std::bad_alloc: std::bad_alloc
16:57:20: The program has unexpectedly finished.
16:57:20: The process was ended forcefully.
16:57:20: [path]/webcamoid/StandAlone/debug/Qt5.13.1/clangpp/x86_64/bin/webcamoid crashed.

## Your Environment

- Webcamoid information: Checked out from last commit bfb3ae84aac968bc7778bf5198dfc632a9aa0adb
- Operating System information: MacOS Mojave 10.14

SheerLin, 2019-10-04:
In Debugger
[image]

SheerLin, 2019-10-04:
**If I run in Debug mode:**

18:45:58: Debugging starts
QML debugging is enabled. Only use this in a safe environment.
2019-10-03 18:46:03.083256-0700 webcamoid[17520:154868] QML Debugger: Waiting for connection on port 51083...
2019-10-03 18:46:11.115267-0700 webcamoid[17520:154769] [AudioHAL_Client] HALC_ShellObject.cpp:212:HasProperty:  HALC_ShellObject::HasProperty: initial inner method call failed, Error: 560947818 (!obj)
2019-10-03 18:46:11.115296-0700 webcamoid[17520:154769] [AudioHAL_Client] HALC_ShellObject.cpp:212:HasProperty:  HALC_ShellObject::HasProperty: initial inner method call failed, Error: 560947818 (!obj)
2019-10-03 18:46:11.117883-0700 webcamoid[17520:154769] [AudioHAL_Client] HALC_ShellObject.cpp:212:HasProperty:  HALC_ShellObject::HasProperty: initial inner method call failed, Error: 560947818 (!obj)
2019-10-03 18:46:11.117925-0700 webcamoid[17520:154769] [AudioHAL_Client] HALC_ShellObject.cpp:212:HasProperty:  HALC_ShellObject::HasProperty: initial inner method call failed, Error: 560947818 (!obj)
[2019-10-03 18:46:11.234, 0x10029b5c0]  IpcBridge::IpcBridge()
libc++abi.dylib: terminating with uncaught exception of type std::bad_alloc: std::bad_alloc
18:46:21: Debugging has finished

GoEZM, 2019-10-04:
I also found this problem, the temporary solution is to delete all Webcamoid config files in ~/Library/Preferences.

SheerLin, 2019-10-04:
> I also found this problem, the temporary solution is to delete all Webcamoid config files in ~/Library/Preferences.

Thank you so much! This workaround works! 

But after I open the application, Webcamoid is not connected to my webcam(FaceTime HD Camera). I mean there is the option for FaceTime HD Camera but no streaming displays while the green light of webcam is on. 

Did you experience the same thing after you delete the config files? (Or maybe this is another issue)

GoEZM, 2019-10-04:
The problem is similar to #192, and I have no answer yet.

SheerLin, 2019-10-04:
> The problem is similar to #192, and I have no answer yet.

Thanks!

hipersayanX, 2019-10-06:
Yes I think this is related to #192, here is a [comment update](https://github.com/webcamoid/webcamoid/issues/192#issuecomment-538701406) to the issue.
@SheerLin can you test the [latest build](https://bintray.com/webcamoid/webcamoid/webcamoid/daily#files/mac)?

hipersayanX, 2019-10-25:
Duplicate of #192, closing.

closed 2019-10-25 by hipersayanX