webcamoid/webcamoid #196: Memory usage continues to grow

kenfa01, 2019-10-08:
In macOS. When the applications connect to the Virtual Camera of the webcamoid, the memory usage of the webcamoid will continue to grow. In the end, it will cause the entire system to run very slowly!

Is there a way to limit the memory cap used by the webcamoid?
Thank you.

Before:
[image]

When an application connected:
[image]

hipersayanX, 2019-10-14:
This was fixed in the [daily build](https://bintray.com/webcamoid/webcamoid/webcamoid/daily#files/mac), install it or copy the portable version to the /Applications folder, remove all virtual cameras, restart Webcamoid and create the virtual camera again.

kenfa01, 2019-10-18:
There is no USB camera now, just test the screenshot.

I have downloaded the latest version from github.
But connecting to "virtual camera" from other programs will have no image output! (The old version has image output)
Thank you.

hipersayanX, 2019-10-18:
Try [resetting the configs](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh#mac).

hipersayanX, 2019-10-25:
Fixed in latest release. Remember to remove all virtual cameras and create a new one.

closed 2019-10-25 by hipersayanX