webcamoid/webcamoid #198: no matter what i tried, when i go to record a video , there is no audio

ladytester, 2019-10-14:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debbuger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates bellow that fit better your issue.

# Reporting a problem?

## Summary

Write here a brief description of the problem.

## Current Behavior

Describe the problem the best as you can, don't omit information.

## Expected Behavior

How it should have work?

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Open this
2.  Click that
3.  Drag those
4.  ...

## Suggestions and tips

How would you solve the problem?

## Your Environment

* Webcamoid information: (version) (architecture)
* Operating System information: (name) (numeric version and codename if applied) (architecture)
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questinons and other matters?

Go ahead!
--------------
10-13-19
this is in the installable version, linux, 64bit, webcamoid 8.7.0
no matter what i try, when i record a video , there is not audio, please fix this

hipersayanX, 2019-10-14:
Which distro?

ladytester, 2019-10-14:
i use linux lite 64bit, i uninstalled the webcamoid 8.7.0 , and downloaded one of the daily builds on 10-13-19, "webcamoid-daily-x86_64.AppImage"    , so far it worked perfectly, pretty nice for a daily build, i was able to record a webcam video with audio no problems.......here is the link to the daily builds https://bintray.com/webcamoid/webcamoid/webcamoid/daily#files/linux , sorry this was not much help,

hipersayanX, 2019-10-14:
Oh ok, so probably it was a bug in the stable release, will check anyway just in case.

hipersayanX, 2019-10-14:
I tested it and it's working find probably the mic was muted, or the wrong audio source was selected, for example in the following picture there are 3 audio sources: the integrated webcam (that has not audio), the speaker monitor (that is audio that comes from your desktop applications, for example your video player), and the mic:

[image]

in which case you must choose 3rd one.
Audio recording needs some improvement anyway.

hipersayanX, 2019-10-25:
Problem seems to be solved, closing.

closed 2019-10-25 by hipersayanX