webcamoid/webcamoid #199: What if the video source is no longer available?

realbsb, 2019-10-21:
The program freezes in Windows 10 if you open it with a video source that is no longer available. It is impossible to change to valid - just the application is not responding.

hipersayanX, 2019-10-21:
Open regedit HKEY_CURRENT_USER\Software\Webcamoid\Webcamoid\StreamConfigs\stream and tell me what is the value.

realbsb, 2019-10-23:
Unfortunately, I have already reset the settings. The problem arose when I tried to watch the already inaccessible RTSP

realbsb, 2019-10-23:
oh, or at that time I really had a virtual camera

hipersayanX, 2019-10-23:
Normally, there is no problem if for example you added a video file and later you erase it from your disk, Webcamoid will simply not play it, the same may happen with RTSP streams, if Webcamoid can't connect to the resource, it will simply not play it, but won't hang.
That's why I think the problem is most probably that you had the vcam enabled and selected, and then when you played it you created a deadlock, which in theory may not happen as far as I had tested.
Beside that I can't give you a better answer, because I have no idea what was your settings at that time.

jmichael2497, 2021-03-07:
on portable 8.8.0 portable 64 w10 seeing similar behavior messing with rtsp and http stream attempts that sometimes lead to hang crash auto-close. usually opening the config ini and deleting the entries in there will fix the issue, otherwise attempts to open give no feedback.

 (i'm trying to find light resource foss methods of using android device as virtual camera for pc video conferencing, so far this program seems a bit inconsistent on old 4GB ram system, but the closest so far.)

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX