webcamoid/webcamoid #201: Virtual Camera Driver Mac

WhiteHatArpit, 2019-11-06:
Hi,

I am working on a Video Processing pipeline which should take input video stream from webcam and output processed video stream to applications like Skype, Hangouts, Zoom etc. 

The pipeline uses OpenCV and is written in Python.
Using Akvcam virtual driver, I have successfully achieved this on Linux.

I am not sure on how to do it for Mac.

Looking at Webcamoid release log, I see that you have written the virtual camera driver for Mac. I was wondering, why can't it be used in the same way as Akvcam? Also, if I want to use it for my application, what do you suggest I should do to integrate?

I can refine my questions according to the directions.

Best,
Arpit

hipersayanX, 2019-11-06:
>I was wondering, why can't it be used in the same way as Akvcam?

In Linux, the virtual camera is implemented as a kernel module (a driver basically), and implements the [V4L2 API](https://linuxtv.org/downloads/v4l-dvb-apis/kapi/v4l2-intro.html) which already define the protocol for capture/output devices and how the frames must be transferred between them.

In Mac the virtual camera is implemented as a [plugin](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/cmio/VirtualCamera) in the user space, the plugin is loaded in the space of the target program and it uses a [central server](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/cmio/Assistant) to communicate data between Webcamoid an the target program. The Windows virtual camera works in a similar way to the Mac virtual camera.

In Mac, CMIO only defines the protocol for the capture side, so I had the define my own protocol for to output. A virtual camera is so close to the internals of the operating systems that is impossible to implement in the same way for all of them, and the shared code is minimum.

>Also, if I want to use it for my application, what do you suggest I should do to integrate?

The best way is linking your program with [VCamIPC](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/cmio/VCamIPC) you can do that [from Python](https://docs.python.org/3/extending/extending.html), The [VirtualCamera element](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/virtualcameraelement.cpp) is a good example on how you can communicate with VCamIPC. The program linking to VCamIPC must be GPLv3 or later.

WhiteHatArpit, 2019-11-06:
Thanks for the very informative answer.

What I understand is:
Linux: virtual camera is installed as a kernel module on its own
Mac/Windows: virtual camera implementation has to be part of an application

I should look through VirtualCameraElement to understand how it uses VCamIPC, and adopt it in my application. I am working on a class project, so I believe everything would be as open source as it can be. I will make sure to give due credits and use appropriate licences.

I am not familiar with development for Mac. Right now, my program is just a python script/process which should run in background. To make use of the VCamIPC, would I need to convert it into a Mac application (package and other things), or can I use this plugin code via Python script?
(The answer might be in the Python resource you have provided, I will go through it too.)

Thanks,
Arpit

hipersayanX, 2019-11-07:
>To make use of the VCamIPC, would I need to convert it into a Mac application (package and other things), or can I use this plugin code via Python script?

Not necessary. Let suppose the easy case in which you have the virtual camera installed by Webcamoid in your system. Now you want to talk with the virtual camera through VCamIPC, VCamIPC is a C++ library (compiled statically by default), you will probably need to write a wrapper dynamic library that links against VCamIPC and that will be loaded from your Python script. PyQt is a good example of a library doing that, and many other libraries do the same.

VCamIPC does not have any Qt dependency ***, and only depends on the system libraries, so you may not have much problem writing a wrapper for it, [the API](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/ipcbridge.h) is pretty straightforward.

Also, for compiling VCamIPC only you must compile [VCamUtils](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/VCamUtils) first, or you can compile the whole camera (and VCamIPC with it) by passing **VIRTUALCAMERAONLY** to [VirtualCamera.pro](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/VirtualCamera.pro), the virtual camera does not have any Qt dependency ***.

*** No Qt dependency for running just for compiling.

closed 2020-02-27 by hipersayanX

andy2301, 2020-03-16:
@WhiteHatArpit I'm on the same mission as you. Just want to quickly check in: did you make progress on this? Maybe we can collaborate to speed up and contribute back to the awesome project @hipersayanX created. Let me know. Thanks.

WhiteHatArpit, 2020-03-17:
@andy2301 For my research project, we chose to go with Linux only in the interest of time.
Thus, we don't have any progress for Mac.