webcamoid/webcamoid #202: Mirror/flip capability needed

adamf663, 2019-11-13:
When working with a usb microscope, it is a major hasle dealing with an image that is flipped and/or mirrored.  When trying to position an object into view, it is a pain when moving the object left causes the image to move right or when moving the object up causes the image to move down.
I couldn't find any plugin or settings page that offered such functionality.

The gnu guvcview application has such functionality.

WolfgangFahl, 2020-03-19:
Good that it's not low priority any more ...

hipersayanX, 2020-03-21:
@WolfgangFahl I had removed all Priority tags because they were subjective.

davipatti, 2020-08-24:
Mirror / flip and rotation operations can be achieved by the Matrix Transform effect.

For instance, enter this matrix for a y-flip:

~~~
-1   0   0
 0  -1   0
~~~

hipersayanX, 2020-08-24:
@davipatti maths magic :wink: 

https://www.khanacademy.org/math/precalculus/x9e81a4f98389efdf:matrices/x9e81a4f98389efdf:matrices-as-transformations/v/transforming-position-vector
https://en.wikipedia.org/wiki/Transformation_matrix

hipersayanX, 2022-02-15:
Feature added in 9.0.0, closing.

closed 2022-02-15 by hipersayanX

dchmelik, 2022-11-09:
> Mirror / flip and rotation operations can be achieved by the Matrix Transform effect.
> 
> For instance, enter this matrix for a y-flip:
> 
> ```
> -1   0   0
>  0  -1   0
> ```
```
-1, 0, 0
 0, 1, 0
```