webcamoid/webcamoid #203: Hide Webcamoid bottom menu bar?

zzzinha, 2019-11-15:
Hello, I am a basic user (not programmer) of Webcamoid. I am wondering if anyone knows a trick to hide the bottom menu bar ... or a way to go full screen that hides the menu bar.

Many thanks,
Teresa

hipersayanX, 2019-11-16:
Isn't possible right now, but no problem I can add that feature.

zzzinha, 2019-11-19:
Thank you so much, HipersayanX! I look forward!

wbraswell, 2020-04-25:
Any progress on this issue?  It is an important feature request.

bryanmtdt, 2020-04-25:
I need this feature too

nolfr, 2021-02-17:
Thanks for the reference to #203, do you have a schedule of when to add the feature?

hipersayanX, 2021-02-17:
@nolfr Nope, I've other priorities right now, but PRs are very welcome if someone want to work on this.

sanette, 2021-04-07:
that would be great indeed, for instance when you broadcast your screen. Maybe a simple trick would be to set the transparency to zero?

hipersayanX, 2021-04-08:
@sanette my idea is to add an option (maybe in the menu) that make a shrink effect and hide all the UI into a button in a corner, something similar to this:

[image]

then click the button and get the UI back with the opposite effect.
But right now I haven't time to implement it.

sanette, 2021-04-08:
did you also consider completely hiding the controls and make them appear automatically as soon as the user hovers the window?
It seems to me that this is a common choice for video players, like youtube

hipersayanX, 2021-04-08:
@sanette could be a mix of both approaches, shrinking the UI to a button, and making the button transparent when the mouse is outside the window.

rajhlinux, 2022-08-22:
I would like this feature too.

hipersayanX, 2023-03-05:
Closing this and joining with #581.

closed 2023-03-05 by hipersayanX