webcamoid/webcamoid #204: Flatpak release on flathub.org

ovari, 2019-11-21:
Are you able to advise when [Webcamoid](https://webcamoid.github.io/) will be available on [flathub.org](https://flathub.org/apps/search/webcamoid) as a flatpak?

Thank you

Linux Mint 19.2 Cinnamon 64-bit
[Jami](https://jami.net/)
[Webcamoid Virtual Camera](https://git.jami.net/savoirfairelinux/ring-daemon/issues/182#note_16303)

hipersayanX, 2019-11-21:
This is a pending task, I need to investigate all the Flatpak thing.

yookoala, 2020-12-16:
Any news on this feature yet?

hipersayanX, 2020-12-17:
@yookoala nope, PRs are very welcome :smile:

iamjameswalters, 2021-09-27:
Here to +1 this issue

Lunarequest, 2021-10-16:
It looks like you need to add this to the cxxflags `-include /usr/include/c++/11*/limits` based since gcc 11 https://stackoverflow.com/questions/4798936/numeric-limits-was-not-declared-in-this-scope-no-matching-function-for-call-t or explicitly add `#include <limits>` to any files that use `std::numeric_limits`

hipersayanX, 2021-10-21:
@Lunarequest you are reporting a bug that was fixed already in the master branch a long time ago, if you want to experiment with the Flatpak thing, do all your experiments with the master brach instead, it will be much more useful.

Lunarequest, 2021-10-21:
Okay will switch

Lunarequest, 2021-11-09:
I've gotten a working flatpak using this commit edab00879feb32311383d0d7cac5e3c946f4fced. There are some things that aren't possible like using the virtual web cam support and privilege escalation as suid binaries can't run in a flatpak

hipersayanX, 2021-11-09:
@Lunarequest it's ok, the point is to have at least a template to start working on, I will keep using AppImage and provide the flatpack as another option.
Also, the next step is integrating the flatpack templates into the [DeployTools](https://github.com/webcamoid/DeployTools).

Lunarequest, 2021-11-09:
There isn't really a need to integrate into deploy tools, flathub handles the building and publishing of flatpaks. Flathub also has https://github.com/flathub/flatpak-external-data-checker which can keep the dependencies up to date in an automated manner

ovari, 2021-11-10:
> There are some things that aren't possible like using the virtual web cam support and privilege escalation as suid binaries can't run in a flatpak

Can [Flatseal](https://flathub.org/apps/details/com.github.tchx84.Flatseal) help?

[image]

Thank you

Lunarequest, 2021-11-10:
> > There are some things that aren't possible like using the virtual web cam support and privilege escalation as suid binaries can't run in a flatpak
> 
> Can [Flatseal](https://flathub.org/apps/details/com.github.tchx84.Flatseal) help?
> 
> [image]
> 
> Thank you

No. The sandbox changes suid binaries to be regular ones. Its why we have to use zypak to override the sandbox of electron apps

hipersayanX, 2022-03-09:
I'm working on this, and as I've learned isn't possible to create self contained packages like in AppImage, but you just create a file with the build instructions, and the only way of providing 3rd-party dependencies is compiling it.
I've added the [flatpak build](https://github.com/webcamoid/webcamoid/blob/master/.github/workflows/flatpak.yml) to the workflow.

hipersayanX, 2022-03-11:
At the end I've discovered that its possible to create [distributable packages](https://docs.flatpak.org/en/latest/single-file-bundles.html), a flatpak package is available in the [daily build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build). The next step is adding it to flathub.

pizzadude, 2022-03-30:
Thank you for creating the flatpak @hipersayanX . The latest AppImage or portable releases don't work for me in Fedora KDE. The flatpak works.

hipersayanX, 2022-03-30:
And now its also available in [flathub](https://flathub.org/apps/details/io.github.webcamoid.Webcamoid).
@pizzadude, did it throw some error message? I did tested it in Fedora 35 until the last moment of the release and it was working fine for me.

pizzadude, 2022-03-31:
@hipersayanX This is the error the AppImage gives me. The flatpak works fine, but I think virtual web cam support won't work (not that I use that).

```
./webcamoid-portable-linux-9.0.0-x86_64.AppImage                                                                                         
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libaccess_alsa_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libaccess_concat_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libaccess_imem_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libaccess_mms_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libaccess_mtp_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libaccess_realrtsp_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libattachment_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libavio_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libcdda_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libdc1394_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libdtv_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libdv1394_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libdvb_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libdvdnav_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libdvdread_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libfilesystem_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libftp_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libhttp_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libhttps_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libidummy_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libimem_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/liblibbluray_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/liblinsys_hdsdi_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/liblinsys_sdi_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/liblive555_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libnfs_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libpulsesrc_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/librtp_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libsatip_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libsdp_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libsftp_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libshm_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libtcp_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libtimecode_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libudp_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libv4l2_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libvcd_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libvdr_plugin.so
[... cut ...]
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/codec/libddummy_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/codec/libdvbsub_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/codec/libedummy_plugin.so
[00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/codec/libfaad_plugin.so

```

hipersayanX, 2022-03-31:
@pizzadude

>@hipersayanX This is the error the AppImage gives me.
> 
> ```
> ./webcamoid-portable-linux-9.0.0-x86_64.AppImage                                                                                         
> [00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libaccess_alsa_plugin.so
> [00005587ea60bf10] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamu9yjUw/lib/vlc/plugins/access/libaccess_concat_plugin.so
>...
> ```
> 
> This is on Fedora 36 KDE.

Just that? Webcamoid should be working fine then. Those messages are because the VLC cache files were located at a different place in the build system, but it won't prevent Webcamoid for working fine.

>The flatpak works fine, but I think virtual web cam support won't work (not that I use that).

Yes, for the moment its like that. I was testing OBS, and in OBS its possible to request root permissions to setup the virtual camera and it seems to be working, but in the other side it returns:

```
[video4linux2,v4l2 @ 0x7f6370000c80] Not a video capture device.
/dev/video4: No such device
```

Not sure if this a problem with v4l2loopback, or because how flatpak works, if the first I can get get through it. The other problem will be running the virtual camera installer (as long as possible) or find an alternative.

pizzadude, 2022-04-01:
@hipersayanX Yes, but the application window doesn't show up after those messages, in the AppImage. I'm on wayland if that matters.

hipersayanX, 2022-04-02:
@pizzadude 

>Yes, but the application window doesn't show up after those messages,

That happens because there are some bundled library that is conflicting with a system one. I debug it taking libraries in and out, one by one or in a bunch, until it launch.

callegar, 2023-07-29:
Don't know if this is really the appropriate place for this, but the flathub flatpack seems to be outdated wrt the flatpack distributed on github.

hipersayanX, 2023-08-02:
@callegar I have already [sent the PR](https://github.com/flathub/io.github.webcamoid.Webcamoid/pull/1) but they won't accept it and I won't remove any feature :man_shrugging:.

callegar, 2023-08-31:
Added a comment to the [pr](https://github.com/flathub/io.github.webcamoid.Webcamoid) kindly suggesting, as a user, that they respond to your query. This was almost 1 month ago and I do not think it resulted in any response.

hipersayanX, 2023-08-31:
@callegar Considering that isn't even possible to compile Qt6 programs with Flatpak, it was a nice experiment, but I still prefer AppImage over those alternatives, Flatpak and Snap just gives me problems instead of solutions.

callegar, 2023-09-01:
Thought it was posssible, for instance `shotcut` uses the `KDE Application Platform` version 6.x (which should be QT6) rather than 5.15.x.  In fact, with QT6 apps, things are rather noisy because the QT6 releases go EOL quite quickly and immeditately flatpak starts complaining that some app relies on a runtime with no security support (e.g,, shotcut with QT6.4 that is eol).

Hoped that flatpack was better than snap which indeed gave me many problems until I gave up.

Is the flatpak on your releases on github which is up to date as good as your appimage or do you still suggest the appimage (or the native distro package, for rolling distros) over it?

hipersayanX, 2023-09-01:
@callegar 

>Thought it was posssible, for instance shotcut uses the KDE Application Platform version 6.x (which should be QT6) rather than 5.15.x.

ok my bad, I spoke too fast, I wasn't updated and thought that Qt6 wasn't available yet, I have updated the [workflow](https://github.com/webcamoid/webcamoid/actions/runs/6048370737).

>May I suggest pulling webcamoid completely from flathub given that only a version > 1 year behind can be found there and that you observe problems with that packaging anyways?

Not sure if this yet, it's still a good advertising and not my fault if updated.

callegar, 2023-09-01:
> Not sure if this yet, it's still a good advertising and not my fault if updated.

I totally understand the advertising point.

As a user, it is rather frustrating that not only the promise "Get the latest versions of all your favorite apps" is actually a delusion, but to find out that attempts like yours (i.e. your prompt pr) to follow up with that promise gets dismissed as questions are not even answered.

Lucky enough for me it is not a problem anymore, since I have discovered that you provide binaries yourself here! So, let me apologize for all the time that I have taken away from you. Thanks again for the great app.

hipersayanX, 2023-09-01:
>As a user, it is rather frustrating that not only Flatuhub's promise "Get the latest versions of all your favorite apps" is actually a delusion, but to find out that they do not even try to follow up to attempt like yours to actually ship the latest version.

From user standpoint, centralized repositories are convenient because (in theory) you can find all programs you need there, and no need to go to random dubious sites to download them. The problem is that the maintainers put a lot of bureaucracy for developers to reach there.
Flatpak/Flathub were supposed to make easy for developers to distribute their apps, but it's just another centralized repository with a lot of bureaucracy on top. And again, I won't remove any feature just to follow their rules, if they want to remove features is their own problem not mine, and won't waste my time for an answer.

callegar, 2023-09-02:
I can only say that I totally agree to your point of view. Thanks again for providing easy to use binaries yourself.

hipersayanX, 2024-06-28:
@callegar I have added the option to disable the virtual camera in Flathub, then point to the full binaries, its fair solution for both sides, I will release the changes for Webcamoid 9.2.0.