webcamoid/webcamoid #205: Videos play super fast or Webcamoid crashes when you switch to them

OscarSilve123, 2019-11-30:
I'm not sure why it is, but some videos play twice as fast in Webcamoid than they do in Media Player or VLC on Linux Mint.  Sometimes if you remove the audio stream from the MP4s they then play at normal speed, sometimes that does not help.  Other times, you can add a video but when you switch to that source while in play mode, the app just crashes then, and if you reopen the app you have to be fast enough to click the Stop button before it crashes again (or edit the config file to set playOnStart=false)

Not sure why every video player I try plays them without issue and Webcamoid can't.

hipersayanX, 2019-11-30:
Can you upload a video in which you have that problem? (preferably < 2 min duration)

OscarSilve123, 2019-12-01:
They are family videos I'd rather not upload.  Is there a log or something I can do and provide that output instead from when it plays really fast?

hipersayanX, 2019-12-01:
>They are family videos I'd rather not upload.

Ok, no problem, you don't need to upload such things. Anyway, if you find any public video doing that, upload or give me a link to it.

>Is there a log or something I can do and provide that output instead from when it plays really fast?

When you launch Webcamoid from a terminal, and you play a video on it, there will be appear a message like this:

```
Input #0, matroska,webm, from 'video.mkv':
  Metadata:
    encoder         : libebml v0.7.7 + libmatroska v0.8.1
    creation_time   : 2009-09-09T01:14:17.000000Z
  Duration: 00:01:00.15, start: 0.000000, bitrate: 11303 kb/s
    Stream #0:0(eng): Video: h264 (High), yuv420p(progressive), 1920x1080 [SAR 1:1 DAR 16:9], 23.98 fps, 23.98 tbr, 1k tbn, 47.95 tbc (default)
    Stream #0:1(eng): Audio: dts (DTS), 48000 Hz, 5.1(side), fltp, 1536 kb/s (default)
    Stream #0:2(eng): Subtitle: subrip (default)
```

and probably some errors will appear after it while playing the video, give me such log.

OscarSilve1233, 2019-12-07:
Can you tell me how exactly do play a video when launching it through the terminal?  If I launch it from a terminal then play a video through the GUI, there's no output like what you show.

hipersayanX, 2019-12-09:
No output at all? Do you have an screen like this in your Preferences?

[image]

Else just give me the output of:

```
ffmpeg -i 'video.mp4'
```

Webcamoid gives the same output as ffmpeg, anyway.

OscarSilve1233, 2019-12-13:
correct, nothing in the terminal window after i click play.  below is the ffmpeg info

Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'test.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
  Duration: 00:21:02.14, start: 0.000000, bitrate: 600 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 640x480 [SAR 1:1 DAR 4:3], 561 kb/s, 30 fps, 360 tbr, 90k tbn, 60 tbc (default)
    Metadata:
      handler_name    : VideoHandler
    Stream #0:1(und): Audio: aac (LC) (mp4a / 0x6134706D), 16000 Hz, mono, fltp, 34 kb/s (default)
    Metadata:
      handler_name    : VideoHandler

hipersayanX, 2019-12-14:
I don't see nothing wrong with the encoding. What about the **General Options** section?

OscarSilve1233-2, 2019-12-20:
In General Options I just have the defaults set with play webcam on start selected.

hipersayanX, 2019-12-20:
The same I have in the capture?

OscarSilve1233-2, 2019-12-21:
Yes except the last one of virtual camera driver.  I have v4l2loopback.  and i do have a virtual camera setup in webcamoid also

hipersayanX, 2019-12-21:
Ok, I think I have enough information with that, will check.

sijanec, 2020-04-16:
This issue also appears on latest version in Ubuntu LTS, when playing mp4 files.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX