webcamoid/webcamoid #206: Segfault and illegal instruction crashes in Webcamoid and AkVCamAssistant

onlynone, 2019-12-12:
## Summary

I noted several crashes and hangs of Webcamoid. Looking through crash reports, I saw several for AkVCamAssistant and one for Webcamoid. Most were segfaults, but two from AkVCamAssistant were illegal instructions.

Here's one for AkVCamAssistant due to a segfault:

```
Crashed Thread:        2  Dispatch queue: com.apple.root.default-qos.overcommit

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [70439]

VM Regions Near 0:
--> 
    __TEXT                 000000010fbdc000-000000010fbf4000 [   96K] r-x/rwx SM=COW  /Library/CoreMediaIO/*/AkVirtualCamera.plugin/Contents/Resources/AkVCamAssistant

Application Specific Information:
dyld2 mode

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff5f6b117a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff5f6b16d0 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff323270c2 __CFRunLoopServiceMachPort + 337
3   com.apple.CoreFoundation      	0x00007fff32326611 __CFRunLoopRun + 1654
4   com.apple.CoreFoundation      	0x00007fff32325d48 CFRunLoopRunSpecific + 463
5   com.apple.CoreFoundation      	0x00007fff32325b54 CFRunLoopRun + 40
6   AkVCamAssistant               	0x000000010fbdd5c5 0x10fbdc000 + 5573
7   libdyld.dylib                 	0x00007fff5f577ed9 start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff5f76a3f8 start_wqthread + 0
1   ???                           	0x0000000054485244 0 + 1414025796

Thread 2 Crashed:: Dispatch queue: com.apple.root.default-qos.overcommit
0   libxpc.dylib                  	0x00007fff5f7a116b xpc_get_type + 6
1   libxpc.dylib                  	0x00007fff5f7a90cb xpc_connection_send_message_with_reply_sync + 28
2   AkVCamAssistant               	0x000000010fbddaa6 0x10fbdc000 + 6822
3   libxpc.dylib                  	0x00007fff5f7aba28 _xpc_connection_call_event_handler + 56
4   libxpc.dylib                  	0x00007fff5f7a97d7 _xpc_connection_mach_event + 446
5   libdispatch.dylib             	0x00007fff5f529e6f _dispatch_client_callout4 + 9
6   libdispatch.dylib             	0x00007fff5f5422fe _dispatch_mach_cancel_invoke + 67
7   libdispatch.dylib             	0x00007fff5f53f876 _dispatch_mach_invoke + 1058
8   libdispatch.dylib             	0x00007fff5f539090 _dispatch_workloop_worker_thread + 603
9   libsystem_pthread.dylib       	0x00007fff5f76a60b _pthread_wqthread + 409
10  libsystem_pthread.dylib       	0x00007fff5f76a405 start_wqthread + 13

Thread 2 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000001  rbx: 0x00007f99af000cb0  rcx: 0x006567617373656d  rdx: 0x00007f99af000cb0
  rdi: 0x0000000000000000  rsi: 0x00007f99af000cb0  rbp: 0x000070000ba18850  rsp: 0x000070000ba18808
   r8: 0x006567617373656d   r9: 0x00007f99af000000  r10: 0x000000001fffffff  r11: 0x00007f989f40d384
  r12: 0x00007f99af000cb0  r13: 0x00007f99acf02d90  r14: 0x00007f99acf02dc8  r15: 0x0000000000000000
  rip: 0x00007fff5f7a116b  rfl: 0x0000000000010246  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x00000004
Trap Number:     14
```

And one due to an illegal instruction:

```
Crashed Thread:        2  Dispatch queue: com.apple.root.default-qos.overcommit

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
[... 109 lines cut ...]
```

For this last one, I had clicked the button to remove all webcam devices and then Webcamoid crashed. For the others, I think I just started up Webcamoid and it hung.

## Expected Behavior

No crashes, no hangs.

## Steps to Reproduce

I'm not able to precisely reproduce, but it seems like adding and removing output devices a bunch can cause the hangs. Once you do get it to hang, the following steps should produce the segfault in the main app:

1. drop to a shell and `kill -HUP` the still running Webcamoid process (it wouldn't end any other way)
2. restart the app
3. go into Preferences -> Output
4. click "Remove All"

## Your Environment

* Webcamoid information: 8.7.1 x86_64
* Operating System information: macos 10.14.3 Mojave x86_64

# Questinons and other matters?

Just wanted to say the app is really useful! I recently [added it to homebrew-cask](https://github.com/Homebrew/homebrew-cask/commit/fb807e4fff15ffc9294ef52f96f81a67e1adb1dc). One thing that might be helpful is if you look over the `zap` stanza and make sure I got everything there to clean up.

hipersayanX, 2019-12-12:
Will check.

> Just wanted to say the app is really useful! I recently [added it to homebrew-cask](https://github.com/Homebrew/homebrew-cask/commit/fb807e4fff15ffc9294ef52f96f81a67e1adb1dc).

Thanks! :smile:

ugurozturk, 2020-02-28:
I wasn't using it. Even i forgot about it, didn't use it for a long time. I dont really remember it if i even used it in my new computer.

[image]

zzk167, 2020-09-06:
Time Awake Since Boot: 5400 seconds

System Integrity Protection: enabled

Crashed Thread:        1  Dispatch queue: com.apple.root.default-qos.overcommit

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [14648]

VM Regions Near 0:
--> 
    __TEXT                 00000001046bc000-00000001046d4000 [   96K] r-x/rwx SM=COW  /Library/CoreMediaIO/*/AkVirtualCamera.plugin/Contents/Resources/AkVCamAssistant

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff67695dfa mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff67696170 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff2d4e9ef5 __CFRunLoopServiceMachPort + 247
3   com.apple.CoreFoundation      	0x00007fff2d4e89c2 __CFRunLoopRun + 1319
4   com.apple.CoreFoundation      	0x00007fff2d4e7e3e CFRunLoopRunSpecific + 462
5   com.apple.CoreFoundation      	0x00007fff2d570489 CFRunLoopRun + 40
6   AkVCamAssistant               	0x00000001046bd5c5 0x1046bc000 + 5573
7   libdyld.dylib                 	0x00007fff67554cc9 start + 1

Thread 1 Crashed:: Dispatch queue: com.apple.root.default-qos.overcommit
0   AkVCamAssistant               	0x00000001046bdb24 0x1046bc000 + 6948
1   libxpc.dylib                  	0x00007fff6779922c _xpc_connection_call_event_handler + 56
2   libxpc.dylib                  	0x00007fff6779806b _xpc_connection_mach_event + 726
3   libdispatch.dylib             	0x00007fff674fb6f8 _dispatch_client_callout4 + 9
4   libdispatch.dylib             	0x00007fff67513fc7 _dispatch_mach_cancel_invoke + 65
5   libdispatch.dylib             	0x00007fff6751195c _dispatch_mach_invoke + 1057
6   libdispatch.dylib             	0x00007fff6750ac09 _dispatch_workloop_worker_thread + 596
7   libsystem_pthread.dylib       	0x00007fff67755a3d _pthread_wqthread + 290
8   libsystem_pthread.dylib       	0x00007fff67754b77 start_wqthread + 15

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff67754b68 start_wqthread + 0

Thread 1 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x00007fff8dcb89a0  rcx: 0x0000000000000000  rdx: 0x00000000000f7cb0
  rdi: 0x0000000000000004  rsi: 0x00007ff4b5400000  rbp: 0x000070000a4598a0  rsp: 0x000070000a459870
   r8: 0x00000000000001da   r9: 0x0000000000000003  r10: 0x00007ff4b5400000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x00007ff4b5605a30  r14: 0x00007ff4b5406160  r15: 0x00007ff4b5605a50
  rip: 0x00000001046bdb24  rfl: 0x0000000000010246  cr2: 0x0000000000000000

zzk167, 2020-09-06:
when I call CMIOObjectAddPropertyListener(kCMIOObjectSystemObject, inAddress, inListener, inClientData) to listen device change event, a carsh alway happens.

const CMIOObjectPropertyAddress propertyAddress = {
            kCMIOHardwarePropertyDevices,
            kCMIOObjectPropertyScopeGlobal,
            kCMIOObjectPropertyElementMaster
    };

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX