webcamoid/webcamoid #208: Question: colour accuracy with YUV2 sources (Logitech BRIO 4K) and tweaking ffmpeg

gordonel, 2019-12-22:
Hi. I have Logitech Brio 4K and I'm using webcamoid to create a loopback on macOS from the YUV2 stream, which has better colour and dynamic range, but I noticed that webcamoid doesn't seem to be using the camera's full colour range. Moreover, there's significant banding in the shadows on the loopback device itself.

I'm not sure how to properly demonstrate this, but I was wondering if it is possible to tweak ffmpeg settings a little bit to see if it would make any difference. Is there a way to do that?

hipersayanX, 2019-12-26:
>there's significant banding in the shadows on the loopback device itself.

Does the image looks like the picture in the left side?

[image]

>I was wondering if it is possible to tweak ffmpeg settings a little bit to see if it would make any difference. Is there a way to do that?

Webcamoid uses the fast bilinear filter in FFmpeg by default, for speed and low CPU usage, it is [hardcoded](https://github.com/webcamoid/webcamoid/blob/20fefd7f656254e4a302a552b4530ed1abe387c0/libAvKys/Plugins/VideoCapture/src/ffmpeg/src/convertvideoffmpeg.cpp#L509), it can't be changed unless you change the source and recompile Webcamoid. But sure, I can add an option to modify the filter in Webcamoid.

gordonel, 2019-12-26:
> [maxresdefault](https://user-images.githubusercontent.com/1249885/71480195-b4eae900-27d6-11ea-9826-9b0ef131033a.jpg)

Yes, it looks like on the left.

Sadly, I wasn't able to compile from the source by following the wiki. Will try again later and file an issue if I find anything.
-------- Original Message --------

gordonel, 2020-01-19:
To bump this, it would be great to have the ability to change the filter somewhere in Webcamoid