webcamoid/webcamoid #210: macOS build doesn't launch. dyld: Library not loaded: libavkys_debug.8.dylib

gordonel, 2019-12-27:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debbuger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates bellow that fit better your issue.

# Reporting a problem?

## Summary

I tried following the raw build instructions and macOS specific ones, but no matter how I build, I get this error when I try to run it:
```
dyld: Library not loaded: libavkys_debug.8.dylib
  Referenced from: /Applications/webcamoid.app/Contents/MacOS/webcamoid
  Reason: image not found
Abort trap: 6
```

## Steps to Reproduce

1.  Follow the Raw build and Run instructions from the wiki
2.  Try running webcamoid

## Suggestions and tips

Dunno, but I strongly suggest making the wiki instructions clearer. I have no trouble building usually, but it took me 2 hours to set everything up

## Your Environment

* Webcamoid information: master at the time of writing, macOS
* Operating System information: macOS Catalina

hipersayanX, 2019-12-28:
You tried copying the app file to the _/Applications_ folder that's the reason of the error, you must run the deploy script to make the app usable.
Just run Webcamoid from Qt creator instead, it will be much easier for debugging, and once  you have it tested, then you run the deploy script to create the definitive app.

gordonel, 2019-12-28:
I can't run it from QT Creator either.

The debug version doesn't even show the ui. Here's the application out:
```bash
18:39:04: Starting /Users/dummy-user/webcamoid/StandAlone/debug/Qt5.14.0/clangpp/x86_64/bin/webcamoid --vcam /Users/dummy-user/webcamoid/libAvKys/Plugins/VirtualCamera/src/cmio/VirtualCamera -r -p /Users/dummy-user/webcamoid/libAvKys/Plugins -q /Users/dummy-user/webcamoid/libAvKys
...
QML debugging is enabled. Only use this in a safe environment.
Error loading plugin  "VideoCapture" : "Cannot load library /Users/dummy-user/webcamoid/libAvKys/Plugins/VideoCapture/release/Qt5.14.0/clangpp/x86_64/bin/libVideoCapture.dylib: (dlopen(/Users/dummy-user/webcamoid/libAvKys/Plugins/VideoCapture/release/Qt5.14.0/clangpp/x86_64/bin/libVideoCapture.dylib, 133): Library not loaded: libavkys.8.dylib\n  Referenced from: /Users/dummy-user/webcamoid/libAvKys/Plugins/VideoCapture/release/Qt5.14.0/clangpp/x86_64/bin/libVideoCapture.dylib\n  Reason: image not found)"
Error loading plugin  "VideoCapture" : "Cannot load library /Users/dummy-user/webcamoid/libAvKys/Plugins/VideoCapture/release/Qt5.14.0/clangpp/x86_64/bin/libVideoCapture.dylib: (dlopen(/Users/dummy-user/webcamoid/libAvKys/Plugins/VideoCapture/release/Qt5.14.0/clangpp/x86_64/bin/libVideoCapture.dylib, 133): Library not loaded: libavkys.8.dylib\n  Referenced from: /Users/dummy-user/webcamoid/libAvKys/Plugins/VideoCapture/release/Qt5.14.0/clangpp/x86_64/bin/libVideoCapture.dylib\n  Reason: image not found)"
Error loading plugin  "DesktopCapture" : "Cannot load library /Users/dummy-user/webcamoid/libAvKys/Plugins/DesktopCapture/release/Qt5.14.0/clangpp/x86_64/bin/libDesktopCapture.dylib: (dlopen(/Users/dummy-user/webcamoid/libAvKys/Plugins/DesktopCapture/release/Qt5.14.0/clangpp/x86_64/bin/libDesktopCapture.dylib, 133): Library not loaded: libavkys.8.dylib\n  Referenced from: /Users/dummy-user/webcamoid/libAvKys/Plugins/DesktopCapture/release/Qt5.14.0/clangpp/x86_64/bin/libDesktopCapture.dylib\n  Reason: image not found)"
Error loading plugin  "DesktopCapture" : "Cannot load library /Users/dummy-user/webcamoid/libAvKys/Plugins/DesktopCapture/release/Qt5.14.0/clangpp/x86_64/bin/libDesktopCapture.dylib: (dlopen(/Users/dummy-user/webcamoid/libAvKys/Plugins/DesktopCapture/release/Qt5.14.0/clangpp/x86_64/bin/libDesktopCapture.dylib, 133): Library not loaded: libavkys.8.dylib\n  Referenced from: /Users/dummy-user/webcamoid/libAvKys/Plugins/DesktopCapture/release/Qt5.14.0/clangpp/x86_64/bin/libDesktopCapture.dylib\n  Reason: image not found)"
Error loading plugin  "MultiSrc" : "Cannot load library /Users/dummy-user/webcamoid/libAvKys/Plugins/MultiSrc/release/Qt5.14.0/clangpp/x86_64/bin/libMultiSrc.dylib: (dlopen(/Users/dummy-user/webcamoid/libAvKys/Plugins/MultiSrc/release/Qt5.14.0/clangpp/x86_64/bin/libMultiSrc.dylib, 133): Library not loaded: libavkys.8.dylib\n  Referenced from: /Users/dummy-user/webcamoid/libAvKys/Plugins/MultiSrc/release/Qt5.14.0/clangpp/x86_64/bin/libMultiSrc.dylib\n  Reason: image not found)"
Error loading plugin  "MultiSrc" : "Cannot load library /Users/dummy-user/webcamoid/libAvKys/Plugins/MultiSrc/release/Qt5.14.0/clangpp/x86_64/bin/libMultiSrc.dylib: (dlopen(/Users/dummy-user/webcamoid/libAvKys/Plugins/MultiSrc/release/Qt5.14.0/clangpp/x86_64/bin/libMultiSrc.dylib, 133): Library not loaded: libavkys.8.dylib\n  Referenced from: /Users/dummy-user/webcamoid/libAvKys/Plugins/MultiSrc/release/Qt5.14.0/clangpp/x86_64/bin/libMultiSrc.dylib\n  Reason: image not found)"
Error loading plugin  "AudioDevice" : "Cannot load library /Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib: (dlopen(/Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib, 133): Library not loaded: libavkys.8.dylib\n  Referenced from: /Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib\n  Reason: image not found)"
Error loading plugin  "AudioDevice" : "Cannot load library /Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib: (dlopen(/Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib, 133): Library not loaded: libavkys.8.dylib\n  Referenced from: /Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib\n  Reason: image not found)"
Error loading plugin  "AudioDevice" : "Cannot load library /Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib: (dlopen(/Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib, 133): Library not loaded: libavkys.8.dylib\n  Referenced from: /Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib\n  Reason: image not found)"
Error loading plugin  "AudioDevice" : "Cannot load library /Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib: (dlopen(/Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib, 133): Library not loaded: libavkys.8.dylib\n  Referenced from: /Users/dummy-user/webcamoid/libAvKys/Plugins/AudioDevice/release/Qt5.14.0/clangpp/x86_64/bin/libAudioDevice.dylib\n  Reason: image [... cut ...]
    /Users/dummy-user/webcamoid/ports/deploy/temp_priv/Applications/webcamoid.app/Contents/Resources/qml/QtQuick/Controls/Styles/Desktop/ScrollViewStyle.qml
    /Users/dummy-user/webcamoid/ports/deploy/temp_priv/Applications/webcamoid.app/Contents/Resources/qml/QtQuick/Controls/Styles/Desktop/ScrollViewStyle.qmlc
    /Users/dummy-user/webcamoid/ports/deploy/temp_priv/Applications/webcamoid.app/Contents/Resources/qml/QtQuick/Controls/Styles/Desktop/SliderStyle.qml
    /Users/dummy-user/webcamoid/ports/deploy/temp_priv/Applications/webcamoid.app/Contents/Resources/qml/QtQuick/

hipersayanX, 2019-12-29:
This could be a bug with Qt 5.14, not sure, Webcamoid [compiled fine](https://travis-ci.org/webcamoid/webcamoid/jobs/623396338) before that Qt release. Will check.

gordonel, 2019-12-29:
I couldn't do it on 5.13 either

gordonel, 2020-01-01:
Any chance of fixing it for Qt 5.14 and Catalina?

hipersayanX, 2020-01-01:
Of course, it will be fixed before the next release for sure.

gordonel, 2020-01-07:
@hipersayanX what's the last known working version of Qt I could use for building?

hipersayanX, 2020-01-07:
If you are planning to install Qt from the official installer, that isn't a good idea, all the plugins that Webcamoid needs to work will not build, like for example all plugins that depends on FFmpeg. Just use the Qt package that comes with homebrew.
The other option you have is checking out  your local  repository to 8.7.1 tag and work from there, I can't guarantee it will work with Qt 5.14, but you can try.
Right now, the master branch is a bit shaky since I'm redesigning the UI, and can't pay attention to other platforms until I finish with that, it will take a while.

gordonel, 2020-01-19:
I noticed that you pushed a commit that has something to do with deps (haven't looked into it yet). Is it related to this problem or not?

hipersayanX, 2020-01-21:
If you are talking about the Qml fix, it affects all platforms in general, but I did not tested in Mac yet, I'm only testing things in GNU/Linux and Android, didn't even tested it in Windows.

hipersayanX, 2020-02-04:
I've checked and there is a bug with GStreamer that crash Webcamoid when launching. First at all clean and update the repository with:

```
git clean -xfd
git reset --hard
git pull
```

Then in _Projects > Build Steps > Additional arguments_ add **NOGSTREAMER=1**, and finally build and run.

dg1sek, 2020-03-03:
I have exactly the same issue and error message when running from within Qt Creator. 

I noted that I only got the error when I tried to launch with the command line parameters to run the application. (Actually I did my first compile&run without indicating command line arguments, that's how I noticed by coincidence).

( By command line arguments I mean the adapted "--vcam /[path]/webcamoid/libAvKys/Plugins/VirtualCamera/src/cmio/VirtualCamera -r -p /[path]/webcamoid/libAvKys/Plugins -q /[path]/webcamoid/libAvKys")

When I run it without command line arguments it actually starts (but then I cannot access to my webcam). Not sure if that observation helps in debugging?

gordonel, 2020-03-08:
@dg1sek Yes. This is an accurate behaviour with this bug. Try disabling gstreamer as shown above. @hipersayanX do you know if the latest (1.16.2) version of gstreamer has this bug? If so, could you please link it for reference

gordonel, 2020-03-08:
To clarify, the current master launches regardless of whether gstreamer is enabled or not, but I get no picture whatsoever and the interface looks weird. I get this in the console:
```bash
$ /Volumes/webcamoid-portable-8.7.1/webcamoid.app/Contents/MacOS/webcamoid 
qrc:/Webcamoid/share/qml/main.qml:523:13: QML ToolBar: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/main.qml:92: ReferenceError: togglePlay is not defined
qrc:/Webcamoid/share/qml/MediaBar.qml:63:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
```

hipersayanX, 2020-03-08:
@gordonel the same code for GStreamer works fine in GNU/Linux, but crash in Mac (I don't remember if it worked in Widows), not sure if that bug is solvable in my side or it is an internal bug in GStreamer.

>NOGSTREAMER=1 had no effect for me

Try cleaning the source directory with git ```clean -xfd``` or if you enabled the shadow build, delete the shadow build folder. When compiling it must appear this table at the very beginning:

```
Checking for alsa... no
Checking for avfoundation... yes
Checking for cmio... yes
Checking for coreaudio... yes
Checking for dshow... no
Checking for ffmpeg... yes
Checking for ffmpeg_avcodec_sendrecv... yes
Checking for ffmpeg_avcodec_subtitledata... yes
Checking for ffmpeg_avformat_codecpar... yes
Checking for ffmpeg_avutil_extraoptions... yes
Checking for ffmpeg_avutil_sampleformat64... yes
Checking for ffmpeg_avresample... yes
Checking for ffmpeg_swresample... yes
Checking for jack... yes
Checking for libuvc... yes
Checking for libusb_has_hotplug... yes
Checking for mediafoundation... no
Checking for ndk_audio... no
Checking for ndk_camera... no
Checking for ndk_media... no
Checking for opensl... no
Checking for pulseaudio... yes
Checking for v4l2... no
Checking for wasapi... no
```

When you use **NOGSTREAMER=1** the **Checking for gstreamer... yes** my not appear there.

>but I get no picture whatsoever and the interface looks weird. I get this in the console:

I'm in the middle of [UI redesign](https://github.com/webcamoid/webcamoid/issues/216) right now, those messages will be common until I finish with it.

gordonel, 2020-03-08:
Do you know if gstreamer has a filed issue on its bug tracker regarding this?

hipersayanX, 2020-03-08:
I did not investigated it yet, GStreamer is a secondary option anyway and the preferred one is FFmpeg.

gordonel, 2020-03-08:
Must it say `Checking for ffmpeg... yes`? Because I'm getting "no" for this and some other deps despite them being installed

gordonel, 2020-03-08:
Outputs:
```
Checking for alsa... no
Checking for avfoundation... yes
Checking for cmio... yes
Checking for coreaudio... yes
Checking for dshow... no
Checking for ffmpeg... no
Checking for jack... no
Checking for libuvc... no
Checking for mediafoundation... no
Checking for ndk_audio... no
Checking for ndk_camera... no
Checking for ndk_media... no
Checking for opensl... no
Checking for pulseaudio... no
Checking for v4l2... no
Checking for wasapi... no```
```bash
$ brew list
aom			leptonica		opus
autoconf		libass			orc
automake		libbluray		p11-kit
berkeley-db		libevent		p7zip
bison			libffi			pango
cairo			libidn2			pcre
ccache			libogg			pixman
ffmpeg			libpng			pkg-config
flac			libsamplerate		pulseaudio
fontconfig		libsndfile		python
freetype		libsoxr			qt
frei0r			libtasn1		readline
fribidi			libtiff			rtmpdump
gdbm			libtool			rubberband
gettext			libunistring		sdl2
giflib			libusb			snappy
glib			libuvc			speex
gmp			libvidstab		speexdsp
gnutls			libvorbis		sqlite
gobject-introspection	libvpx			tesseract
gradle			little-cms2		texi2html
graphite2		lzo			theora
gst-plugins-base	meson			unbound
gstreamer		nasm			watchman
harfbuzz		nettle			webp
icu4c			ninja			x264
jack			opencore-amr		x265
jpeg			openjdk			xvid
json-c			openjpeg		xz
lame			openssl@1.1```

hipersayanX, 2020-03-08:
FFmpeg must say "yes", Gstreamer "no" (or not appear), and everything else depends on your system, for example there is no v4l2 in Mac so it will say "no", because it's a GNU/Linux and BSD only feature.

hipersayanX, 2020-03-08:
Check step 14 in the [wiki](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Mac). You must add _/usr/local/bin_ to _PATH_ variable, that way you will be able to use **pkg-config**
.

gordonel, 2020-03-08:
It **is** added. However, I have homebrew installed to ~/homebrew, so the bin is ~/homebrew/bin

hipersayanX, 2020-03-08:
What is the result of this ```ls /usr/local/bin/pkg-config```?

gordonel, 2020-03-08:
```bash
ls: /usr/local/bin/pkg-config: No such file or directory
```

gordonel, 2020-03-08:
On the other hand:
```bash
$ ls ~/homebrew/bin/pkg-config
/Users/gordonel/homebrew/bin/pkg-config
```

hipersayanX, 2020-03-08:
And did you put _~/homebrew/bin_ as is in the PATH? try with _/Users/gordonel/homebrew/bin_.

gordonel, 2020-03-08:
I put in the absolute path. Also, `$ sudo ln -s ~/homebrew/bin/pkg-config /usr/local/bin` didn't help

hipersayanX, 2020-03-08:
Anyways hombrew must have created the _/usr/local/bin_ folder when installing, if you have that folder then try linking pkg-config with ```brew link pkg-config```.

gordonel, 2020-03-08:
Tried
```bash
brew link pkg-config
Warning: Already linked: /Users/gordonel/homebrew/Cellar/pkg-config/0.29.2
To relink:
  brew unlink pkg-config && brew link pkg-config
```

hipersayanX, 2020-03-08:
>I put in the absolute path. Also, $ sudo ln -s ~/homebrew/bin/pkg-config /usr/local/bin didn't help

Undo that and do as I say before.

gordonel, 2020-03-08:
Did that, no effect. Homebrew never touched /usr/local/bin on my machine as far as I know

hipersayanX, 2020-03-08:
Let make the check list again:

Do you have the file _/Users/gordonel/homebrew/bin/pkg-config_?
Did you put _/Users/gordonel/homebrew/bin:/other/paths:/here_ in PATH? (pay attention to the colons **:** separators please)
Did you cleaned the sources directory and the shadow build, and rebuilded?

gordonel, 2020-03-08:
1. Yes
2. Yes. It reads: `/Users/gordonel/homebrew/Cellar/qt/5.14.1/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Users/gordonel/homebrew/bin`
3. Shadow build is disabled according to wiki. Yes to the other two

hipersayanX, 2020-03-08:
You followed all the steps and ffmpeg is not enabled still? mmm... there is something wrong with you system, but not sure what is :confused:

gordonel, 2020-03-08:
How can we debug this "checking for" thing? Can we have it log the paths it's checking and how it does that?

\-------- Original Message --------

[view it on GitHub]: https://github.com/webcamoid/webcamoid/issues/210?email_source=notifications&email_token=AKYUTQVMR6JMS7GO32LTCWLRGPMHBA5CNFSM4KABXY52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEOE4VKY#issuecomment-596232875
[unsubscribe]: https://github.com/
[AKYUTQR5RIELVDCHJUNMW23RGPMHBA5CNFSM4KABXY52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEOE4VKY.gif]: https://github.com/notifications/beacon/AKYUTQR5RIELVDCHJUNMW23RGPMHBA5CNFSM4KABXY52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEOE4VKY.gif

hipersayanX, 2020-03-08:
The compile log is in _webcamoid/libAvKys/config.log_. The test result for FFmpeg must appear under the section **executing config test ffmpeg**.

gordonel, 2020-03-08:
Does this tell you anything?
```bash
executing config test ffmpeg
+ cd /Users/gordonel/webcamoid/libAvKys/Tests/ffmpeg && /Users/gordonel/homebrew/Cellar/qt/5.14.1/bin/qmake  -spec /Users/gordonel/homebrew/Cellar/qt/5.14.1/mkspecs/macx-clang "CONFIG -= qt debug_and_release app_bundle lib_bundle" /Users/gordonel/webcamoid/libAvKys/Tests/ffmpeg
> Project ERROR: libavformat development package not found
test ffmpeg FAILED
executing config test gstreamer
+ cd /Users/gordonel/webcamoid/libAvKys/Tests/gstreamer && /Users/gordonel/homebrew/Cellar/qt/5.14.1/bin/qmake  -spec /Users/gordonel/homebrew/Cellar/qt/5.14.1/mkspecs/macx-clang "CONFIG -= qt debug_and_release app_bundle lib_bundle" /Users/gordonel/webcamoid/libAvKys/Tests/gstreamer
> Project ERROR: gstreamer-1.0 development package not found
test gstreamer FAILED
executing config test jack
+ cd /Users/gordonel/webcamoid/libAvKys/Tests/jack && /Users/gordonel/homebrew/Cellar/qt/5.14.1/bin/qmake  -spec /Users/gordonel/homebrew/Cellar/qt/5.14.1/mkspecs/macx-clang "CONFIG -= qt debug_and_release app_bundle lib_bundle" /Users/gordonel/webcamoid/libAvKys/Tests/jack
> Project ERROR: jack development package not found
test jack FAILED
```

hipersayanX, 2020-03-08:
That message come from pkg-config, so pkg-config is detected, but isn't locating the *.pc files for ffmpeg. What if you do this?:

```
find /Users/gordonel/homebrew/Cellar/ffmpeg -name '*.pc'
```

gordonel, 2020-03-08:
```bash
$ find /Users/gordonel/homebrew/Cellar/ffmpeg -name '*.pc'
/Users/gordonel/homebrew/Cellar/ffmpeg/4.2.2_2/lib/pkgconfig/libswscale.pc
/Users/gordonel/homebrew/Cellar/ffmpeg/4.2.2_2/lib/pkgconfig/libswresample.pc
/Users/gordonel/homebrew/Cellar/ffmpeg/4.2.2_2/lib/pkgconfig/libavdevice.pc
/Users/gordonel/homebrew/Cellar/ffmpeg/4.2.2_2/lib/pkgconfig/libavcodec.pc
/Users/gordonel/homebrew/Cellar/ffmpeg/4.2.2_2/lib/pkgconfig/libavresample.pc
/Users/gordonel/homebrew/Cellar/ffmpeg/4.2.2_2/lib/pkgconfig/libavformat.pc
/Users/gordonel/homebrew/Cellar/ffmpeg/4.2.2_2/lib/pkgconfig/libavutil.pc
/Users/gordonel/homebrew/Cellar/ffmpeg/4.2.2_2/lib/pkgconfig/libavfilter.pc
/Users/gordonel/homebrew/Cellar/ffmpeg/4.2.2_2/lib/pkgconfig/libpostproc.pc
```

hipersayanX, 2020-03-08:
```
pkg-config --list-all | sort | awk '{print $1}' | grep '^libav'
```

?

gordonel, 2020-03-08:
Nada. `pkg-config --list-all` returns
```bash
$ pkg-config --list-all
libecpg_compat libecpg_compat - PostgreSQL libecpg_compat library
osxfuse        fuse - OSXFUSE
fuse           fuse - OSXFUSE
apr-1          APR - The Apache Portable Runtime library
apr-util-1     APR Utils - Companion library for APR
libpq          libpq - PostgreSQL libpq library
libpcre        libpcre - PCRE - Perl compatible regular expressions C library with 8 bit character support
libpgtypes     libpgtypes - PostgreSQL libpgtypes library
libecpg        libecpg - PostgreSQL libecpg library
libpcreposix   libpcreposix - PCREPosix - Posix compatible interface to libpcre
libiodbc       iODBC - iODBC Driver Manager
```

hipersayanX, 2020-03-08:
But [here](https://github.com/webcamoid/webcamoid/issues/210#issuecomment-596229171) say ffmpeg is installed, but pkg-config [doesn't recognize](https://github.com/webcamoid/webcamoid/issues/210#issuecomment-596235647) it. Then again there is something wrong with you install, [here](https://travis-ci.org/webcamoid/webcamoid/jobs/659658256#L3584) is a build log from a few hour later, and it's working fine.
In the worst case you will have to set the [PKG_CONFIG_PATH](https://linux.die.net/man/1/pkg-config) variable manually to add FFmpeg *.pc files path.

hipersayanX, 2020-03-08:
@gordonel I've thinking, maybe it did not added ffmpeg to the path because you have installed pkg-config after after ffmpeg? maybe reinstalling ffmpeg may solve the problem?

gordonel, 2020-03-08:
@hipersayanX thanks a million. fixed by setting PKG_CONFIG_PATH to ~/homebrew/lib/pkgconfig

hipersayanX, 2020-03-08:
Finally :smile:

gordonel, 2020-03-08:
Can you make it so if tests fail, they get logged as warnings or errors?

gordonel, 2020-03-08:
Well, despite all this, I still don't see anything, no matter which device I select. I do get a fair share of warnings when compiling.

Also nothing works when I try to build off of 8.7.1 tag, but I get more stuff in the console.

gordonel, 2020-03-08:
I take it back. Everything works when I set video convert to ffmpeg

hipersayanX, 2020-03-09:
>Can you make it so if tests fail, they get logged as warnings or errors?

Is not necessary,  qmake already has options for [verbose debugging](https://doc.qt.io/qt-5/qmake-running.html#general-options) (-d option), and anyway there is the log file where you can read those errors.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX