webcamoid/webcamoid #211: Make UI accessible to screen readers or add CLI options

devinprater, 2019-12-27:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debbuger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates bellow that fit better your issue.

# Reporting a problem?

## Summary

Webcamoid is not usable with the VoiceOver screen reader on macOS 10.15.

## Current Behavior

I've been looking for a way to use <http://www.seeingwithsound.com/webvoice.htm> not only with an actual camera on my computer, but with applications, to visualize them better. I tried installing Webcamoid with the installer found on the site, but VoiceOver could not see anything on the installer window. I used Home (cask) to install Webcamoid, but the main screen of the program is still not accessible to VoiceOver. I tried using the command line, but didn't get far. It said

```
Webcam capture application.

Options:
  -h, --help                              Displays this help.
  -v, --version                           Displays version information.
  -c, --config <PATH>                     Load settings from PATH. If PATH is
                                          empty, load configs from application
                                          directory.
  -q, --qmlpaths <PATH1;PATH2;PATH3;...>  Semi-colon separated list of paths to
                                          search the Qml interface.
  -r, --recursive                         Search in the specified plugins paths
                                          recursively.
  -p, --paths <PATH1;PATH2;PATH3;...>     Semi-colon separated list of paths to
                                          search for plugins.
  -b, --no-load <PATH1;PATH2;PATH3;...>   Semi-colon separated list of paths to
                                          avoid loading.
  --vcam <PATH1;PATH2;PATH3;...>          Semi-colon separated list of paths to
                                          search for virtual camera driver
```

but if I type ./webcamoid Safari, for example, it just goes back to th main screen, and no virtual camera is created.

## Expected Behavior

Webcamoid should either have been accessible to [Voiceover](https://developer.apple.com/accessibility/), or there should be a way to easily control it through the command line.

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Press Command+F5 on older Macs, or hold Command and triple press the Touch ID button on Macs that have it. This is a toggle for VoiceOver.
2.  Open Webcamoid.
3.  Press Control+Option+Right arrow to navigate an item to the right.
4. Nothing happens.

## Suggestions and tips

If the GUI of Webcamoid is made using an inaccessible framework, like GTK (Mac), or QT4, there may not be much you could do with the GUI, but the command line in accessible to all screen readers on any platform where it is readily available, so good command-line options and explanations on how to use them would solve it as well.

## Your Environment

* Webcamoid information: (Webcamoid 8.7.1 64 bit)
* Operating System information: (macOS 10.15.2)(64-bit)
* MacBook Pro 2019)

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questinons and other matters?

Go ahead!

hipersayanX, 2019-12-28:
>If the GUI of Webcamoid is made using an inaccessible framework, like GTK (Mac), or QT4, there may not be much you could do with the GUI

Actually, according to [Qt documentation](https://doc.qt.io/qt-5/qml-qtquick-accessible.html) it would be possible to enable accessibility in QML. Will check.

hipersayanX, 2023-04-27:
Forgot to close, this is already done.

closed 2023-04-27 by hipersayanX