webcamoid/webcamoid #213: Can I add a Can I add a surveillance camera to my webcamoid? I am using Ubunti 18.04.3 LTS.

Krassegrand, 2020-01-07:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debbuger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates bellow that fit better your issue.

# Reporting a problem?

## Summary

Write here a brief description of the problem.

## Current Behavior

Describe the problem the best as you can, don't omit information.

## Expected Behavior

How it should have work?

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Open this
2.  Click that
3.  Drag those
4.  ...

## Suggestions and tips

How would you solve the problem?

## Your Environment

* Webcamoid information: (version) (architecture)
* Operating System information: (name) (numeric version and codename if applied) (architecture)
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questinons and other matters?

Go ahead!

hipersayanX, 2020-01-07:
Just add the URL of the stream, it may work. 

[image]

**The URL in the example is illustrative.**

Krassegrand, 2020-01-08:
Did I do the Url right? I wrote this
Http://localhost:192.168.1.135:88/video.mp4            and 
Http://localhost:192.168.1.135.88                              and
Http://192.168.1.135:88                                            

I think I need a lot of help.

hipersayanX, 2020-01-08:
Read your security camera manual.
Also [here](https://public-camera.com/en/how-to-connect-ipcamera.html) is a URL generator for different models, and [here](https://www.insecam.org/) you can find public IP cameras you can test with.

Krassegrand, 2020-01-08:
Well, the manual is for Windows and VMS.
The other yips I will try later.
Thank You!

closed 2020-02-27 by hipersayanX