webcamoid/webcamoid #215: Feature suggestion: support 3D lookup tables (3D LUTs)

gordonel, 2020-01-13:
3D LUTs are used in video production to map one colour space to another. They can also be used to correct the colour space by using special LUTs made with the use of colour targets:
https://en.wikipedia.org/wiki/3D_lookup_table

hipersayanX, 2020-01-13:
Can you give me an example on how will you use that? for example in an effect or something?

gordonel, 2020-01-13:
I could feed log footage to the computer and use a LUT to transform the colour space to normal
-------- Original Message --------

hipersayanX, 2020-01-13:
Oh, get it.

https://www.youtube.com/watch?v=iilou58SyZY
https://www.rocketstock.com/blog/tips-for-log-color-space-compositing/
https://www.bhphotovideo.com/explora/video/tips-and-solutions/understanding-log-format-recording