webcamoid/webcamoid #216: UI redesign

hipersayanX, 2020-01-16:


gordonel, 2020-01-19:
Looks great, but I can't enjoy it since I can't compile webcamoid on my mac properly. You know why :)

hipersayanX, 2020-02-21:
This is how Webcamoid looks now.

[image]

hipersayanX, 2020-02-21:
Ok, with the style more or less defined is time to make the big real change. Let start with the main screen.

# Main screen - Photo capture

[image]

This is the very first screen you'll see when opening Webcamoid. Webcamoid will be ready to take a picture.  
In in the top left corner there will be an icon to apply video effects to the camera, desktop, video, etc.. Bellow it, an icon for enabling/disabling the camera flash (either software or hardware), and the option to set the timeout before taking the photo.  
In the top right corner is the settings icon that offers even more options, I'll talk about it later.  
In the bottom, you have access to the most important options for the current view.  
In the middle you've a big button to take a picture, in the left you can click/touch and have direct access to the last picture taken, in the right you have the recording icon that you can use to switch to the recording mode.  
And in the bottom you have a progress bar indicating the photo timeout.

# Main screen - Video recording

[image]

When you touch the recording icon, the last photo preview will disappear, the take photo icon will be displaced to the left, the record icon to the middle, and the last video preview will appear in the right. The photo related controls disappears.  
When recording, a floating control will appear showing the current recording time.  

This design may works well in both, small and big screens.

hipersayanX, 2020-02-21:
# Video effects

[image]

When you touch the effect icon, a drawer in the left side will appear showing all applied effects (clean by default). If you want to add a new effect you touch the homonym option. Each effect is listed as a swipe control, this is a control type that when you drag it you can discover new options, in this case options for configuring and removing the effect.

# Video effect selection

[image]

When you press "Add effect", a dialog showing all available effects and a preview will appear. You can use the top left TextField to search the specific effect you want to use. You will also have the option to overlap the effect (apply a new effect over an applied one), if you don't want to overlap the effects, the new selected effect will replace the current one (default action). The overlap option is the equivalent of the old "Advanced effect mode" but more intuitive and easier to use.

# Video effect options

[image]

When you press "Configure", you will able to configure the effect, just like before.

hipersayanX, 2020-02-21:
# Settings menu

[image]

The settings menu has the most important options ready to use. You can also play and stop the playback from there.

hipersayanX, 2020-02-21:
# Sources bar

[image]

If you select the video options, this drawer will appears, where you can use to set and configure the video sources and outputs, I'll talk about video outputs later.  
It will list all available webcams and the desktop by default, and you can add new sources like a video file or a GIF if you want.  

# Adding files

[image]

You can add a file just like before.

# Adding multimedia streams

[image]

In the old _add file dialog_, it was already possible to add an URL but it was counter intuitive for many users, so this time _file and URL add_ is split between two tabs.

# Sources settings

[image]

And this is the setting menu for the source selected, similar to the old one.

hipersayanX, 2020-02-21:
# Video outputs

[image]

Video outputs is the equivalent of the virtual camera but this time you have direct access to it. This will be used for virtual cameras but in the future is intended to be used with other kind of outputs like for example video streaming.

# Adding a new output

[image]

You will be able to define which formats will be offered by your virtual camera.

# Advanced options

[image]

And you will be also able to control how the stream will be delivered to the client.

# Format dialog

[image]

This dialog will appear when you want to define a new format.

hipersayanX, 2020-02-21:
# Audio inputs

[image]

Audio settings menu brings you here, which is not much different from the old one.

# Audio outputs

[image]

Here is the audio outputs menu

# Audio settings

[image]

And here you have the audio settings.   

There are not big changes in this section.

hipersayanX, 2020-02-21:
# Image Capture

[image]

It will be possible to configure pictures saving options.

# Video Recording

[image]

Video saving options too.

# Video Recording - Codec options

[image]

All options where they must be.

# General Options

[image]

Same as before.

# Plugins - Paths

[image]

One tab for plugins paths

# Plugins - Plugins

[image]

and other for managing them.

# Updates

[image]

Same as before.

# About

[image]

Same as before.

# Contributors

[image]

# License

[image]

Same as before.

# 3rd Party Licenses

[image]

Same as before.

hipersayanX, 2020-02-21:
And that is my current plan for the UI redesign.

hipersayanX, 2020-03-04:
Current progress:

[image]

Effect button in the top left corner is dummy.
Video recording button in the bottom right corner is dummy.
**Video**, **Audio** and **Preferences** options in the setting menu are dummy. 
**Play sources** replaces the **Play/Stop** option in the toolbar.
Photo shoot is fully working. 

[image]

hipersayanX, 2020-03-08:
More updates on the **Preferences** menu:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

**Image Capture** and **Video Recording** are dummys.

hipersayanX, 2020-03-25:
Image capture and video recording menus finished.

[image]
[image]
[image]
[image]
[image]

hipersayanX, 2020-04-24:
Effects menus ready

[image]
[image]
[image]

hipersayanX, 2020-04-30:
Audio settings ready.

[image]
[image]

hipersayanX, 2020-05-05:
Video sources ready.

[image]
[image]

hipersayanX, 2020-05-23:
Virtual camera working, with some bugs that must be fixed yet.

[image]

Virtual camera formats are now customizable .

[image]
[image]
[image]

You can also control the behavior in the client side.

[image]

Also, have removed the tray icon for announcing new version updates and have replaced by a dialog.

[image]

Slion, 2021-04-14:
Can't find that place in the settings where you can define virtual cam output resolution. I must be getting old. Running 8.8.0 on Windows.

hipersayanX, 2021-04-16:
@Slion these changes are for Webcamoid 9 (currently daily build), Webcamoid 8 has not such options.

hipersayanX, 2021-09-07:
And finally this is the last feature I want to add to Webcamoid 9. Webcamoid will advice to the user if the virtual camera is not installed or is outdated, and offer to download and install it. Alternatively it will inform if the virtual camera is not supported in your operating system.

[image]

These will be more or less the download dialogs:

[image]

Once I finish with this feature, I'll start reviewing all pending issues and fixing bugs, and make the final release.

hipersayanX, 2021-09-10:
And that's all,  every single feature I wanted for Webcamoid 9 is done, now is time to polish the code and make the final release.

[image]
[image]

shawnk0723, 2021-12-10:
Hi, I'm currently testing latest daily build (webcamoid-installer-linux-daily-master-1-x86_64.run) on arch linux. It says one of the updated feature from new UI design is "direct access to the last picture taken..." I do see the last photo preview icon but it's not clickable. Any idea what might be the issue? thank you for the great app!! :)

shawnk0723, 2021-12-10:
> @cjbarth will check, but for me is ok as is right now.

Hi, I'm currently testing latest daily build (webcamoid-installer-linux-daily-master-1-x86_64.run) on arch linux. It says one of the updated feature from new UI design is "direct access to the last picture taken..." I do see the last photo preview icon but it's not clickable. Any idea what might be the issue? thank you for the great app!! :)

hipersayanX, 2021-12-11:
@shawnk0723 The preview is in the bottom left corner

[image]

>I do see the last photo preview icon but it's not clickable. Any idea what might be the issue? 

I did checked and it's working fine for me, try downloading the daily build again, it should open the picture in your default image viewer.

shawnk0723, 2021-12-12:
> @shawnk0723 The preview is in the bottom left corner
> 
> [image]
> 
> > I do see the last photo preview icon but it's not clickable. Any idea what might be the issue?
> 
> I did checked and it's working fine for me, try downloading the daily build again, it should open the picture in your default image viewer.

Thank you for the quick reply. I'll try to download again and let you know. :)

shawnk0723, 2021-12-12:
> @shawnk0723 The preview is in the bottom left corner
> 
> [image]
> 
> > I do see the last photo preview icon but it's not clickable. Any idea what might be the issue?
> 
> I did checked and it's working fine for me, try downloading the daily build again, it should open the picture in your default image viewer.

I just reinstalled fresh downloaded daily build. It still doesn't work even after cleaning all previous cached/config files. I remember it used to work on my system. I think it stopped working after trying latest stable version (8.8.0) and went back to daily build. I also have cheese webcam app. it has similar feature and it opens the photo when preview icon is clicked without any issues. Not sure why it stopped working... I really like the UI of webcamoid but this is the biggest showstopper for me now. :(

shawnk0723, 2021-12-12:
I wonder if it's just linux specific bug since it works fine on your machine.

hipersayanX, 2021-12-12:
@shawnk0723 

>I think it stopped working after trying latest stable version (8.8.0) and went back to daily build.

That feature is not affected by any configuration, so switching versions won't make any change. That feature only depends on which image viewer you have configured by default on your system.

>I wonder if it's just linux specific bug since it works fine on your machine.

Again, I'm using Arch Linux and it's working fine for me, so it's unlikely that this is a GNU/Linux general bug. Which distribution are you using?

shawnk0723, 2021-12-12:
> @shawnk0723
> 
> > I think it stopped working after trying latest stable version (8.8.0) and went back to daily build.
> 
> That feature is not affected by any configuration, so switching versions won't make any change. That feature only depends on which image viewer you have configured by default on your system.
> 
> > I wonder if it's just linux specific bug since it works fine on your machine.
> 
> Again, I'm using Arch Linux and it's working fine for me, so it's unlikely that this is a GNU/Linux general bug. Which distribution are you using? also, which desktop environment? maybe Wayland or Xorg?

That is super weird! FYI, I'm currently using garuda, kde, x11. All I know is it used to work just fine up until I switched back & forth from 8.8.0 to daily build. My default image viewer is  LXimage. I changed it to gthumb just to see if it might fix the issue but no luck with this either. I also tried to troubleshoot by executing webcamoid.sh from terminal. This is the error i got: 
**********************************************************************************
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
/usr/bin/xdg-open: line 613: 162175 Aborted                 (core dumped) kde-open${KDE_SESSION_VERSION} "$1"
**********************************************************************************

Then I found a solution from someone who previously had similar issue on google. His was fixed by installing qtcreator but unfortunately, it didn't work on my machine. yeah so, I'm still stuck with this same issue. Any suggestion would be greatly appreciated. Thanks man!

hipersayanX, 2021-12-12:
>I also tried to troubleshoot by executing webcamoid.sh from terminal. This is the error i got:

But you did executed Webcamoid on first place and it worked, then how did you executed it on first place? did you downloaded the AppImage, the installer, or the portable version?

shawnk0723, 2021-12-13:
> @shawnk0723
> 
> > I also tried to troubleshoot by executing webcamoid.sh from terminal. This is the error i got:
> 
> ~But you did executed Webcamoid on first place and it worked, then how did you executed it on first place? did you downloaded the AppImage, the installer, or the portable version?~
> 
> **EDIT**
> 
> I've downloaded and tested the daily build, and I'm experiencing the same exact bug, it could be a packaging problem. @shawnk0723 can you please open a bug for that?

Oh, so it's happening on your machine too? yeah, i don't think it's anything big. might be packaging issue like you said bc it was  working just fine. I just opened a bug for this. Please let me know once this bug is fixed. I'll test it out for you soon as it's fixed. 
Thanks a lot for the awesome app once again hipersayanX!

hipersayanX, 2021-12-13:
@shawnk0723 yep, I didn't noticed because building from sources doesn't throw this error :neutral_face: , I think it's probably because is missing some Qt plugin.

shawnk0723, 2021-12-23:
Thank you for the quick reply. I tried with source code but still no luck. it's weird that all the other image viewers work except for gthumb. I guess it's my luck. LOL I'll just use other viewer for now. Thanks man. :)

hipersayanX, 2022-02-15:
New UI released with Webcamoid 9.0.0, closing.

closed 2022-02-15 by hipersayanX