webcamoid/webcamoid #217: Error when creating a virtual camera

ryanhardt41, 2020-02-05:
Hello guys,

I'm trying to run Webcamoid 8.7.1 on Zorin OS (a distro based on Ubuntu 18.04 LTS).
The application turns on and everything works fine, EXCEPT when I try to create a virtual camera.
The checkbox of "virtual camera" works **however** when I try to add a virtual camera it shows "Error creating camera:" even though I have entered the correct password.

This is an image of the issue I've been getting:
[image]
This is an image of my settings:
[image]

hipersayanX, 2020-02-05:
I think this could be related to [this](https://github.com/webcamoid/akvcam/issues/4).

lcarva, 2020-02-14:
I'm having the same issue on Fedora 30. I see the same error as initially reported when running both the AppImage and portable linux binary.

tuxflo, 2020-03-19:
I'm having the same issue on manjaro. Any ideas what to do? I also tried v2loopback but the resolution of the fake webcam is way too bad for screen sharing.

JosefLitos, 2020-11-27:
I am using arch, I managed to start the v4l2loopback using insmod and the extra/v4l....ko.xz module, but I can't use that camera anyway and can't save the virtual camera in webcamoid neither.

Kotarou0, 2021-06-18:
Same with Arch

JosefLitos, 2021-06-21:
I have solved this some time ago, you have to have a graphical polkit interface agent running in background (such as polkit-gnome).

hipersayanX, 2022-02-15:
Error messages have been improved in Webcamoid 9.0.0, closing.

closed 2022-02-15 by hipersayanX