webcamoid/webcamoid #218: Webcamoid crashes when pressing play with Virtual Camera enabled

pcgamer31098, 2020-02-08:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debbuger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates bellow that fit better your issue.

# Reporting a problem?

## Summary

Webcamoid crashes every time I press the play button at the bottom left when I have a video file selected as the source and I have the Virtual Camera option enabled.

## Current Behavior

Same as the summary.

## Expected Behavior

The program should playback the video file that I've selected.

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Select a video file as the source (in my case it was an .mp4 file)
2.  Enable the Virtual Camera option
3.  Press the play button at the bottom left

## Suggestions and tips

How would you solve the problem?

## Your Environment

* Webcamoid information: Webcamoid 8.7.1
* Operating System information: Windows 10
* Any other useful information: Screenshot of the error that showed up after the program crashed
[image]

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questinons and other matters?

Go ahead!

yboules-gwh, 2020-02-10:
I am running Webcamoid 8.7.1 on Windows 10 v1803.17134.1246. I use it primarily for the Virtual Camera feature, so that I can use a video as a camera source. I do not get a crash dialog, but I when I use a video as the camera source and minimize Webcamoid, a few minutes later I find the window is closed and Webcamoid is no longer running.
Virtual cam driver is a great and useful feature. I wish you can design Webcamoid, as a feature request, to run in the background and stream a video source as a virtual camera (perhaps minimize to system tray).
I do not know if it is closing after a few minutes because it is designed to do so or because it is crashing, but I would like it to continue to serve my video as a virtual camera source continuously in the background.
Thank you. I do not know if this is more of a bug or a feature request.

yboules-gwh, 2020-02-10:
I am running Webcamoid 8.7.1 on Windows 10 v1803.17134.1246. I use it primarily for the Virtual Camera feature, so that I can use a video as a camera source. I do not get a crash dialog, but I when I use a video as the camera source and minimize Webcamoid, a few minutes later I find the window is closed and Webcamoid is no longer running.

Virtual cam driver is a great and useful feature. I wish you can design Webcamoid, as a feature request, to run in the background and stream a video source as a virtual camera (perhaps minimize to system tray). Also, would be nice if it can play still images. It did for a minute, but then would not let me afterwards.

I do not know if it is closing after a few minutes because it is designed to do so or because it is crashing, but I would like it to continue to serve my video as a virtual camera source continuously in the background.

Thank you. I do not know if this is more of a bug or a feature request.

hipersayanX, 2020-02-12:
@yboules-gwh is a bug, Webcamoid is not supposed to do that, if possible open a new issue for reporting that specific bug.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX