webcamoid/webcamoid #219: FPS dropdown.. 25fps?

gregharewood, 2020-02-10:
Hi

My entire hopeful use case for webcamoid on the Mac is to force a more useful framerate.  Is that possible?  Will it control the hardware, or just reformat whatever is coming in?  In any case, my camera is defaulting to 30fps I think, but it a room with 50hz fluorescent lighting, it's horrible with the beat frequency.  I was hoping to use webcamoid to select 25fps (or 50fps), but I can't seem to.  24 is in the list.  The only fps in the list that avoids the problem is 5fps, and I AM using that but it is obviously jerky.  

Should I be able to type in a value?  Should it be available?  Is it showing a list of what the camera directly supports?  Is is simply missing because the software is US centric?  Please help!

Thx
Greg

hipersayanX, 2020-02-12:
All settings for format, resolution and FPS are provided by AV Fundation, Webcamoid hasn't direct access to the hardware.
Also, even if you set the FPS of the camera to a specific value, the real FPS could be lower, even without applying any processing step in-between, and that is a hardware limitation that can't be fixed by software.
Even if I allow you to type a FPS you will never get the desired frame rate because is not technically possible for the hardware and the port speed.

gregharewood, 2020-02-12:
Thank you so much for replying!!  I did think that maybe there wasn’t that level of control, but I hoped anyway,

Kind regards
Greg

closed 2020-02-27 by hipersayanX