webcamoid/webcamoid #220: Virtual Camera not visible in Photo Booth (MacOS)

BoD, 2020-02-21:
## Summary

This doc https://github.com/webcamoid/webcamoid/wiki/Virtual-camera-support says that I should see the virtual webcam in Photo Booth, but I don't see it.

## Current Behavior
The virtual cam is not in the list in Photo Booth.

## Expected Behavior
It should be in the list.

## Steps to Reproduce

1. Install Webcamoid
1. Run it
1. In Preferences, check Virtual Camera then Add and choose a name
1. Run Photo Booth
1. Check the Camera item in the menu bar: there's only 1 choice: your "real" camera 

## Suggestions and tips

N/A

## Your Environment

* Webcamoid information: 8.7.1 MacOS 64bits
* Operating System information: MacOS Catalina 10.15.2 (19C57)
* Any other useful information: N/A

mthorta001, 2020-03-14:
I have the same issue.   Photo Booth can't choose my virtual camera.   
But the virtual camera can be recognized by Skype/Chrome.
What's the problem.

jenniferhuang, 2020-03-14:
I have the same issue. Photo Booth can't choose my virtual camera.
But the virtual camera can be recognized by Skype/Chrome.

hipersayanX, 2020-03-14:
If I remember well, when testing the virtual camera on Photo Booth I disabled the [SIP](https://www.macworld.com/article/2986118/how-to-modify-system-integrity-protection-in-el-capitan.html), otherwise it will require the virtual camera plugin to be signed, which is impossible for me right now.

jenniferhuang, 2020-03-15:
@hipersayanX  I disabled the SIP,  Photo Booth can choose virtual camera now.  thanks.

mvpereirabr, 2020-04-18:
I'm using macOS Cataline (10.15.4). Downloaded webcamoid 8.7.1 to try to use my GoPro Hero3 as webcam. Couldn't have any success after trying to use CameraSuite and ffmpeg. What I realized is that when I check "Virtual Camera", the "Devices" dropdown list is always empty, no matter the order I start CameraSuite, ffmpeg or webcamoid, and to try that, I chose my "Screen 0" as video source before going to output settings. I think this is the reason I can't add a Virtual Camera to my system.

tedcarnahan, 2020-04-22:
Try an upgrade to Google Chrome dev channel, they have changed the code to make it possible to load unsigned libraries for this kind of thing.

hipersayanX, 2022-02-15:
As said before, the virtual camera must be signed, or the user must disable the SIP.

closed 2022-02-15 by hipersayanX