webcamoid/webcamoid #223: MacOS binary version is not notarized, cannot run on MacOS without paying for Apple Developer Credentials

dg1sek, 2020-03-03:
When running the binary release on MacOS Catalina 10.15.3 I get an error message saying the developer cannot be verified. 

[image]

Unfortunately on my system with recent Catalina, it's not possible to workaround this message in Preferences-Security as it was the case in previous MacOS versions.

I also cannot notarize it manually over command line, as I don't have a developer account.

Would it be possible for you to notarize the MacOS binary?

hipersayanX, 2020-03-04:
Did you tried [this](https://support.apple.com/guide/mac-help/open-a-mac-app-from-an-unidentified-developer-mh40616/mac)?

>Would it be possible for you to notarize the MacOS binary?

Impossible, I have no way to pay for signing.

core-code, 2020-03-05:
just use right-click=>open as has always been necessary. or use the "open anyway" button in the system preferences

[image]
[image]

core-code, 2020-03-05:
the bug report is misleading, you can easily launch this and any other non-notarized app even on Catalina. no developer membership is needed, i am not sure where this myth is coming from.

ferdi2005, 2020-03-21:
Notarizing your software is **FREE**, please do it, you won't need to pay anything
https://developer.apple.com/documentation/xcode/notarizing_macos_software_before_distribution

hipersayanX, 2020-03-21:
@ferdi2005 will check, thanks!

closed 2022-02-15 by hipersayanX