webcamoid/webcamoid #226: Unable to run webcamoid: Unknown option 'q'.

gordonel, 2020-03-08:
## Summary

When running webcamoid from QT Creator, I get the following in the console

## Current Behavior

```bash
18:19:49: Starting /Users/gordonel/webcamoid/StandAlone/release/Qt5.14.1/clangpp/x86_64/bin/webcamoid --vcam /Users/gordonel/webcamoid/libAvKys/Plugins/VirtualCamera/src/cmio/VirtualCamera -r -p /Users/gordonel/webcamoid/libAvKys/Plugins -q /Users/gordonel/webcamoid/libAvKys...
Webcamoid: Unknown option 'q'.
18:19:49: /Users/gordonel/webcamoid/StandAlone/release/Qt5.14.1/clangpp/x86_64/bin/webcamoid exited with code 1
```

## Expected Behavior

No "Webcamoid: Unknown option 'q'." error

## Steps to Reproduce

1. Clone the repo afresh
2. Install the deps afresh
3. In short, follow all the instructions in the wiki for macOS
4. Attempt running from qt-creator

## Your Environment

* Webcamoid information: master
* Operating System information: macOS Catalina 10.15.3

hipersayanX, 2020-03-08:
I have not updated the wiki, the **q** option was removed in master, isn't necessary anymore.

gordonel, 2020-03-08:
That fixes it, but still, the app is unusable. See my other issue. Here's what I have in the console:
```
18:57:19: Starting /Users/gordonel/webcamoid/StandAlone/release/Qt5.14.1/clangpp/x86_64/bin/webcamoid --vcam /Users/gordonel/webcamoid/libAvKys/Plugins/VirtualCamera/src/cmio/VirtualCamera -r -p /Users/gordonel/webcamoid/libAvKys/Plugins...
qrc:/Webcamoid/share/qml/main.qml:523:13: QML ToolBar: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/main.qml:92: ReferenceError: togglePlay is not defined
qrc:/Webcamoid/share/qml/MediaBar.qml:63:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
18:57:55: /Users/gordonel/webcamoid/StandAlone/release/Qt5.14.1/clangpp/x86_64/bin/webcamoid exited with code 0
```

hipersayanX, 2020-03-08:
I'm in the middle of [UI redesign](https://github.com/webcamoid/webcamoid/issues/216) right now, those messages will be common until I finish with it.

beppe9000, 2020-03-21:
webcamoid-portable-daily-win64 has this error

hipersayanX, 2020-03-22:
@beppe9000 remove the _-q "%~dp0/path"_ part of the command. I will update the script later.

commit 3f1d6ff2 (2020-03-26): Removed old FFmpeg code. Using Screen.width and Screen.height instead of
Screen.desktopAvailableWidth and Screen.desktopAvailableHeight, this may
fix #177. Removed -q option in Windows launcher, this may fix #226.
closed 2020-03-26 by hipersayanX