webcamoid/webcamoid #227: Add advanced OBS-like settings for input sources

gordonel, 2020-03-09:
# Want a new feature?

Here's how webcam settings look in OBS.
[image]
As you can see, it allows greater control over the colour space, stream choice etc.

# Questinons and other matters?

Would this be possible to add?

hipersayanX, 2020-03-09:
I don't see any difference with what already offers.

[image]

gordonel, 2020-03-09:
> [![Screenshot\_20200309\_124236][Screenshot_20200309_124236]][Screenshot_20200309_124236]

There's no selection for colour space (rec.709) and range (partial vs full)

\-------- Original Message --------

[Screenshot_20200309_124236]: https://user-images.githubusercontent.com/1249885/76231443-04e92180-6204-11ea-8f52-4c557da51126.png
[view it on GitHub]: https://github.com/webcamoid/webcamoid/issues/227?email_source=notifications&email_token=AKYUTQVL46HDSAPWW7OYFVDRGUFSJA5CNFSM4LEHVMT2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEOHZBQA#issuecomment-596611264
[unsubscribe]: https://github.com/
[AKYUTQUWTYZ2JZBC24VOS23RGUFSJA5CNFSM4LEHVMT2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEOHZBQA.gif]: https://github.com/notifications/beacon/AKYUTQUWTYZ2JZBC24VOS23RGUFSJA5CNFSM4LEHVMT2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEOHZBQA.gif

gordonel, 2020-03-09:
Also, there is no detailed description of the stream. The OBS's description tells me the stream has 4:2:2 chroma subsampling and it's an 8 bit stream

hipersayanX, 2020-03-09:
>The OBS's description tells me the stream has 4:2:2 chroma subsampling and it's an 8 bit stream

That is just a mere description of the format and does not make much difference.

I'll investigate the other two, but I'm pretty sure that those are options for rendering rater than source options; cameras, desktops and videos does not provides any option like that.

Buffering option doesn't make any sense unless the source is a video stream.