webcamoid/webcamoid #230: Virtual camera video track freezes but audio continues on macOS Catalina

andrewodri, 2020-03-18:
When accessing the camera using Google Chrome, the video feed freezes while audio continues to play. This typically happens between 1 and 10 secs after the camera is activated via [MediaDevice.getUserMedia()](https://developer.mozilla.org/en-US/docs/Web/API/MediaDevices/getUserMedia).

## Current Behavior

In the Chrome Dev Tools I've exported the `MediaStream` element for the camera both before and after the issues occurs. I tried this with video & audio enabled, as well video only. In all instances the result of the object below was the same, with the video freezing between 1 and 10 seconds into the feed, audio continuing (when enabled), and with the active property returning true.

```
MediaStream {
  id: "QXfmN8ZZDdaDSGvQI8brMcVZykKB3gXqC7oU",
  active: true,
  onaddtrack: null,
  onremovetrack: null,
  onactive: null,
  oninactive: null,
  __proto__: MediaStream
}
```

In the webcamoid interface, the video stream plays continuously without any issues... Which would lead me to think that the virtual camera doesn't like something with the input stream, or it's tripping up on something during the transcoding process.

The input source is a TCP stream that is currently being fed in from GStreamer. I've tried both low latency and high latency variations of streaming from from my iPhone's camera using the following command that utilizes [qvh](https://github.com/danielpaulus/quicktime_video_hack):

I've also tried this with every combination of transcoders available in the webcamoid settings (i.e. ffmpeg and generic).

_Low latency, low sync_
```
go run main.go gstreamer --pipeline "queue name=audio_target ! wavparse ignore-length=true ! audioconvert ! faac ! aacparse ! mux.
  queue name=video_target ! h264parse ! vtdec ! videoconvert ! aspectratiocrop aspect-ratio=16/9 ! videoscale ! video/x-raw,width=1280,height=720 ! videorate ! video/x-raw,framerate=15/1 ! x264enc tune=zerolatency speed-preset=superfast ! h264parse ! mux.
  mpegtsmux name=mux ! tcpserversink host=localhost port=7000 sync=false"
```

_High latency, high sync_
```
go run main.go gstreamer --pipeline "queue name=audio_target ! wavparse ignore-length=true ! audioconvert ! faac ! aacparse ! queue2 ! mux.
  queue name=video_target ! h264parse ! vtdec ! videoconvert ! aspectratiocrop aspect-ratio=16/9 ! videoscale ! video/x-raw,width=1280,height=720 ! videorate ! video/x-raw,framerate=15/1 ! x264enc tune=zerolatency speed-preset=superfast ! h264parse ! queue2 ! mux.
  mpegtsmux name=mux ! tcpserversink host=localhost port=7000 recover-policy=keyframe sync-method=latest-keyframe"
```

## Expected Behavior

The virtual camera will continue playing the video, or crash gracefully is an error is encountered.

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Install Go@1.13, clone qvh (`git clone https://github.com/danielpaulus/quicktime_video_hack.git`), and following the building instructions
2.  Run either/both of the commands above (i.e. "Low latency" and "High latency")
3.  Add a source in webcamoid with the path `tcp://localhost:7000`. The video and audio tracks should be detected, and the preview will continue for hours
4.  Open Chrome, and try to visit a website that will access and preview your camera, such as Google Hangouts

## Suggestions and tips

I only just started playing around with GStreamer today, and I just don't have enough knowledge with macOS development to get to the bottom of this one at this point in time... Any insight in where in the process transcoding is happening, what AkVirtualCamera wants/needs in order to work smoothly, and what kind of stream/encapsulation formats I could tweak on my end would super helpful! Thanks!

## Your Environment

* Webcamoid information: 8.7.1 macOS (Downloaded binary)
* Operating System information: macOS Catalina 10.15.3

andrewodri, 2020-03-18:
Here are some system errors that are generated whenever Chrome attempts to access the camera:

```
error	13:44:26.948714-0700	Google Chrome Helper (Plugin)	AddInstanceForFactory: No factory registered for id <CFUUID 0x7f8b9cc08d50> F8BB1C28-BAE8-11D6-9C31-00039315CD46
error	13:44:26.979245-0700	Google Chrome Helper (Plugin)	 HALC_ShellDriverPlugIn::Open: Can't get a pointer to the Open routine
error	13:44:26.995718-0700	Google Chrome Helper (Plugin)	CMIO_DAL_Object.cpp:720:AddPropertyListener object hasn't been initialized, object ID: 34 class ID: astr
error	13:44:26.995750-0700	Google Chrome Helper (Plugin)	CMIOHardware.cpp:667:CMIOObjectAddPropertyListener listener was already added
error	13:44:27.148730-0700	Google Chrome Helper (Plugin)	CMIO_DAL_Stream.cpp:104:CopyBufferQueue stream hasn't been initialized, object ID: 33 class ID: adev
```

andrewodri, 2020-03-19:
Just an update: I was playing around with various colour profiles to attempt to match it as closely as possible to the virtual camera defaults (e.g. muxing into matroska with YUY2 which is the virtual camera default, changing the frame rates to 30 fps, etc.), and noticed that when the virtual camera is selected as the source, the colours on the test frame fluctuate... So even with the test frame selected, I will get colour fluctuations for up to 10 secs before the fluctuations stop.

hipersayanX, 2020-03-21:
That depends on what do you mean with color fluctuation, if for example the yellow color in YUV2 format is darker and the yellow color is brighter in RGB, that's ok, YUV formats uses between 12 to 16 bits to represent a color, while RGB uses 24 bits, so colors doesn't looks the same in both formats.

andrewodri, 2020-03-21:
> That depends on what do you mean with color fluctuation, if for example the yellow color in YUV2 format is darker and the yellow color is brighter in RGB, that's ok, YUV formats uses between 12 to [...]

Yup, all good! I just mentioned that to point out that even with a different (and likely more thoroughly tested) video source, the issue of the freezing camera still occurs. Also, I didn't mention before; but thank you for all your hard work!

tedcarnahan, 2020-04-07:
I think I have this bug too.  Also installed through `brew cask`, also running webcamoid v8.7.1 on Catalina. By running Chrome from Terminal I get this message on stderr at the time I start the virtual camera in a browser window:

> 2020-04-07 12:00:55.214 Google Chrome[28263:960712] Error loading /Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera:  dlopen(/Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera, 265): no suitable image found.  Did find:
/Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera: code signature in (/Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera) not valid for use in process using Library Validation: mapped file has no cdhash, completely unsigned? Code has to be at least ad-hoc signed.
dyld: warning, LC_RPATH @executable_path/../Frameworks in /Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera being ignored in restricted program because of @executable_path (Codesign main executable with Library Validation to allow @ paths)
dyld: warning, LC_RPATH @executable_path/../Frameworks in /Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera being ignored in restricted program because of @executable_path (Codesign main executable with Library Validation to allow @ paths)`

hipersayanX, 2020-04-12:
@tedcarnahan Check [this](https://github.com/webcamoid/webcamoid/issues/220#issuecomment-599091191).

tedcarnahan, 2020-04-22:
I was able to get it working without disabling SIP by upgrading to Google Chrome Version 84.0.4115.5 (Official Build) dev (64-bit).  Apparently they moved the interface code that calls the virtual webcam library to a separate process that doesn't require code signing.

hipersayanX, 2020-04-24:
@tedcarnahan Considering how Google handles security, is more probable that the development version does not check for that (for debugging purposes), while the release version does.

tedcarnahan, 2020-04-24:
Take a look at Chromium [Issue 990381: Virtual cameras no longer show up in Chrome because lacking com.apple.security.cs.disable-library-validation entitlement](https://bugs.chromium.org/p/chromium/issues/detail?id=990381)

hipersayanX, 2020-04-25:
@tedcarnahan Oh get it, if I understand it well, they manage the camera from a separated process (it does not matter if the binaries are signed or not) and send the frames to the main process with some IPC mechanism, so they don't risk Chome it self, clever.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX