webcamoid/webcamoid #231: Windows: Keeps crashing immediately after start

jekay001, 2020-03-18:
First it worked for about an hour. Now it keeps crashing immediately after start. Already reinstalled several times, tried portable versions, 32 and 64bit. Nothing works anymore.
Win10.

jekay001, 2020-03-18:
I used the new virtual cam.

beppe9000, 2020-03-19:
I had limited success by deleting the app folder and extract again the portable release.
Is there a place where things are logged? The 'app crashed' dialogue of windows is pretty useless, although it does offer the option of debugging it... I shall test it but I think it needs to be an explicit debug build with pdb files for visual studio to work well with it.

hipersayanX, 2020-03-21:
@beppe9000 error logging is one of the features I want to add.

beppe9000, 2020-03-21:
no way to get you a stack trace from the crash ?

beppe9000, 2020-03-22:
I found a bunch of crash dumps in windows app data. IDK if they are useful to diagnose the problems
[CrashDumps.zip](https://github.com/webcamoid/webcamoid/files/4365308/CrashDumps.zip)

hipersayanX, 2020-03-22:
@beppe9000 will check, thanks!

valbok, 2020-09-02:
Can somebody rebuild directshow plugin in QtMultimedia without 
hr = pVideoControl->GetFrameRateList(pPin, iIndex, size, &listSize, &frameRates);
and check if it works?
can it be an issue with negotiating the frame rates?

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX