webcamoid/webcamoid #232: Memory leak back on 8.7.1?

perdja, 2020-03-18:
Hello,

I saw that on a previous version there was a memory leak which was updated according to records (post #159 ). By behaviour it seems that this leak has returned in version 8.7.1 as I can confirm the described behaviour recorded for previous version. After some use I am forced to restart computer due to memory. Can this be confirmed?

hipersayanX, 2020-03-18:
Do you remember which was the target program?

perdja, 2020-03-19:
Hello, thanks for quick response! 
Well at the time I was channeling to Teams from OBS using the virtual cam capabilities, but please be aware that I experienced this also a few days ago without any particular destination, which also was indicated in post #159 . I operate with Mojave 10.4.6. I was really happy being able to use the webcamiod as virtual camera capability and really appreciate your work!

esnho, 2020-04-01:
I'm experiencing something similar but related to hard drive space on MacOS Catalina, I've got a two screen setup, grabbing one of the desktop quickly fill my hard drive.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX