webcamoid/webcamoid #233: VirtualCamera (Mac) support for "Zoom" app

xpluscal, 2020-03-21:
## Summary

I have created a virtual camera and it shows on skype, but it does not show on Zoom [https://zoom.us/](https://zoom.us/) - a popular video meeting app.

Operating system: MacOS 10.14.6

## Current Behavior

The virtual camera doesn't show (only facetime HD camera is shown)

## Expected Behavior

The test camera to appear.

hipersayanX, 2020-03-21:
Did you tried [this](https://support.apple.com/guide/mac-help/open-a-mac-app-from-an-unidentified-developer-mh40616/mac)?

beppe9000, 2020-03-21:
on windows webcamoid needs to be restarted completely to see the virtual webcam, maybe on mac too ?

hipersayanX, 2020-03-22:
@beppe9000 if that's the case, open a new bug report specifically for Windows.

karipov, 2020-03-23:
Can confirm this issue on MacOS

neufeldtech, 2020-03-24:
I was able to get it to work with the Zoom app on OSX.
- Quit Zoom completely (CMD + Q)
- Quit Webcamoid Completely
- Open Webcamoid (right click and select Open) and configure virtual cam output
- Press "Stop" And "Play" button in Webcamoid again to start output to virtual cam
- Open Zoom and select the virtual camera output

SergiyZhyrov, 2020-04-14:
https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_security_cs_disable-library-validation
seems Zoom start using `library-validation` 
- library validation prevents an app from loading frameworks, plug-ins, or libraries unless they’re either signed by Apple or signed with the same team ID as the app.

ikester, 2020-04-15:
Will disabling the SIP as recommended here https://github.com/webcamoid/webcamoid/issues/220#issuecomment-599091191 for the PhotoBooth issue also work for this?

ikester, 2020-04-15:
Also, how do we know Zoom started using library-validation? Is that also what FaceTime and PhotoBooth do these days? Because the virtual camera is not available on those first party apps either.

hipersayanX, 2020-04-16:
@ikester Apple introduced the SIP thing a long time ago, sadly I can't do much about that :confused:

ikester, 2020-04-16:
Thanks @hipersayanX. Is that what's happening with all the first party apps? Skype seems to be OK with it.

I wonder how webcam manufacturers get plug&play working. I don't remember having to install any signed DALs for other webcams to work.

hipersayanX, 2020-04-16:
@ikester

>I wonder how webcam manufacturers get plug&play working. I don't remember having to install any signed DALs for other webcams to work.

Webcam manufacturers just need to comply with [UVC standard](https://en.wikipedia.org/wiki/USB_video_device_class), they don't need to write any driver since the support is provided by the default Mac driver, and if they do (for example, to provide features not available with the default driver), they must pay for the signing.

SergiyZhyrov, 2020-04-16:
Any USB webcam use Apple plugin
`/System/Library/Frameworks/CoreMediaIO.framework/Versions/A/Resources/VDC.plugin`, previous macOS located at
`/Library/CoreMediaIO/Plug- Ins/DAL/`
See `DAL Sample Explained.pdf` (https://github.com/lvsti/CoreMediaIO-DAL-Example/blob/master/Documentation/DAL%20Sample%20Explained.pdf):
Here is an example of the values used for the Apple-provided VDC camera plug-in:
    <key>CMIOHardwarePlugInLazyLoadingInfo</key>
    <array>
        <dict>
            <key>IOProviderClass</key>
            <string>IOUSBDevice</string>
            <key>idProduct</key>
            <integer>34049</integer>
            <key>idVendor</key>
            <integer>1452</integer>
</dict>
<dict>
            <key>IOProviderClass</key>
            <string>IOUSBDevice</string>
            <key>bDeviceClass</key>
            <integer>239</integer>
            <key>bDeviceSubClass</key>
            <integer>2</integer>
            <key>bDeviceProtocol</key>
            <integer>1</integer>
        </dict>
    </array>

So all virtual-cam put it plug-ins to `/Library/CoreMediaIO/Plug- Ins/DAL/`. And if any app compiled with `library validation`, then method 
[AVCaptureDevice devices] (https://developer.apple.com/documentation/avfoundation/avcapturedevice/1386237-devices?language=objc) does not return virtual cam. Method find plugin binaries but unable load them with `dlopen`:
`Error loading /Library/CoreMediaIO/Plug-Ins/DAL/ManyCamVideoDeviceCMIO.plugin/Contents/MacOS/ManyCamVideoDeviceCMIO:  dlopen(/Library/CoreMediaIO/Plug-Ins/DAL/ManyCamVideoDeviceCMIO.plugin/Contents/MacOS/ManyCamVideoDeviceCMIO, 262): no suitable image found.  Did find:
	/Library/CoreMediaIO/Plug-Ins/DAL/ManyCamVideoDeviceCMIO.plugin/Contents/MacOS/ManyCamVideoDeviceCMIO: code signature in (/Library/CoreMediaIO/Plug-Ins/DAL/ManyCamVideoDeviceCMIO.plugin/Contents/MacOS/ManyCamVideoDeviceCMIO) not valid for use in process using Library Validation: mapping process and mapped file (non-platform) have different Team IDs`
Maybe now plugin can be signed and distributed via AppStore as `System Extension` (https://developer.apple.com/videos/play/wwdc2019/702/)

ikester, 2020-04-16:
@hipersayanX interesting...

Would it make sense to try to emulate a USB Video Device Class with IOKit or something like it? Instead of doing it with a DAL? So it can use the standard Apple webcam plugin @SergiyZhyrov mentioned?

I searched around and couldn't find any virtual/emulated USB projects for Mac, and only a couple of references on StackOverflow (e.g. https://stackoverflow.com/questions/20064648/creating-virtual-usb-device-in-mac-os-x) that suggested it would be difficult. But that was back 7 years ago. Maybe there's a better approach now?

beppe9000, 2020-04-17:
this kind trusted computing is obnoxious....

hipersayanX, 2020-04-17:
@ikester I had the same thought while writing the plugin, but then I would need to study the USB and UVC protocol, plus finding code examples doing that, all of that in a short time, a complete nightmare with absolute zero advantages, because even if I do that it still will require code signing.

@beppe9000 You see... but they do that to "protect" the user, even though Linux has not such things and the attacks on driver layer are rare.

SergiyZhyrov, 2020-04-17:
That's why Zoom enable `library-validation` :
https://nvd.nist.gov/vuln/detail/CVE-2020-11470
https://objective-see.com/blog/blog_0x56.html

ikester, 2020-04-22:
@hipersayanX I agree that it would be a heavy lift to figure out the USB/UVC protocol and implementation.

Do you know if the library validation restriction also applies to USB hubs/drivers? If it doesn't, then the code signing shouldn't be an issue. As long as the device emulates a regular USB webcam. As you said above, webcam manufacturers don't have to do anything special, other than comply with the proper USB standards.

hipersayanX, 2020-04-24:
@ikester

>Do you know if the library validation restriction also applies to USB hubs/drivers?

It applies to everything from drivers to plugins and apps (the restriction is a bit lax in the last one), the idea is to put a barrier to malware so it can't infect the system, but it also put a barrier for FOSS developers that can't pay for the signing, and even if you can pay for the signing, Apple can still kick you off for whatever reason because they don't like your project.

jlakser49, 2020-04-24:
I am getting this issue as well. 
Would like to have it working without disabling SIP. 
anyone have a solution?

what strange is some Macs work and some don't.
same os. 10.14.6

anyideas?

jlakser49, 2020-04-24:
found a solution here: 
https://www.reddit.com/r/VIDEOENGINEERING/comments/fy7xi3/fyi_zoom_v4610_breaks_blackmagic_capture_devices/

Hopefully this resolves itself in the next few updates. Sounds like a complex issue. Not just Webcamoid is affected.

ikester, 2020-04-24:
@hipersayanX 

> It applies to everything from drivers to plugins and apps (the restriction is a bit lax in the last one), the idea is to put a barrier to malware so it can't infect the system, ...

OK. So maybe it cannot be done in software, but USB webcams are still able to stream video without special authorization right?

I know this is a bit esoteric, but how about a physical loopback device? A small Arduino or something like it that plugs into the USB port, receives video frames from an application and sends them back in using standard UVC protocol? Does that sound like a nice hobby project for someone? :)

jlakser49, 2020-04-24:
> @hipersayanX
>
>
> OK. So maybe it cannot be done in software, but USB webcams are still able to stream video without special authorization right?
>
> I know this is a bit esoteric, but how about a [...]

I am doing this with a AJA Utap, however they are sold out everywhere because of the pandemic.
Thats how i found Webcamoid, Seraching for a software only solution for more users.

ikester, 2020-04-24:
@jlakser49 that's cool but, as you said, they're sold out and $345!

I did some googling around and a DIY approach may not be as hard as I originally thought. It seems like a RaspberryPi Zero can be used to emulate USB Gadgets and recent Linux kernels have made this easier to use:
http://www.linux-usb.org/gadget/
https://github.com/torvalds/linux/blob/master/Documentation/usb/gadget-testing.rst#18-uvc-function
https://www.kernel.org/doc/Documentation/usb/gadget_configfs.txt
https://www.isticktoit.net/?p=1383
https://blog.gbaman.info/?p=699

The trick will be figuring out a configuration that will allow the Pi to both, receive video data form the host, and loop it back emulating a UVC device.

With some time, ingenuity and patience we may be able to create a video loopback device for $5.

jlakser49, 2020-04-24:
@ikester 
would love to see that put together. 
not sure i can wrap my head around how we would get the video out to the pi.

hipersayanX, 2020-04-25:
@ikester well, what you are proposing is the very definition of hacking :laughing: 
A solution would be sending the stream to the RaspberryPi, the RaspberryPi sends the stream to the hdmi port, and finally you use a [hdmi capture card](https://www.youtube.com/watch?v=H7CEyVAJf5k)

ikester, 2020-04-25:
Ha!

But if I could do it all in a tiny loopback RPi Zero it would be more elegant, and WAY cheaper :)

Like all other tech "arm races" I wonder if the next stage is to disable UVC devices and only allow encrypted, signed, "trusted" connections (like HDMI/HDCP).

hipersayanX, 2020-04-25:
@ikester do you mean just connecting the RPi and PC ports together? even in that case you still need an extra peripheral.

https://www.quora.com/Is-it-possible-to-connect-2-computers-with-a-USB-cable
https://superuser.com/questions/795053/how-do-i-connect-two-computers-using-usb-3-0

I think the hdmi solution is expensive but much simpler because you are using the capture card as a display, you just need to write a program that receives the video and shows it full screen, you don't need to write any kernel code, maintenance becomes much simpler, and you don't need any special privileges for that. You can transfer the data using your local network.

ikester, 2020-04-25:
@hipersayanX With a RaspberryPi Zero, Model A and now the 4, the USB port can be used OTG (or Gadget) mode. Which means it can act as a slave instead of a hub. So you can use a regular USB cable and connect it from a PC/Mac's USB port to a Pi Zero directly. Here's a simple tutorial for making a Pi appear as an Ethernet adapter:
https://www.circuitbasics.com/raspberry-pi-zero-ethernet-gadget/

This way you can power and connect to the Pi with a single cable and without a need for WiFi.

In the the links I posted above, there's information about how to use a different kernel module (driver) to make the Pi appear as a UVC webcam instead.

Here's a C program that reportedly does exactly that (abeit flakey):
https://github.com/jdonald/uvc-gadget

Modern Linux kernel versions supposedly allow a multi-mode configuration. So, in theory, it's possible that a Pi Zero could be configured to operate as both, an Ethernet adapter, and a UVC webcam.

If this can be made to work reliably, it shouldn't be too hard to get video frames out to the Pi over the "network" and have the Pi stream the video back in through the UVC interface. How cool would that be?!

hipersayanX, 2020-04-25:
@ikester Interesting, I didn't know that, for $10 we can have a fake camera device,  we can use either wifi or bluetooth for the stream, or can we use the OTG port at same time to send the stream?

ikester, 2020-04-25:
@hipersayanX  That's the idea. But it's easier said that done.

I actually took out my Pi Zero for the first time in years to try this out. I'm already connected to it as a virtual Ethernet device. I have no doubts we can send video out that way.

The difficult part is making it work as UVC **AND** Ethernet simultaneously. I want to see if that code works with a modern kernel. If I get anything remotely working I'll let you know.

But even if it can't work in multi-mode, if just the UVC part works we can always get the video out via WiFi as you suggested.

shanika, 2020-05-02:
If snap cam or manycam works for an app on OSX does it mean webcamoid also should work? I believe all these apps use similar technology. I still had no luck with getting webcamoid virtual cam listed on any of the other apps such as MS Teams, Skype.

hipersayanX, 2020-05-05:
@shanika

>If snap cam or manycam works for an app on OSX does it mean webcamoid also should work?

Because they can pay for code signing, I can not.

PatientZerobeat, 2020-05-09:
> I was able to get it to work with the Zoom app on OSX.
>
> * Quit Zoom completely (CMD + Q)
> * Quit Webcamoid Completely
> * Open Webcamoid (right click and select Open) and configure virtual cam output
> * [...]

I understand the first two instructions completely, but the next two are a [partial] mystery to me. Can you explain it to somebody way less computer literate? 

Why right-click and select "Open"? How is this different from double-clicking on the application or launching from the Dock? I've never known there to be a difference. 

What exactly is "configure virtual cam output?" I clicked on the checkmark for "Virtual Camera" and then pressed the "Add" button and gave it a personal name. Anything more?

I only have one button which toggles between Play and Stop. How can I press both buttons as the instruction appears to state? If I am seeing a representation of my screen inside the small Webcamoid window, does this mean I am "output to virtual cam"?

In no case does the Video Preference inside Zoom reveal a camera source (my computer MacPro doesn't have any kind of camera; hence my desire to use my video card's output as a source for Zoom)

MacOs 10.4.6, Webcamoid 8.7.1, Zoom 4.6.11

hipersayanX, 2020-05-11:
@PatientZerobeat 

>Why right-click and select "Open"? How is this different from double-clicking on the application or launching from the Dock? I've never known there to be a difference.

Because the code is not signed, you must give explicit permission to execute Webcamoid. You do that step only once.

>What exactly is "configure virtual cam output?" I clicked on the checkmark for "Virtual Camera" and then pressed the "Add" button and gave it a personal name. Anything more?

No, that's all. What you did is correct.

>I only have one button which toggles between Play and Stop. How can I press both buttons as the instruction appears to state? If I am seeing a representation of my screen inside the small Webcamoid window, does this mean I am "output to virtual cam"?

Once you have created the the virtual camera, if you are playing a webcam, the desktop or a video, it will already send the video to the virtual camera.

SergiyZhyrov, 2020-06-03:
Strange but snap-camera work in Zoom and OBS studio. In zoom there are error in terminal
"Error loading /Library/CoreMediaIO/Plug-Ins/DAL/SnapCamera.plugin/Contents/MacOS/SnapCamera:  dlopen(/Library/CoreMediaIO/Plug-Ins/DAL/SnapCamera.plugin/Contents/MacOS/SnapCamera, 265): no suitable image found.  Did find:
	/Library/CoreMediaIO/Plug-Ins/DAL/SnapCamera.plugin/Contents/MacOS/SnapCamera: code signature in (/Library/CoreMediaIO/Plug-Ins/DAL/SnapCamera.plugin/Contents/MacOS/SnapCamera) not valid for use in process using Library Validation: mapping process and mapped file (non-platform) have different Team IDs"
but camera present in list and work.

But in QuickTime (which work fine with all virtual cameras) show blank image for snap-camera

SergiyZhyrov, 2020-06-16:
zoom have inside bundle one more app compiled without library validation, in this way zoom can work with other virtual cam, it use another one to receive frames. To make this second app to work with your virtual cam, this app should be "notarized" by zoom. See
https://support.zoom.us/hc/en-us/articles/360044801671-Supported-virtual-cameras-for-macOS

hipersayanX, 2020-06-16:
@SergiyZhyrov will check, thanks!

PatientZerobeat, 2020-06-17:
Unfortunately, nothing I try works. I'm just trying to get Zoom to see my display's video instead of the video coming from a camera. It's weird that this isn't built into the software (Zoom). How can it be that this is such a unimplemented feature, to output the video that I'm seeing on my display to Zoom, especially since Zoom already has this feature called "share screen"?

varunTheBaboon, 2020-12-07:
> @hipersayanX That's the idea. But it's easier said that done.
>
> I actually took out my Pi Zero for the first time in years to try this out. I'm already connected to it as a virtual Ethernet device. I [...]

@ikester Any updates on the project?

ikester, 2020-12-07:
> @ikester Any updates on the project?

@varunTheBaboon unfortunately not. I could never get the gadget to work as both UVC and Eth simultaneously. I gave up after trying many, many iterations. If anyone else has a breakthrough I'd love to hear about it.

varunTheBaboon, 2020-12-07:
> @varunTheBaboon unfortunately not. I could never get the gadget to work as both UVC and Eth simultaneously. I gave up after trying many, many iterations. If anyone else has a breakthrough I'd love [...]

@ikester Did you try the alternate way that you were suggesting by getting video out through wifi while emulating UVC?

ikester, 2020-12-07:
> @ikester Did you try the alternate way that you were suggesting by getting video out through wifi while emulating UVC?

@varunTheBaboon No. I didn't get to try that because I was never able to send "synthetic" video over UVC. Meaning that I wasn't able to send a static image or a video file as a webcam feed. The only way I got it to work was by sending video directly from a v4l2 source (a RPi camera module). For that to work I would first need to figure out how to re-format/-encode a video stream so that it works as a source for the UVC driver.

hipersayanX, 2020-12-17:
@ikester just take my advice, encode the video in the PC as for example as webm, send it through the network, read the video stream in the Raspberry and play it full screen (using ffplay, mplayer or mpv), then connect the Raspberry to a HDMI capture card to make it look like a webcam, no kernel programming needed, for 35 bucks or less you get the full thing working, extra points if you can send the video directly to the HDMI port.

ikester, 2020-12-17:
> @ikester just take my advice, encode the video in the PC as for example as webm, send it through the network, read the video stream in the Raspberry and play it full screen (using ffplay, mplayer or [...]

Well, if I had an HDMI capture card then I wouldn't need any of this to begin with :) You're forgetting to factor in the price of the HDMI capture card, which is much higher than $35.

hipersayanX, 2020-12-18:
@ikester look at [this](https://www.amazon.com/TenYua-Capture-Camcorder-Recording-Streaming/dp/B089LGRN14), what do you think? would it work? :smiley:

ikester, 2020-12-18:
> @ikester look at [this](https://www.amazon.com/TenYua-Capture-Camcorder-Recording-Streaming/dp/B089LGRN14), what do you think? would it work? 😃

Wow, that's surprisingly affordable! I guess I was thinking of the 4K capture cards :)

hipersayanX, 2020-12-18:
@ikester I've spotted that captured recently, and have found some local stores selling it, that's not a 4K gaming capturer but is more than enough for tricking the average online teacher :grin: 
Also if it works, I'm planning to just take out the soft vcams and going for this solution.

hipersayanX, 2022-01-26:
I've tested the new driver with the Zoom test call and it seems to be [working fine](https://github.com/webcamoid/akvirtualcamera/wiki/Support-status#mac).
So I changed this issue title to take and put in practice the idea of @ikester in the future.