webcamoid/webcamoid #234: qml start problem on Ubuntu 20.04

RoadRunnr, 2020-03-21:
## Summary

Does not start on Ubuntu 20.04
```
QQmlApplicationEngine failed to load. component
qrc:/Webcamoid/share/qml/main.qml:214 Type SettingsMenu unavailable
qrc:/Webcamoid/share/qml/SettingsMenu.qml:24 Type Menu unavailable
qrc:/Webcamoid/share/themes/Default/Menu.qml:88 Non-existent attached object
```

## Your Environment

* Webcamoid information: git, master
* Operating System information: Ubuntu 20.04 (dev), full updated, x64
* dependencies installed according to CI script

```
# dpkg -l | grep "qml"
ii  libqt5qml5:amd64                             5.12.5-5                                        amd64        Qt 5 QML module
ii  qml-module-qt-labs-folderlistmodel:amd64     5.12.5-5                                        amd64        Qt 5 folderlistmodel QML module
ii  qml-module-qt-labs-settings:amd64            5.12.5-5                                        amd64        Qt 5 settings QML module
ii  qml-module-qtgraphicaleffects:amd64          5.12.5-2                                        amd64        Qt 5 Graphical Effects module
ii  qml-module-qtqml-models2:amd64               5.12.5-5                                        amd64        Qt 5 Models2 QML module
ii  qml-module-qtquick-controls:amd64            5.12.5-1                                        amd64        Qt 5 Quick Controls QML module
ii  qml-module-qtquick-controls2:amd64           5.12.5+dfsg-2                                   amd64        Qt 5 Qt Quick Controls 2 QML module
ii  qml-module-qtquick-dialogs:amd64             5.12.5-1                                        amd64        Qt 5 Dialogs QML module
ii  qml-module-qtquick-extras:amd64              5.12.5-1                                        amd64        Qt 5 Quick Extras QML module
ii  qml-module-qtquick-layouts:amd64             5.12.5-5                                        amd64        Qt 5 Quick Layouts QML module
ii  qml-module-qtquick-privatewidgets:amd64      5.12.5-1                                        amd64        Qt 5 Private Widgets QML module
ii  qml-module-qtquick-templates2:amd64          5.12.5+dfsg-2                                   amd64        Qt 5 Qt Quick Templates 2 QML module
ii  qml-module-qtquick-window2:amd64             5.12.5-5                                        amd64        Qt 5 window 2 QML module
ii  qml-module-qtquick2:amd64                    5.12.5-5                                        amd64        Qt 5 Qt Quick 2 QML module
ii  qt5-qmltooling-plugins:amd64                 5.12.5-5                                        amd64        Qt 5 qmltooling plugins

```

hipersayanX, 2020-03-22:
Create a **test.qml** file with the following content:

```
import QtQuick 2.12
import QtQuick.Controls 2.5

ApplicationWindow {
    width: 400
    height: 300
    visible: true

    Button {
        text: "Open menu"
        onClicked: contextMenu.popup()
    }

    Menu {
        id: contextMenu
        MenuItem { text: "Cut" }
        MenuItem { text: "Copy" }
        MenuItem { text: "Paste" }
    }
}
```

and run it with ```qmlscene-qt5 test.qml```, does it works?

phord, 2020-03-23:
I'm having this same problem all the sudden.  Webcamoid _was_ working fine, but after a hang in video effects settings, it won't start again.  I see the same errors as OP during load on the console.

I tried removing my local webcamoid config, assuming something was corrupted there, but it didn't help.

I'm on Ubuntu 19.10.

I don't have qmlscene-qt5, but I have qmlscene.  I tried the test code from @hipersayanX and I get this:

    $ qmlscene test.qml 
    qmlscene: could not find a Qt installation of ''

viking, 2020-03-24:
This is an issue for me in Fedora 31. The test.qml file above does work for me but I'm seeing the same messages as the original poster when running Webcamoid.

hipersayanX, 2020-03-25:
@phord install ```qmlscene``` package and run with ```qmlscene test.qml```.
@viking let's clarify this, you see this message:

```
...Menu.qml:88 Non-existent attached object
```

when running the test, right?

viking, 2020-03-25:
The test works, and there are no messages at all in the console. However, I still see the messages when running a freshly compiled and installed webcamoid:

```
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:214 Type SettingsMenu unavailable
qrc:/Webcamoid/share/qml/SettingsMenu.qml:24 Type Menu unavailable
qrc:/Webcamoid/share/themes/Default/Menu.qml:88 Non-existent attached object
```

hipersayanX, 2020-03-25:
~~The problem is, that the message is pointing [here](https://github.com/webcamoid/webcamoid/blob/445bc3e6ef2c3f0dc36b334bb39ec5572239ae00/StandAlone/share/themes/Default/Menu.qml#L88), and when you check the [official implementation](https://github.com/qt/qtquickcontrols2/blob/ed92154c712b0e54caa5baa72d832d7f8286307f/src/imports/controls/material/Menu.qml#L81) in Qt is the exact same.~~

Try changing

```
import QtQuick.Controls 2.5
import QtQuick.Templates 2.5 as T
```

to

```
import QtQuick.Controls 2.12
import QtQuick.Templates 2.12 as T
```

Does it fix the problem?

viking, 2020-03-25:
I changed it everywhere I could find in the git repository, recompiled and installed, and running it still shows the error messages.

viking, 2020-03-25:
Here's the diff: [patch.txt](https://github.com/webcamoid/webcamoid/files/4383174/patch.txt)

phord, 2020-03-27:
> @phord install `qmlscene` package and run with `qmlscene test.qml`.
I do have qlmscene, and it reports the error I showed in my comment.

I have a new datapoint, though:  another application now fails when it tries to use Qt libraries.  I suspect this is caused by the same issue, so there's probably some Qt config problem on my system.

phord, 2020-03-28:
I tried the replacing the 2.5 with 2.12.  No change.

I installed the akvcam driver.  No change

    $ webcamoid 
    libv4l2: error getting pixformat: Invalid argument
    QQmlApplicationEngine failed to load component
    qrc:/Webcamoid/share/qml/main.qml:214 Type SettingsMenu unavailable
    qrc:/Webcamoid/share/qml/SettingsMenu.qml:24 Type Menu unavailable
    qrc:/Webcamoid/share/themes/Default/Menu.qml:88 Non-existent attached object

fiffeek, 2020-04-01:
I am having the same issue on `bionic` : `Description:	Ubuntu 18.04.3 LTS Release:	18.04` with qt version `5.12.7`,  anyone tried downgrading to `5.12.0` or so?

koubeba, 2020-04-01:
I am having the same issue with qt 5.12.0

lucag73, 2020-04-10:
If I just comment line 88 in 
./StandAlone/share/themes/Default/Menu.qml
then webcamoid starts; I do not know about missing functionalities, though.

hipersayanX, 2020-04-12:
Fixed this in the master, may work now.

closed 2020-04-12 by hipersayanX

viking, 2020-04-13:
webcamoid runs now, but none of the icons at the bottom have any images.
[image]

hipersayanX, 2020-04-16:
@viking open a new issue for that, will check later.

hipersayanX, 2020-04-17:
@viking try installing **qt5-image-formats-plugins**.