webcamoid/webcamoid #236: Downloaded app doesn't work at all (no windows launched) on MacOS 10.15.3

ajmas, 2020-03-23:
# README

I did look at issue #83, but that doesn't seem to help.

# Reporting a problem?

## Summary

The application launched, asks for permission to access video camera, but then never gets beyond that point. No windows or menus are displayed.

## Current Behavior

The application launched, asks for permission to access video camera, but then never gets beyond that point. No windows or menus are displayed.

## Expected Behavior

How it should have work?

## Steps to Reproduce

1. Download non-portable version (did try that one too)
2. Launch installed
3. Go to system preferences to allow it be installed
4. In installer specify /Applications as location
5. Once installed, launch

I did try launch it from the CLI as `/Applications/webcamoid/webcamoid.app/Contents/MacOS/webcamoid`, which gave me some logging output.

## Suggestions and tips

Not sure at this time.

## Your Environment

* Webcamoid information: (8.7.1) (Mach-O 64-bit executable x86_64)
* Operating System information: (macOS) (10.15.3) (xnu-6153.81.5~1/RELEASE_X86_64 x86_64)
* Any other useful information: 
  * Log: [webcamoid-log.txt](https://github.com/webcamoid/webcamoid/files/4367060/webcamoid-log.txt)
# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questions and other matters?

Go ahead!

ajmas, 2020-03-23:
Trying now the window comes up, but I am not sure what changed. Will see if I can reproduce why it wasn't working.

I did have an external webcam connected. 

BTW this is on an 2019 16" MacBook Pro.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX