webcamoid/webcamoid #237: ERROR: streaming an image (or video) to a virtual cam on Mac

andy2301, 2020-03-23:
I'm trying to write a wrapper program with IpcBridge to stream an image (ultimate goal is video but starting with image) to a virtual cam "/akvcam/video0", but failed. Any help is appreciated!

These are my steps:
1. compiled the latest code on Mac OS (10.14) with Qt Creator, and started webcamoid.
2. created a virtual cam "/akvcam/video0" with description of "vcam1".
3. compiled and linked the following code with IpcBridge, and ran it but ended up with a malloc error. (I was following the guide from @hipersayanX in Issue #201 )

**Code**:
```
#include <CoreFoundation/CFRunLoop.h>
#include <xpc/xpc.h>
#include <xpc/connection.h>

#include "ipcbridge.h"
#include "VCamUtils/src/image/videoformat.h"
#include "VCamUtils/src/image/videoframe.h"
#include "VCamUtils/src/logger/logger.h"

int main(int argc, char **argv)
{
    AkVCam::IpcBridge ipcb;
    ipcb.connectService(false);
    auto devices = ipcb.listDevices();

    AkLoggerLog("loading frame ...");
    AkVCam::VideoFrame frame1("/app/video/webcamoid/libAvKys/Plugins/VirtualCamera/share/TestFrame/TestFrame1.bmp");
    AkLoggerLog("to write a frame ...");
    ipcb.write("/akvcam/video0", frame1);
    ipcb.disconnectService();
    
    return EXIT_SUCCESS;
}
```

**ERROR**:
```
[2020-03-23 00:29:16.217, 0x113f045c0]  IpcBridge::IpcBridge()
[2020-03-23 00:29:16.218, 0x113f045c0]  IpcBridge::connectService()
[2020-03-23 00:29:16.218, 0x113f045c0]  IpcBridge::registerPeer()
[2020-03-23 00:29:16.220, 0x113f045c0]  SUCCESSFUL
[2020-03-23 00:29:16.220, 0x113f045c0]  IpcBridge::listDevices()
[2020-03-23 00:29:16.220, 0x113f045c0]  Devices:
[2020-03-23 00:29:16.220, 0x113f045c0]      /akvcam/video0
[2020-03-23 00:29:16.221, 0x113f045c0]  loading frame ...
vcam(15061,0x113f045c0) malloc: can't allocate region
*** mach_vm_map(size=2305843009212772352) failed (error code=3)
vcam(15061,0x113f045c0) malloc: *** set a breakpoint in malloc_error_break to debug
libc++abi.dylib: terminating with uncaught exception of type std::bad_alloc: std::bad_alloc
Abort trap: 6
```

hipersayanX, 2020-03-25:
You must select at least one of the devices returned by ipcb.listDevices(), because the device id is not warranted, then [start the device](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/virtualcameraelement.cpp#L528), and keep sending the frame over and over with ipcb.write() otherwise it will show only the default screen.

andy2301, 2020-03-25:
Thank you, @hipersayanX. 

I was able to make it work yesterday (frame sent is reflected in the viewer of the virtual cam). There were 2 problems in my code: 
1. I didn't do "deviceStart" as you suggested; 
2.I reduced the size of the bmp file (previously 1.2M) and it avoided the malloc issue. 

Now I'm facing a new problem: frame rate is super low at 1.6 FPS (as you can see from the log, it takes ~0.6 sec to send a frame).
```
[2020-03-25 00:08:59.852, 0x112ba65c0]  IpcBridge::write()
[2020-03-25 00:09:00.454, 0x112ba65c0]  IpcBridge::write()
[2020-03-25 00:09:01.35, 0x112ba65c0]  IpcBridge::write()
[2020-03-25 00:09:01.623, 0x112ba65c0]  IpcBridge::write()
```

How can I optimize for higher FPS? A couple of thoughts:
1) I noticed that the `IpcBridge::write()` is sending the frame in sync mode with `xpc_connection_send_message_with_reply_sync`. Would async mode help? 

2) is there any other approach that's better for streaming video with at least >20 FPS? How about the approach used in #85?

hipersayanX, 2020-03-25:
>How can I optimize for higher FPS?

That's a REALLY good question and will appreciate a lot suggestions for this. 

>I noticed that the IpcBridge::write() is sending the frame in sync mode with xpc_connection_send_message_with_reply_sync. Would async mode help?

It was async in the past, and worked faster, but then it caused a memory grow #196.

>is there any other approach that's better for streaming video with at least >20 FPS? How about the approach used in #85?

#85 is unrelated, it was about playing a video stream.

andy2301, 2020-03-25:
> #85 is unrelated, it was about playing a video stream.

My original intention is to stream video, it’s just that I wanted to start from frame first. How does streaming video with the AKElement pipeline approach avoid the delay and low frame rate issue? Is it not using sync mode?

hipersayanX, 2020-03-25:
Is the exact same implementation, the difference could be in the frame and format size, if you send a high resolution frame it takes longer to send and render in the target.

hipersayanX, 2021-02-16:
Check [akvirtualcamera](https://github.com/webcamoid/akvirtualcamera/issues/3) instead, not linking required.

closed 2021-02-16 by hipersayanX