webcamoid/webcamoid #242: webcamoid-portable-8.7.1-win64 version starts 'not responding' and never recovers

paulvirtuel, 2020-03-24:
# README

# Reporting a problem?
yes

## Summary
webcamoid-portable-8.7.1-win64 version starts 'not responding' and never recovers

## Current Behavior
I have waited 30 minutes and it still shows 'not responding'

## Expected Behavior
When using VLC media player, I can select the same web cam and it works.

## Steps to Reproduce
Nothing special, after installing it, start it

## Suggestions and tips

## Your Environment
webcamoid-portable-8.7.1-win64 version
Windows 8.1

# Want a new feature?

# Questions and other matters?

gijbelsy, 2020-03-26:
Same issue here, only I haven't configured the webcam yet, so no virtual cam's are available

Aquind, 2020-04-15:
I just installed Webcamoid and I also get the (Not Responding) issue upon launch.  I have tried rebooting, reinstalling, uninstalling rebooting then installing again.  I have also left it running for hours with still no response.  There are no driver issues and multiple other programs use the webcam fine.

Is there any other info I can provide to help troubleshoot this?  Any ideas?

[image]

hipersayanX, 2020-04-16:
@Aquind Can you give me the video information from **TANDBERG Video**, for example from VLC or similar? is that a streamed video or a local file?

[image]

Also try [Reset configs and starting fresh](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh)

paulvirtuel, 2020-09-01:
1- The application does this even without any webcam connected.

2- I connected a web cam and got this from VLC:
Stream 0
   Type: Video
   Codec: Packed YUV 4:2:2, Y:U:Y:V (YUY2)
   Resolution: 1280x720
   Display resolution: 1280x720
   Frame rate: 10
Stream 1
   Type: Audio
   Codec: PCM S16 LE (araw)
   Channels: Stereo
   Sample rate: 44100 Hz
   Bits per sample: 16

paulvirtuel, 2020-09-22:
I am trying to understand why this issue is closed pointing to #345 since #345 is also pointing to this issue and closed as duplicate.

hipersayanX, 2020-09-22:
@paulvirtuel This issue is actually open, check bellow the title, #345 is pointing to this issue as reference ([check](https://guides.github.com/features/issues/) this and [this](https://docs.github.com/en/github/writing-on-github/autolinked-references-and-urls)) and closed because the similarities with this issue. Please, take some time to learn how github works.

>I should create a new issue with the same information since the issue is still present?

Don't try to do that, that's basically spamming and I will ban you and anyone doing that, if there was a problem with current version that you believe that was not actually solved just reopen it.

hipersayanX, 2020-09-22:
And I forgot to mention, I've updated the daily build for Windows these days, check if you still have this problem.

paulvirtuel, 2020-09-23:
Thanks for your response.

I just tried the daily build and the 'not responding' is not present at startup but appears as soon as I select the web cam source 'USB 2.0 Camera'.

From VLC it works.  Here is the codec info shown by VLC:
Stream 0
Type: Video
Codec: Packed YUV 4:2:2, Y:U:Y:V (YUY2)
Resolution: 1280x960
Display resolution: 1280x960
Frame rate: 30.000030

Stream 1
Type: Audio
Codec: PCM S16 LE (araw)
Channels: Stereo
Sample rate: 44100 Hz
Bits per sample: 16

hipersayanX, 2020-09-23:
@paulvirtuel I can't reach to any conclusion with that information, Webcamoid handles fine that format in other webcams, so I think the problem comes from somewhere else. Could it be that you have installed the virtual camera? Did you tried [resetting the configurations](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh#windows)?

Aquind, 2020-09-27:
sorry for the delay.  Actually I am getting this from the codec information in VLC (The video works fine in VLC) 
[image]

Aquind, 2020-09-27:
BTW, the daily build also freezes but instead of freezing on a white window it freezes with the following:
[image]

hipersayanX, 2020-10-03:
@Aquind again, I can't reach to any conclusion with this information https://github.com/webcamoid/webcamoid/issues/242#issuecomment-699569116, Webcamoid works fine with those formats. It should be something else.

>BTW, the daily build also freezes but instead of freezing on a white window it freezes with the following:

If you you go to _Video/Sources_, what formats does it show? Do you have the virtual camera enabled in _Video/Outputs_?

cjbarth, 2021-10-29:
I am getting this problem to with the latest daily builds of Win32 and Win64 both the portable and installed versions. Webcamoid starts, sometimes turns on the camera and shows a single frame, and then reports (Not Responding) in the title bar and never does anything. I eventually close it and it reports the problem to Microsoft. What debugging or logging steps can I take to help get to the bottom of this?

hipersayanX, 2021-10-30:
@cjbarth beside using gdb//lldb, there are not much options for debugging right now. Improving error logging is one of my pending tasks.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX