webcamoid/webcamoid #244: [featuring] Auto crop cameraman

mpvphd, 2020-03-25:
### **Git-Hub**
Hi @hipersayanX  hi everyone. I have been working with firends on this project :
https://github.com/lguilmard/camtrackerwebrtcstreaming/

### **TOPIC**
Have ever experienced making video call with people that dont know how to be in the camera field ? The project we are working on is already able to focus on people in the field of view. The principal problem is : how to make this cross platform compatible ?

### **ISSUE**
Is there a way to pipe video feed into webcamoid in command line ? Actualy I am trying to use it **on mac OSX** like this

- creat vCam with webcamoid 
- update png file in tmpfs RAM mounted DIR with camtrackerwebrtcstreaming 
- make vcam point on that png 

Unfortunatly this solution is highly unstable ...

### **QUESTION**
So : is ther a way to make camtrackerwebrtcstreaming and webcamoid work together in a more reliable way ?

### **thanks !**

mpvphd, 2020-03-25:
ok sorry I found this : https://github.com/webcamoid/webcamoid/issues/201

seems to be recurrent issue

hipersayanX, 2020-03-25:
Closing as it seems solved for you.

closed 2020-03-25 by hipersayanX