webcamoid/webcamoid #245: Mac os Catalina Segmentation fault because of rg.webcamoid.cmio.DAL.VirtualCamera

vetermanve, 2020-03-25:
## Summary

I've installed this software and my mac starts unexpectedly log me out, found the crash report and it points to virtual camera plugin 

## Current Behavior

Just unexpected crashes after installation and creating virtual camera. Crashed continue appear after virtual camera was removed, and possible was before, but i can't say for sure

## Expected Behavior

Installation of this software of whatever configuration inside should not crash mac os =)

## Steps to Reproduce

Possible 
1. Install last version of software on last version of mac os Catalina
2. Create virtual camera
3. At some point of time, even not opening this soft mac os will crash, but not fully, just log you out as result

## Suggestions and tips
[BunchOfCrashLogs.zip](https://github.com/webcamoid/webcamoid/files/4381027/BunchOfCrashLogs.zip)

Now i'm removed this software, no ideas how i can fix it other way

## Your Environment

* Webcamoid information: webcamoid-8.7.1.dmg
* Operating System information: Mac os Catalina 10.15.3 on MacBook Pro (15-inch, 2018)
* Any other useful information: top of crash log

`Process:               PerfPowerServices [8485]
Path:                  /usr/libexec/PerfPowerServices
Identifier:            PerfPowerServices
Version:               94.81.1
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
Responsible:           PerfPowerServices [8485]
User ID:               0

PlugIn Path:             /Library/CoreMediaIO/*/AkVirtualCamera.plugin/Contents/MacOS/AkVirtualCamera
PlugIn Identifier:       org.webcamoid.cmio.DAL.VirtualCamera
PlugIn Version:          8.7.1 (8.7.1)

Date/Time:             2020-03-25 16:02:40.871 +0300
OS Version:            Mac OS X 10.15.3 (19D76)
Report Version:        12
Bridge OS Version:     4.2 (17P3050)
Anonymous UUID:        BBD7E2C4-86E2-BE7A-469C-2F4C6F3FE2A4

Sleep/Wake UUID:       89A11FF8-1AE8-4B14-84BC-B5336B2FDD81

Time Awake Since Boot: 36000 seconds
Time Since Wake:       15000 seconds

System Integrity Protection: disabled

Crashed Thread:        6  Dispatch queue: com.apple.powerlog.workqueue.PLCameraAgent

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [8485]

VM Regions Near 0:
--> 
    __TEXT                 000000010398f000-0000000103990000 [    4K] r-x/r-x SM=COW  /usr/libexec/PerfPowerServices

Application Specific Information:
dyld3 mode
`

MarFre22, 2020-06-15:
This situation also happened to me. I hope some fix, as it makes MacOS unusable.
My solution was to remove everything related to this and try another app.
My issue was initially published here:
https://github.com/daliansky/XiaoMi-Pro-Hackintosh/issues/390#issuecomment-644096425

kokroo, 2020-09-08:
This was crashing my laptop for 9 months and I just figured out this was the problem. Jesus Christ.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX