webcamoid/webcamoid #247: webcamoid 64bit macOS version crashes after adding virtual camera

m0rhadin, 2020-03-26:
Hi Guys, 
recently I installed Webcamoid to play around with some video calls. Unfortunately after I created two virtual camera via options and "adding virtual camera" the program just crashed and won't reopen but crashes everytime. Maybe interesting: I tried to use it in combination with zoom.us. Which I of course closed, then reopened webcamoid and so on.

I thought that it just might add the cameras so that I can choose them as an output and from there on I can add video, effects etc. to the virtual device. Since the software is crashing everytime I try to open it, I am not even able to delete the devices.

I am using the 64bit mac version of webcamoid which you can download from the website. My operating system is macOS Catalina (Version 10.15.2) with an 2,7 GHz Quad-Core Intel Core i7.

> Crashed Thread:        0  Dispatch queue: com.apple.main-thread
> 
> Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
> Exception Codes:       KERN_INVALID_ADDRESS at 0x00000034aebb6b38
> Exception Note:        EXC_CORPSE_NOTIFY
> 
> Termination Signal:    Segmentation fault: 11
> Termination Reason:    Namespace SIGNAL, Code 0xb
> Terminating Process:   exc handler [76196]
> 
> Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
> 0   org.qt-project.QtQml          	0x000000010afe0925 0x10ad8b000 + 2447653
> 1   org.qt-project.QtQml          	0x000000010aea11e7 QV4::QObjectWrapper::getProperty(QV4::ExecutionEngine*, QObject*, QQmlPropertyData*) + 231
2   org.qt-project.QtQml          	0x000000010ae86ef1 QV4::QQmlContextWrapper::lookupScopeObjectProperty(QV4::Lookup*, QV4::ExecutionEngine*, QV4::Value*) + 529
3   ???                           	0x0000000146409eb8 0 + 5473607352
4   org.qt-project.QtQml          	0x000000010aebde8c 0x10ad8b000 + 1257100
5   org.qt-project.QtQml          	0x000000010ae663b6 QV4::Function::call(QV4::Value const*, QV4::Value const*, int, QV4::ExecutionContext const*) + 342
6   org.qt-project.QtQml          	0x000000010afe00f5 QQmlJavaScriptExpression::evaluate(QV4::CallData*, bool*) + 597
7   org.qt-project.QtQml          	0x000000010afe8157 0x10ad8b000 + 2478423
8   org.qt-project.QtQml          	0x000000010afe627e QQmlBinding::update(QFlags<QQmlPropertyData::WriteFlag>) + 318
9   org.qt-project.QtQml          	0x000000010aff9856 QQmlObjectCreator::finalize(QQmlInstantiationInterrupt&) + 326
10  org.qt-project.QtQml          	0x000000010af80a1f 0x10ad8b000 + 2054687
11  org.qt-project.QtQml          	0x000000010af807ff QQmlEnginePrivate::incubate(QQmlIncubator&, QQmlContextData*) + 415
12  org.qt-project.QtQml          	0x000000010b053dc7 0x10ad8b000 + 2919879
13  org.qt-project.QtQuick        	0x0000000109ff4a33 0x109df0000 + 2116147
14  org.qt-project.QtQuick        	0x000000010a007e25 0x109df0000 + 2194981
15  org.qt-project.QtQuick        	0x0000000109ff5cd4 0x109df0000 + 2120916
16  org.qt-project.QtQuick        	0x0000000109ff1bb4 0x109df0000 + 2104244
17  org.qt-project.QtQuick        	0x0000000109fedf8a QQuickItemView::setCurrentIndex(int) + 122
18  org.qt-project.QtQuick        	0x0000000109ff9997 QQuickItemView::qt_metacall(QMetaObject::Call, int, void**) + 71
19  org.qt-project.QtQuick        	0x000000010a0096c8 QQuickListView::qt_metacall(QMetaObject::Call, int, void**) + 24
20  org.qt-project.QtQml          	0x000000010aea313c QV4::QObjectWrapper::setProperty(QV4::ExecutionEngine*, QObject*, QQmlPropertyData*, QV4::Value const&) + 1644
21  org.qt-project.QtQml          	0x000000010aea496e QV4::QObjectWrapper::virtualPut(QV4::Managed*, QV4::PropertyKey, QV4::Value const&, QV4::Value*) + 414
22  org.qt-project.QtQml          	0x000000010ae78ae3 QV4::Object::virtualResolveLookupSetter(QV4::Object*, QV4::ExecutionEngine*, QV4::Lookup*, QV4::Value const&) + 355
23  org.qt-project.QtQml          	0x000000010aebe9fb 0x10ad8b000 + 1260027
24  org.qt-project.QtQml          	0x000000010aebde5b 0x10ad8b000 + 1257051
25  org.qt-project.QtQml          	0x000000010ae6943f 0x10ad8b000 + 910399
26  org.qt-project.QtQml          	0x000000010aebefde 0x10ad8b000 + 1261534
27  org.qt-project.QtQml          	0x000000010aebde5b 0x10ad8b000 + 1257051
28  org.qt-project.QtQml          	0x000000010ae663b6 QV4::Function::call(QV4::Value const*, QV4::Value const*, int, QV4::ExecutionContext const*) + 342
29  org.qt-project.QtQml          	0x000000010afe00f5 QQmlJavaScriptExpression::evaluate(QV4::CallData*, bool*) + 597
30  org.qt-project.QtQml          	0x000000010af8707a QQmlBoundSignalExpression::evaluate(void**) + 1162
31  org.qt-project.QtQml          	0x000000010af8784b 0x10ad8b000 + 2082891
32  org.qt-project.QtQml          	0x000000010afc6c45 QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) + 581
33  org.qt-project.QtCore         	0x000000010b52392f QMetaObject::activate(QObject*, int, int, void**) + 127
34  org.qt-project.QtQml          	0x000000010aff9c4d QQmlObjectCreator::finalize(QQmlInstantiationInterrupt&) + 1341
35  org.qt-project.QtQml          	0x000000010af7d9e9 QQmlComponentPrivate::complete(QQmlEnginePrivate*, QQmlComponentPrivate::ConstructionState*) + 73
36  org.qt-project.QtQml          	0x000000010af7b8b6 QQmlComponentPrivate::completeCreate() + 102
37  org.qt-project.QtQml          	0x000000010af7d2a9 QQmlComponent::create(QQmlContext*) + 121
38  org.qt-project.QtQml          	0x000000010afea407 QQmlApplicationEnginePrivate::finishLoad(QQmlComponent*) + 87
39  org.qt-project.QtQml          	0x000000010afeaa92 QQmlApplicationEngine::load(QUrl const&) + 34
40  com.webcamoidprj.webcamoid    	0x0000000109bfe9f0 0x109bf5000 + 39408
41  com.webcamoidprj.webcamoid    	0x0000000109bfa907 0x109bf5000 + 22791
42  libdyld.dylib                 	0x00007fff6f0387fd start + 1

driq, 2020-03-30:
I have the same issue.

**Steps to reproduce**
1. Install Webcamoid 
   * I downloaded https://github.com/webcamoid/webcamoid/releases/download/8.7.1/webcamoid-portable-8.7.1-x86_64.dmg and dragged the webcamoid app to the applications folder
2. Open Webcamoid
3. In Webcamoid, go to the 'settings' tab
4. Check the 'virtual camera' box
5. Wait for the application  to crash

Crash report:
```Process:               webcamoid [1593]
Path:                  /Applications/webcamoid.app/Contents/MacOS/webcamoid
Identifier:            com.webcamoidprj.webcamoid
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           webcamoid [1593]
User ID:               504

Date/Time:             2020-03-30 11:06:26.043 +0200
OS Version:            Mac OS X 10.15.4 (19E266)
Report Version:        12
Anonymous UUID:        F20FC9A2-21EB-7944-4DE8-15327AEA6142

Time Awake Since Boot: 1200 seconds

System Integrity Protection: enabled

Crashed Thread:        14  Thread (pooled)

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000001296eb000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [1593]

VM Regions Near 0x1296eb000:
    MALLOC_LARGE           0000000129692000-00000001296eb000 [  356K] rw-/rwx SM=PRV  
--> 
    CG image               00000001298a5000-00000001298a9000 [   16K] rw-/rwx SM=PRV  

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff69066dfa mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff69067170 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff2efde0b5 __CFRunLoopServiceMachPort + 247
3   com.apple.CoreFoundation      	0x00007fff2efdcb82 __CFRunLoopRun + 1319
4   com.apple.CoreFoundation      	0x00007fff2efdbffe CFRunLoopRunSpecific + 462
5   com.apple.HIToolbox           	0x00007fff2dc0fabd RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00007fff2dc0f7d5 ReceiveNextEventCommon + 584
7   com.apple.HIToolbox           	0x00007fff2dc0f579 _BlockUntilNextEventMatchingListInModeWithFilter + 64
8   com.apple.AppKit              	0x00007fff2c25ac99 _DPSNextEvent + 883
9   com.apple.AppKit              	0x00007fff2c2594e0 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
10  com.apple.AppKit              	0x00007fff2c24b1ee -[NSApplication run] + 658
11  libqcocoa.dylib               	0x0000000104ac24f3 0x104a8e000 + 214259
12  org.qt-project.QtCore         	0x0000000102a5913f QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 431
13  org.qt-project.QtCore         	0x0000000102a5e2d2 QCoreApplication::exec() + 130
14  com.webcamoidprj.webcamoid    	0x000000010114b90c 0x101146000 + 22796
15  libdyld.dylib                 	0x00007fff68f25cc9 start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff69125b68 start_wqthread + 0

Thread 2:: QQmlThread
0   libsystem_kernel.dylib        	0x00007fff6906d3d6 poll + 10
1   org.qt-project.QtCore         	0x0000000102ab41b0 qt_safe_poll(pollfd*, unsigned int, timespec const*) + 608
2   org.qt-project.QtCore         	0x0000000102ab5a0a QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 858
3   org.qt-project.QtCore         	0x0000000102a5913f QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 431
[... 598 lines cut ...]
JS VM Gigacage                    4096K        1 
JS VM Isolated Heap               6400K        5 
Kernel Alloc Once                    8K        1 
MALLOC                           254.6M      193 
MALLOC guard page                   32K        7 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
Memory Tag 242                      12K        1 
STACK GUARD                       56.1M       15 
Stack                             15.1M       15 
VM_ALLOCATE                       1032K       63 
__DATA                            40.6M      426 
__DATA_CONST                       272K        7 
__FONT_DATA                          4K        1 
__LINKEDIT                       399.7M      114 
__OBJC_RO                         32.2M        1 
__OBJC_RW                         1888K        1 
__TEXT                           274.7M      403 
__UNICODE                          564K        1 
mapped file                      494.8M      337 
shared memory                      632K       14 
===========                     =======  ======= 
TOTAL                              1.9G     1730 
TOTAL, minus reserved VM space     1.6G     1730 

```

qdtjvszxc, 2020-04-04:
You can just stop camera before adding or deleting any new camera or source.

pavlobu, 2020-05-02:
I confirm, this also happens on my system.
MacOS Catalina 10.15.3
MacBook Pro (Retina, 15-inch, Mid 2015)

m0rhadin, 2020-05-03:
Any new findings regarding this issue? If you need more information or some screen capture of the apps behaviour just let me know.

StuffAndyMakes, 2020-05-05:
I installed the latest version today, ran it for a few minutes, then macOS started popping up the "Force Quit" dialog, telling me system memory was low, asking me to force quit some apps. This was on a 2013 Mac Pro (Late 2013) with 64GB of RAM. First time in over 6 years I've seen that box pop up. Pasting the computer specs and the crash log from AkVCamAssistant.

macOS Catalina 10.15.4 (19E287)
3.5 GHz 6-Core Intel Xeon E5
64 GB 1866 Mhz DDR3
1 TB internal SSD (about 50% available)

```
Process:               AkVCamAssistant [40022]
Path:                  /Library/CoreMediaIO/*/AkVirtualCamera.plugin/Contents/Resources/AkVCamAssistant
Identifier:            AkVCamAssistant
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           AkVCamAssistant [40022]
User ID:               501

Date/Time:             2020-05-04 17:16:10.384 -0700
OS Version:            Mac OS X 10.15.4 (19E287)
Report Version:        12
Anonymous UUID:        1226D623-89D2-1D33-9983-3C9E5AB910CE

Time Awake Since Boot: 890000 seconds

System Integrity Protection: enabled

Crashed Thread:        1  Dispatch queue: com.apple.root.default-qos.overcommit

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [40022]

VM Regions Near 0:
--> 
    __TEXT                 000000010ea73000-000000010ea8b000 [   96K] r-x/rwx SM=COW  /Library/CoreMediaIO/*/AkVirtualCamera.plugin/Contents/Resources/AkVCamAssistant

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff72ff3dfa mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff72ff4170 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff38eb9bf7 __CFRunLoopServiceMachPort + 247
3   com.apple.CoreFoundation      	0x00007fff38eb86b1 __CFRunLoopRun + 1329
4   com.apple.CoreFoundation      	0x00007fff38eb7b23 CFRunLoopRunSpecific + 466
5   com.apple.CoreFoundation      	0x00007fff38f4020c CFRunLoopRun + 40
6   AkVCamAssistant               	0x000000010ea745c5 0x10ea73000 + 5573
7   libdyld.dylib                 	0x00007fff72eb2cc9 start + 1

Thread 1 Crashed:: Dispatch queue: com.apple.root.default-qos.overcommit
0   libxpc.dylib                  	0x00007fff730f0980 xpc_get_type + 10
1   libxpc.dylib                  	0x00007fff730f9804 xpc_connection_send_message_with_reply_sync + 28
2   AkVCamAssistant               	0x000000010ea74aa6 0x10ea73000 + 6822
3   libxpc.dylib                  	0x00007fff730fb2bc _xpc_connection_call_event_handler + 56
4   libxpc.dylib                  	0x00007fff730fa0fb _xpc_connection_mach_event + 726
5   libdispatch.dylib             	0x00007fff72e596f8 _dispatch_client_callout4 + 9
6   libdispatch.dylib             	0x00007fff72e71fc7 _dispatch_mach_cancel_invoke + 65
7   libdispatch.dylib             	0x00007fff72e6f95c _dispatch_mach_invoke + 1057
8   libdispatch.dylib             	0x00007fff72e68c09 _dispatch_workloop_worker_thread + 596
9   libsystem_pthread.dylib       	0x00007fff730b7a3d _pthread_wqthread + 290
10  libsystem_pthread.dylib       	0x00007fff730b6b77 start_wqthread + 15

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff730b6b68 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff730b6b68 start_wqthread + 0

Thread 1 crashed with X86 Thread State (64-bit):
[... 72 lines cut ...]
ReadOnly portion of Libraries: Total=399.8M resident=0K(0%) swapped_out_or_unallocated=399.8M(100%)
Writable regions: Total=118.1M written=0K(0%) resident=0K(0%) swapped_out=0K(0%) unallocated=118.1M(100%)
 
                                VIRTUAL   REGION 
REGION TYPE                        SIZE    COUNT (non-coalesced) 
===========                     =======  ======= 
Activity Tracing                   256K        1 
Dispatch continuations            24.0M        1 
Kernel Alloc Once                    8K        1 
MALLOC                            84.1M       27 
MALLOC guard page                   24K        5 
STACK GUARD                       56.0M        4 
Stack                             9752K        4 
__DATA                            2050K       46 
__DATA_CONST                        20K        1 
__LINKEDIT                       388.1M        3 
__OBJC_RO                         32.2M        1 
__OBJC_RW                         1892K        2 
__TEXT                            11.7M       45 
__UNICODE                          564K        1 
mapped file                         28K        1 
shared memory                       44K        5 
===========                     =======  ======= 
TOTAL                            610.4M      148 
```

hipersayanX, 2020-05-05:
@m0rhadin just drop any information you have, I'll check it later.

mailinglists35, 2020-12-30:
one more crash report from here.
mac mini late 2014, big sur
to avoid harvesting bots, the password of crash report is the name of the project spelled backwards

https://cryptobin.co/a3j1o9o7

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX