webcamoid/webcamoid #250: Virtual Backgrounds

ck-kc, 2020-03-26:
I would like webcamoid to have a Virtual Background feature (like in Zoom) that works without a green screen. Would it be difficult to implement in webcamoid?

ackbar3, 2020-03-28:
I'm exactly searching for something like that.

hipersayanX, 2020-03-29:
A lot of people ask for this feature.

>Would it be difficult to implement in webcamoid?

Not really, I'll do it when had some time.

egeland, 2020-03-30:
With a lot more people WFH due to COVID-19, I'm sure this feature would be very much appreciated! 😀

doitux, 2020-04-02:
That would be really great. So i could stop using Zoom!

fangfufu, 2020-04-06:
@hipersayanX, is there any ways for us to donate to you, for this specific plugin?

hipersayanX, 2020-04-07:
@fangfufu #267

beppe9000, 2020-04-11:
see also this for background manipulation without relying on chroma key
https://elder.dev/posts/open-source-virtual-background/

utak3r, 2020-04-11:
I'm also going to implement the "multi-chromakey", the key built from many color samples, using ramps and so on. It can give really good results - not as great as using neural networks, but faster. I was already doing this for huge commercial projects. It's just... have to find time for this ;)

beppe9000, 2020-04-11:
great!

fangfufu, 2020-04-12:
@beppe9000, the blogpost you linked contains some mistakes. I fixed them, and created a turnkey solution. It should be fairly easy to follow. 

https://github.com/fangfufu/Linux-Background-Blur-Webcam

beppe9000, 2020-04-12:
nice!

hipersayanX, 2020-04-12:
@beppe9000 @fangfufu

Damn, is that much complex to do that? as complex as using Tensorflow for that?
I was thinking on something much simpler, adapting the code from [ColorReplace](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/ColorReplace/src/colorreplaceelement.cpp), then ask the user to get out of the screen for a second, take a snapshot of the background, and use that reference to replace the background.
Anyway if there are more documentation and examples for this, post it.

beppe9000, 2020-04-12:
> @beppe9000 @fangfufu
>
> Damn, is that much complex to do that? as complex as using Tensorflow for that?
> I was thinking on something much simpler, adapting the code from [...]

i always wondered if such a method can react to changing light conditions or camera noise
I think an alternate approach could be using the openCV motion tracking with a very long memory level

hipersayanX, 2020-04-12:
@beppe9000 Camera noise can be removed with a light de-noise filter and light changes could be fixed normalizing the luma channel, but I'm talking based on my knowledge, I could be wrong.

beppe9000, 2020-04-12:
i hope to try it soon or later then

fangfufu, 2020-04-12:
@hipersayanX, you definitely don't need a convolutional neural network for that. The network in my example can segment out human without learning the background. Not learning the background is not a requirement. 

I only tidied up his blog post because I can't be bothered to write my own code, and I want something working quickly.

fangfufu, 2020-04-12:
@hipersayanX, you probably want to use a segmentation approach rather than colour replace, because the skin's RGB value can be quite close to the background RGB value, in some lighting condition. Certain parts of the skin maybe replaced.

fangfufu, 2020-04-12:
@hipersayanX, finally don't even think about using the bodypix network, because the model is only available in nodejs format... 

So in the code I gathered, it does this: 
Webcam -> Python -> HTTP-> NodeJS->Tensorflow->Mask.

And yes, it spins up two Docker containers, and they communicate with each other via HTTP...

It is really stupid.

hipersayanX, 2020-04-12:
@fangfufu 

>Webcam -> Python -> HTTP-> NodeJS->Tensorflow->Mask.

Damn, and they forgot to include Ruby on Rails, Angular, React and JQuery on that list :rofl:

beppe9000, 2020-04-12:
where is the obligatory mongo database ?!

fangfufu, 2020-04-12:
@hipersayanX 
> Damn, and they forgot to include Ruby on Rails, Angular, React and JQuery on that list

It was not "they", it was a one-man job, done by a Google engineer... He even had time to add a hologram effect to the foreground, but he didn't have time to type up the correct command to spin up the Docker containers... 

I think it says a LOT about Google as a company.

fangfufu, 2020-04-12:
https://www.reddit.com/r/computervision/comments/g043z9/could_anyone_suggest_a_image_segmentation/

https://www.reddit.com/r/MachineLearning/comments/g04daq/d_could_anyone_suggest_a_image_segmentation/

People on Reddit are recommending CNNs... It does make me wonder if people in academia are out of their mind... CNNs are so impractical for a little webcam project...

beppe9000, 2020-04-12:
they like their nets :P
[image]

RazZziel, 2020-04-13:
Apparently it's possible to replicate Benjamin Elder's example in good old C++ and TensorFlow: https://github.com/floe/deepbacksub

fangfufu, 2020-04-13:
I think it would be quite a nightmare if webcamnoid suddenly depend on tensorflow-lite to build. I have had difficulty with getting his code working, 

Bodypix does not seem to available in Python (https://github.com/tensorflow/tfjs/issues/2043).

beppe9000, 2020-04-14:
wouln't make sense to have that implementation as a separate plugin?

fangfufu, 2020-04-14:
I reckon that's what needs to be done. It doesn't make sense to make webcamoid to depend on Tensorflow to build. Or at least set it as an optional plugin.

allo-, 2020-04-15:
Hi.

I created a webcam script for the python bodypix project (here is my [fork](https://github.com/allo-/simple_bodypix_python). In fact the main dependency is `tfjs_graph_converter` and not the rest of the project.

Testing it:

- https://github.com/allo-/simple_bodypix_python/blob/master/fake_webcam_background.py
- https://github.com/allo-/simple_bodypix_python/blob/master/get-model.sh
- all of requirements.txt plus `python-opencv`, `pyyaml` and `tensorflow>=2.0.0` or `tensorflow-gpu>=2.0.0`.
- TF 1.15 should work but you probably need to change in the script `tf.config.experimental.set_memory_growth` to TF 1.x syntax

Then run it:

- Use `get-model.sh` to download a model. Currently I can recommend `bodypix/mobilenet/float/050/model-stride16`
- Create a v4l2loopback device
- Possibly you need to set `v4l2loopback-ctl set-fps 10 /dev/video2` and `v4l2loopback-ctl set-caps any /dev/video2`
- copy config.yaml.example to config.yaml and a jpg image with ratio 1.77 to background.jpg
- possibly change the devices (`video0`, `video2`) in the script
- run the `fake_webcam_background.py` script.

You now should now be able to see your webcam, e.g., by running `mplayer tv://`. With tensorflow-gpu it runs in real-time for me.

allo-, 2020-04-15:
Current Open Problem: Tuning of erode/blur and choosing the best model to get a good segmentation.
Possibly rewriting some code of the nodejs module https://github.com/tensorflow/tfjs-models/tree/master/body-pix/src that is needed for segmentPerson, as it manages somehow (see the [blog post](https://elder.dev/posts/open-source-virtual-background/)) to get a nice shape, while my code still creates a rather blocky shape.

hipersayanX, 2020-04-16:
>I think it would be quite a nightmare if webcamnoid suddenly depend on tensorflow-lite to build.

In the last case I'll strip just the minimum code from Tensorflow (or whatever library), and include inside Webcamoid, as I did with OpenCV and [FaceDetect](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/FaceDetect/src/haar), of course, if the license allow me to do that.

fangfufu, 2020-04-16:
> if the license allow me to do that
Tensorflow / Tensorflow-lite is on Apache license. You should be fine.

nariox, 2020-04-16:
> https://www.reddit.com/r/computervision/comments/g043z9/could_anyone_suggest_a_image_segmentation/ [...]

Academia is about research, so the "cool fancy new" techniques are all the rage. CNNs (and probably RNNs) would be great at detecting people/body vs background, but I agree this is overkill. I wonder if we could implement a simple solution where you take a picture of your background (without anyone) and then let the filter compare it to the current frame. If the difference is smaller than a certain threshold, set the alpha channel to 1.0.

It would probably fail badly if the lighting changes, but I feel it should work well enough for most people and probably be significantly less CPU intensive than any ANN solution. That said, I have had the hardest time using OpenCV, so I can't help much in this regard.

If this works "okay", we could implement minor improvements, like soft thresholding, automatic threshold detection, temporal averaging of the background (would take of slow lighting changes due to sunrise/set) etc.

fangfufu, 2020-04-16:
Personally I think computing the intrinsic image of the background would be easiest approach. It doesn't involve CNN, the algorithm itself is simple enough to be done in OpenCV. 

But it is known that Microsoft Team uses CNN to blur the background.

nariox, 2020-04-16:
But blurring needs much less precise accuracy. Worst that could happen would be a softer edge on the hair etc. My guess is that they are probably running a small CNN in this case.

I mean, bodypix is not just detecting the foreground, but also segmenting into parts, estimating pose, etc. That said, TBH, bodypix segmentation is not all that great (at least from what I see here: https://storage.googleapis.com/tfjs-models/demos/body-pix/index.html)

allo-, 2020-04-16:
@nariox 
> I wonder if we could implement a simple solution where you take a picture of your background (without anyone) and then let the filter compare it to the current frame. If the difference is smaller than a certain threshold, set the alpha channel to 1.0.
Then you will filter out people that do not move enough ;-).

My Bodypix Code is working quite good now (with similar results as the body-pix npm module used in the blog post). I will probably sooner or later create a repository for my script that does not depend on the simple_bodypix_python repo, when you're having interest you will find it then in the list of my public repos.

When you continue here without using ML, I think will unsubscribe this issue in a few posts. When you decide to use bodypix later we can still work together.

nariox, 2020-04-16:
> Then you will filter out people that do not move enough ;-).
As long as you take a picture without any people first, this should work "okay". (Maybe couches or other flexible things might not work either). It is nowhere as good as the ML solution, but it would have much lower CPU usage.

I have tried yours, but for whatever reason it uses all 8 threads of my CPU. I could have limited it to 10fps or something, but it is still a lot of CPU. That said, I don't see why both solutions can't coexist. It's not like any of these will replace traditional chroma-key in any way either. Maybe we can open two issues and continue the discussion over there?

allo-, 2020-04-16:
I do not say that they cannot or should not co-exist. I just think that we currently cannot help each other when we do different projects.

> I have tried yours, but for whatever reason it uses all 8 threads of my CPU.

That's the thing with AI. For me it works quite okay with GPU and mobilenet 0.5/16 and okay on CPU, but without limiting the frame rate it will use 100% CPU.

I only said I may unsubscribe because when I do not have anything to add to the discussion and do not use your solution either, so you know that I will not get the bug mails then.
This was not meant to criticize your project.

fangfufu, 2020-04-19:
Right, I updated my repository, and removed all the Docker related nonsense, and by sampling downsampling the image sent to bodypix, I managed to dramatically improve the performance. 

Check this out: 
https://github.com/fangfufu/Linux-Fake-Background-Webcam

But anyways, I don't really get the woke software developers these days. Everything seems to be more complicated than it needs to be.

fangfufu, 2020-04-19:
@hipersayanX , I just want to say, if you want to steal ideas on running neural network, please go and steal @allo- 's Virtual Background Webcam. It is much neater than mine - s/he managed to run bodypix without Tensorflowjs.

OceanWolf, 2020-04-19:
Hmm, surely the solution exists as much simpler than that; at first I was thinking you can just combine the edge detect and the face detect plugin, but then a cheating method hit me!  Why not take a picture on a delay timer!  That picture gets used as a reference, anything different from that exists as foreground!

1. User tells webcamoid how long they need to get out of shot and presses okay
2. User gets out of shot while webcamoid counts down.
3. Picture gets taken
4. User comes back in and voila!

It exists as a problem if the camera moves or other drastic changes, hence why I describe this method as a "cheat" but maybe that can get dealt with. :)

nariox, 2020-04-20:
@fangfufu and @OceanWolf : christmas is here early (?)
I have (badly) implemented this option. It is NOT production ready, nor it is bug free (outputs lots of garbage), but it is working. CPU usage is still abysmal, but I think it is probably due to the garbage being spit into STDOUT. Haven't had time to modify Readme or documentation yet (next step, but I am procrastinating on my other responsibilities, so I should probably get back to those first) Check out my fork: https://github.com/nariox/Linux-Fake-Background-Webcam
PRs (and forks) are (very) welcome.

@allo- & @fangfufu  : thank you for your efforts. I would much rather use a "set-and-forget-it" option. I have tried more recently and seems like the CPU usage is at a reasonable level.

diegovazquez, 2020-04-22:
Hi everybody, english is not my first language and i'm still learning, sorry for grammar and spelling mistakes.
 I have an implementation for a filter similar to https://elder.dev/posts/open-source-virtual-background/. The ouput is ASCII Art, but the background removal is Tensorflow (No tensorflow JS.) so, you dont need a Docker for the backgorund mask. The model is Deeplab Mobilenetv2.

Git:
https://github.com/diegovazquez/ASCIILinuxVirtualWebCam

Explanation:
https://www.reddit.com/r/linux/comments/g5mnfx/ascii_art_webcam_effect_for_linux/

You can use tensorflow lite for get a quantized model if you needit to run in CPU. 

https://www.tensorflow.org/lite/performance/post_training_quantization

i hope this is useful.

allo-, 2020-04-22:
@diegovazquez Cool effect. The program looks similar to what I am building, just using another network.
Deeplab does not support identification of body parts, does it?

When you mention quantization, do you have a hint what I need to do to work with quantized models? Loading the model like when full floats are used does not work, so I probably need some preprocessing for the input and postprocessing for the output when I use a quantized model?

diegovazquez, 2020-04-22:
Deeplab does not support identification of body parts.
In mi experience for quantized models you dont need change the input or ouput of the model (for android i use the same code, only change the model).

I recommend this link whit a pretrained and quantized model: 
https://www.tensorflow.org/lite/models/segmentation/overview

allo-, 2020-04-22:
My current code does not work when I use the body-pix quant2 or quant1 models, but it seems to run with an acceptable speed on CPU.

I think I will stick to body-pix, because I have a lot of ideas to implement filters that need head tracking and more.

Svetlana-T, 2020-05-04:
Hi all. Impressed that this task indeed is possible to implement using either python or cpp. Is there a webcamoid plugin for the background blur or replacement, so that it is easier to install? I'm using Debian GNU/Linux.

hipersayanX, 2020-05-05:
>Is there a webcamoid plugin for the background blur or replacement, so that it is easier to install?

Webcamoid has a video effect with the same name, but it applies to the whole picture.

azumukupoe, 2020-05-05:
Can you do the opposite of face detection?

hipersayanX, 2020-05-05:
@azumukupoe what do you mean with "opposite"?

azumukupoe, 2020-05-05:
@hipersayanX 
like blurring the undetected area

hipersayanX, 2020-05-05:
@azumukupoe but then you need a haar cascade that detects the human body in every possible position, form and size. There ways of implementing it that are much better that using haar cascades.

nariox, 2020-05-05:
Hi @azumukupoe, I can suggest something that is not using webcamoid, but
you can combine it with webcamoid:
https://github.com/nariox/Linux-Simple-Fake-Background-Webcam
I have a "blur.py" script. At my place it works well enough (better than
Skype's IMO). It works better if your background color is sufficiently
different from your skin color, but even if not, it is "okay".

Pedro Nariyoshi

jiangjianping, 2020-05-15:
@all,
Is it possible to use bodypix get backgroud/forground pictures automatically(not running always) and feed to  @nariox's CV scrpit which consume's less cpu?

allo-, 2020-05-15:
@jiangjianping I am not sure what you want to achieve.

Of course you can run the background / foreground detection one time and then re-use the mask. But when the person moves, the mask is wrong and the person may be (partially)  outside of the mask.

You could also try to run the detection only every n frames, but on the other hand I even calculate the mean over 2-3 frames to avoid flickering at the boundary of the mask. When you are skipping frames you will increase the artifacts from an unperfect segmentation.

jiangjianping, 2020-05-15:
@allo-,

Thanks for your great response. I hope the virtual backgroud can be done in normal performance PCs.   The bodypix is great but cost too much.

nariox, 2020-05-15:
@jiangjianping  : in my scripts, the mask gets recomputed every frame, just like in @fangfufu's and @allo-'s. The difference is that I am using more traditional background/foreground extraction algorithms. 

In the beginning I did it "by hand" by computing the background "offline" and then comparing it to the current frame, the areas with "large differences" were the detected as foreground. My latest commit uses the built-in OpenCV foreground extraction algorithms. The issue is that they are designed for foreground that is constantly moving, so I had to "hack" it not to update the background estimation unless the user presses a button.

This works well enough for blurring the background, but the virtual background requires a bit more post-processing and adjustment (it works fine if the background is sufficiently different from you (my walls are "skin toned" unfortunately, so I have to set up a sheet on the wall to increase contrast at night, during the day, it works okay).

The issue seems to be that bodypix is doing much more computation than simply foreground extraction (people "shape" detection, limb segmentation, etc). I have tried to improve the detection with contours and cv.connectedcomponents, but I couldn't make it fast enough (but probably because I am missing out something important). The detection is pretty good though.

Right now, the script still uses about 100% of one of my cores (i5-8350U), so it is not necessarily lightweight. Probably, an improvement I should try would be to use the same approach @fangfufu used and scale the image down before applying segmentation to reduce CPU usage.

allo-, 2020-05-15:
The body-pix programs (i.e. the javascript library that @fangfufu is using and the reimplementation in python that I am using) have an `internal_resolution` parameter that scales the image. You need to experiment a bit, because it is set to the next valid resolution, such that some values create a foreground image that has an offset.

And when you have an Nvidia card, use CUDA. You probably need to download libnvinfer in addition to the usual cuda libs. I tested the python program with CUDA 10, but it should probably work with older versions as well.

jiangjianping, 2020-05-16:
@nariox 
Thanks. Have you had a look: https://grail.cs.washington.edu/projects/background-matting/ , which  may be done in mobile phone?

jiangjianping, 2020-05-16:
@allo- ,
Which hardware configuration can keep 20-30 fps? C++ version may perform better than python?

allo-, 2020-05-16:
I am using my python script and with CUDA the net seems to be very fast and with a recent CPU it seems to be quite good as well. I think the bottlenecks in my script are inefficient filters, e.g. opencv blur (and possibly converting between numpy and opencv).
I am still optimizing stuff, though. Using C++ could make things a bit faster, but probably not much as tensorflow uses a C++ backend anyway.

[81 more replies]