webcamoid/webcamoid #251: Audio distorted when making a movie

littlefico, 2020-03-28:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debbuger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates bellow that fit better your issue.

# Reporting a problem?

## Summary

No matter which mic i use no matter which codec i use as output for audio and video, the result is always the same. audio is so distorted, you cannot understand anything

## Current Behavior

When watching videos, made with the program they have distorted audio, not understandable

## Expected Behavior

i should have hear my voice

## Steps to Reproduce

1.) Configure Audio Settings to your mic
2.) Make a Video
3.) Watch it with VLC player

## Suggestions and tips

## Your Environment

* Webcamoid information: 8.7.1 x64
* Operating System information: win10 pro x64, intel hades canyon barebone
* Any other useful information: 2x mics tested (bluetooth headset sennheiser mx 150, and integrated mic of a webcam, dericam) --> i could test both mics positive and working properly

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questinons and other matters?

Go ahead!

littlefico, 2020-03-31:
could you reproduce that?

littlefico, 2020-04-01:
i have an update.
i installed a jack microphone and it works with the same cam, when the audio output source is another microphone than the inbuild one in the webcam.

littlefico, 2020-05-03:
anything new?

hipersayanX, 2020-05-05:
@littlefico Nope, I'll will check this later, but if you have new information or patch to solve this bug, just drop it.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX