webcamoid/webcamoid #253: webcamoid crashes after start virtual camera and start a video [win10 1909]

gittyhub2018, 2020-03-30:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debbuger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates bellow that fit better your issue.

# Reporting a problem?

## Summary

Write here a brief description of the problem.

I start the program (with or without admin privilege make no difference, both is tested and create the same problem), then i activate the virtual camera. After i start a video to stream to the virtual camera, the Program crashes completely. Video size is not important. It crashes with only 11 Mb and 100 Mb file size.

Describe the problem the best as you can, don't omit information.

I will start and stop a video stream to the virtual camera without crashes.

How it should have work?

Nothing.

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Open webcamoid 
2.  Click the settings icon
3.  Activate virtual camera
3.1 if virtual camera doesn't exist create it now (it makes no changes)
4. Go to the source settings
5. Add a mp4 file 
6. Start the video.
7. Stop the video after few seconds.
-> Crash occure
PS: I have start the program under a normal and a admin account, the error occured the same! The mp4 file is in the C:\Users\public . That makes it easier to test it without rights permission problems.

## Suggestions and tips
Error in the RAM writing.
How would you solve the problem?
I don't have solved it.

## Your Environment

* Webcamoid information: (version) (architecture) 8.7.1 win32; I have test the 8.7.1 win64 Version, too.   it creates the same issue.

* Operating System information: Windows 10 Education 1909 Update date 30.03.2020; Lenovo G500; RAM: 6GB; CPU: Intel i3; SSHD;
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)
Win 10 Sys logs
[Win10_error_log.zip](https://github.com/webcamoid/webcamoid/files/4405014/Win10_error_log.zip)
# Want a new feature?
No
Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questinons and other matters?

Go ahead!

gittyhub2018, 2020-03-31:
PS: I have another error log from my desktop pc with follow config:
Win 10 1909 (Update Status: 30.03.2020); CPU: Ryzen 7 2700; Graphic Card: NVIDIA RTX 2070; SSD: M.2 Samsung
[Error_Win10_webcamoid_Desktop.zip](https://github.com/webcamoid/webcamoid/files/4407327/Error_Win10_webcamoid_Desktop.zip)

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX