webcamoid/webcamoid #254: Can't set up webcamoid for use with Skype

jim1001, 2020-04-01:
Thanks for sharing webcamoid and you help.

## Summary

Get infinity tunnel of windows in webcamoid and can't see webcamoid effects in Skype.

## Current Behavior

I tried following step posted in issue #90, i.e. 

> 1. Open Webcamoid 64 bits, go to Outputs and Remove All virtual webcams.
> 2. Uninstall Webcamoid 64 bits and remove all files related to it from the PC. To be sure, uninstall everything related to Webcamoid.
> 3. Install (or download the portable) Webcamoid 32 bits, go to Outputs and add the webcam.
> 4. Close and exit from all target programs (Firefox, Skype, Chrome, Brave, etc.) and reopen it.
> 5. Go to the webcam icon in Webcamoid, select for example Screen 0 and press the play button if not playing already.
> 6. At this point all programs must be able to see the webcam.
> 

Please tell me what I'm doing wrong (due to my lack of understanding) - I see an infinity tunnel of windows in webcamoid as shown in attached screenshots for steps 3 and 5. 

## Expected Behavior

Hoping to see Webcamoid effects in Skype

## Your Environment

* Webcamoid information: v8.7.1 Windows Portable 32 bit
* Operating System information: Microsoft Windows 10 Enterprise
Version 10.0.17763 Build 17763

[image]
[image]

hipersayanX, 2020-04-04:
Depends on which version of Skype, [Skype for desktop](https://www.skype.com/es/get-skype/download-skype-for-desktop/) may work, the one integrated with Windows does not.

>Please tell me what I'm doing wrong (due to my lack of understanding) - I see an infinity tunnel of windows in webcamoid as shown in attached screenshots for steps 3 and 5.

You are doing a desktop capture.

jim1001, 2020-04-06:
Thanks for reply.

>Depends on which version of Skype, Skype for desktop<https://www.skype.com/es/get-skype/download-skype-for-desktop/> may work, the one integrated with Windows does not.

I had read that but assumed (wrongly!) that because I was using a desktop I was bound to have Skype for desktop. Now I’ve read the details and see Skype for desktop is same as Skype Classic but is different to the Skype app.

>You are doing a desktop capture.

Ah, hadn’t realised the Play button was depressed and that was capturing desktop!

Thanks again ☺

hipersayanX, 2020-04-07:
>I had read that but assumed (wrongly!) that because I was using a desktop I was bound to have Skype for desktop. Now I’ve read the details and see Skype for desktop is same as Skype Classic but is different to the Skype app.

The difference is that the desktop version still uses DirectShow (the virtual camera is actually a DirectShow filter), while the app use Media Foundation, and MF can't read DS filters, so the camera can't be detected there, to make the camera detectable in MF it must be a signed Windows driver afaik.

jim1001, 2020-04-15:
I see, thank you for the explanation. It’s nice to know the technical background. All the best to you.

hipersayanX, 2020-04-16:
Also, did not tested, but maybe it could work in Skype Web with Chrome or Firefox.

closed 2021-02-16 by hipersayanX