webcamoid/webcamoid #255: Mirror-inverted stream

Dangerfreak, 2020-04-01:
I'm using Webcamoid v8.7.1 (64 bit, Windows 10) for my YouTube live streams (via virtual webcam), because I wanted to be able to switch between two webcams and to show my desktop during the stream. It all works well, except for two details:
1.) The YouTube stream is mirror-inverted, that's perfect for the webcam that shows me, but not for the desktop view. Is there a way (maybe a plugin/effect?) to flip the stream horizontally?
2.) I have two screens, one is 16:9 resolution, the other one is 4:3, my webcams are 16:9. The stream also is 16:9. If I switch the source to the 16:9 screen, the picture in the stream is perfect concerning the aspect ratio. But if I switch to the 4:3 screen, the aspect ratio gets lost, the picture is stretched horizontally to fit the stream.
Is there a solution for these problems?

hipersayanX, 2020-04-04:
A temporary solution is going to _Effects > Matrix Transform_ and set the following matrix:

```
-1 0 0
 0 1 0
```

Dangerfreak, 2020-04-08:
That's a very good solution - thank you very much! :-)

hipersayanX, 2022-02-15:
Feature added in 9.0.0, closing.

closed 2022-02-15 by hipersayanX