webcamoid/webcamoid #257: App not working properly when building on Windows

demiantres, 2020-04-01:
I am trying to debug the current "master" branch of the webcamoid project on Windows using shadow builds with Qt 5.13 and 5.15. Therefor I followed the instructions from https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Windows. 

But the program starts with a screen like this and nothing works:
[image]

The installer packages provided by your website work fine.
Is there a special setting required to make it work?

demiantres, 2020-04-01:
Qt Creator/qmake creates for each plugin multiple sub-directories of the structure:

> libAvKys\Lib\debug\Qt5.15.0\gpp\i686\bin\<here is the plugin's dll>

Could this be the problem?

hipersayanX, 2020-04-04:
https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Windows#build-from-qt-creator

>8. In **Build & Run > Run > Command line arguments** add:
  
```
-r -p C:\[path]\webcamoid\libAvKys\Plugins --vcam C:\[path]\webcamoid\libAvKys\Plugins\VirtualCamera\src\dshow\VirtualCamera
```

demiantres, 2020-04-04:
I already added the command line arguments (I exactly followed the build instructions).
Interestingly when I compile and run the 8.7.1 source code then the app completely fails to start with these messages:
```
qrc:/Webcamoid/share/qml/main.qml:27:1: module "AkQmlControls" is not installed
qrc:/Webcamoid/share/qml/main.qml:24:1: module "AkQml" is not installed
qrc:/Webcamoid/share/qml/main.qml:27:1: module "AkQmlControls" is not installed
qrc:/Webcamoid/share/qml/main.qml:24:1: module "AkQml" is not installed
```

I am using Qt Creator 4.11.1 on Windows with mingw and Qt 5.15 32 bit (also tested other Qt versions with the same result). I tested it with and without shadow build.

Here is what I have done:

1. Download webcamoid.pro in Qt Creator.
2. Set the FFMPEGINCLUDES
3. Set the command line arguments 
4. Set the run environment path to the ffmpeg\bin
5. Click Run in Qt Creator which start compilation.

hipersayanX, 2020-04-04:
8.7.1 and master are completely different versions, in 8.7.1 you must use the -q option to tell where the AkQml folder is located, master does not have such problem, if you want to test or debug Webcamoid use the master branch.
Also, in Windows is a lot much easier to setup the working environment using [MSYS2](https://www.msys2.org/), uninstall Qt and then install MSYS2 and all [dependencies](https://github.com/webcamoid/webcamoid/blob/master/ports/ci/appveyor/install_deps.sh#L21) from there. FFMPEGINCLUDES and all those flags are not required. I will update the wiki later.

demiantres, 2020-04-04:
I got the master branch working on another machine - thanks. The app crashes during creation of the virtual camera now but I will post another issue for this problem.

demiantres, 2020-04-05:
https://github.com/webcamoid/webcamoid/issues/264

closed 2020-04-05 by demiantres