webcamoid/webcamoid #258: new camera on linux only outputing a black screen

campbebj, 2020-04-01:
# Reporting a problem?

## Summary

Write here a brief description of the problem.
When i create a virtual webcam, and set output to be displayed, i attempted to open it up in a variety of applications (VLC being the one i fell back on as the most likely to work) but all i see is a black screen with no content.

## Current Behavior

Describe the problem the best as you can, don't omit information.
I see a black screen with no content on the virtual webcam i created (created using akvcam)

## Expected Behavior

How it should have work?
I expected i should see output from webcamoid coming through on the video feed.

## Your Environment

* Webcamoid information: 8.7.1 linux 64bit
* Operating System information: ubuntu 18.04 64bit

hipersayanX, 2020-04-04:
Which driver are you using, akvcam or v4l2loopback?

campbebj, 2020-04-04:
I tried both. Akvcam was what I ended up leaving setup.

hipersayanX, 2020-04-04:
So, akvcam works fine and v4l2loopback is the one that have this problem?

campbebj, 2020-04-04:
Sorry to be confusing. Nether work, but akvcam is what's currently turned
on.

campbebj, 2020-04-09:
anything i can provide to assist with debugging?

hipersayanX, 2020-04-12:
the **dmesg** related to akvcam and your _/etc/akvcam/config.ini_.

campbebj, 2020-04-13:
here is my config https://pastebin.com/VVPJaaf2

i couldnt figure out how to get my dmesg for this

im managing the module with DKMS and i dont see the .ko file referenced for loading and unloading the driver to change the loglevel

this is my v4l2-compliance for my "camera" https://pastebin.com/cqS2XgPf

hipersayanX, 2020-04-16:
>i couldnt figure out how to get my dmesg for this

Unload the driver

```sudo modprobe -r akvcam```

Clean dmesg buffer

```sudo dmesg -C```

Listen to dmesg new messages

```dmesg -w```

Open a new terminal and load the driver

```sudo modprobe akvcam loglevel=7```

All related messages starts with  ```[number] akvcam:```.
Also, check if ```ls /etc/akvcam/default_frame.bmp``` is ok.

campbebj, 2020-04-16:
ls /etc/akvcam/default_frame.bmp found the file

modprobe: FATAL: Module akvcam not found.

instead of doing the dkms install i did the sudo make install and i was able to get the modprobe to work.
here is the dkms 
https://pastebin.com/ZAcmP3rs

now that i did the install that way i cant even edit the cameras in webcamoid. so i removed them and cant seem to use webcamoid to add cameras anymore, as webcamoid just freezes when i add a new camera.

im going to try uninstalling everything and installing it again using the sudo make install manner

campbebj, 2020-04-17:
I reinstalled everything from scratch, using the sudo make install method, reinstalled webcamoid.

i was able to get a non blank screen to show up on the webcam, but its not a frozen screen showing my monitor as i was changing the source in webcamoid.
ive attached a screenshot frmo a webcam tester here it is completely static and doesnt change.
[image]
(the numbers in the corner are from the test site, you can ignore those)

I was saving dmesg this entire time, so here it is. but its quite a lot.
[dmesg for akvcam.txt](https://github.com/webcamoid/webcamoid/files/4494213/dmesg.for.akvcam.txt)
there were messages colored red at 1368 related to a segfault, and at  1414, 1623, 1696, 1698 related to uvcvideo.

hipersayanX, 2020-04-17:
The log seems to be right, it's calling  akvcam_ioctl_dqbuf() and akvcam_ioctl_qbuf() as expected, 1368 is probably indicating a segfault in user space (probably Webcamoid), this is probably related to #264.

campbebj, 2020-04-20:
If you think its webcamoid then would this be pertient
VideoCapture: Can't set format: "YUYV"
Segmentation fault
this has started happening when i try and change the source to a gif in webcamoid. (it didnt used to do that back when the akvcam driver wasnt loaded)

hipersayanX, 2020-04-24:
@campbebj 

>is it the length thing?

It's more probably an align problem, use videos and gifs who's width is multiple of 32 when possible. As I said before, it could be related to #264, will check it later.

campbebj, 2020-04-24:
it definitely is the same as 264. im going to close this.

closed 2020-04-24 by campbebj