webcamoid/webcamoid #259: Windows: app does not exit when using URL source

demiantres, 2020-04-02:
#49 Reporting a problem?
Yes.

## Summary

When using an url as the image source then the app hangs on exit with high CPU usage.

## Steps to Reproduce

1.  Add a webcam with the url "udp://@:1234".
2.  Click the start button. The stream will be shown in the webcamoid window.
3.  Exit the app.
4.  The app is still running (as seen in the task manager).
5.  Manually kill the app in the task manager.
6.  Start the webcamoid app again. The url source does not work.
7.  Restart Windows. Now the url source in webcamoid works again.

## Your Environment

* Webcamoid information: 8.7.1 32 bit binary package from website
* Operating System information: Win10 64Bit

hipersayanX, 2020-04-04:
Webcamoid hangs because FFmpeg keep waiting for you to send some data through a UDP port, that's how it woks, not sure if there's a way to set a timeout for it.
If you can't start Webcamoid after setting a UDP source, [reset the settings](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh).

demiantres, 2020-04-04:
A timeout can be set as follows (this could be an option in the user interface). If the user sets the timeout to a value greater than 0 then this code executes:

`VFormatContext *formatCtx = nullptr;
AVDictionary *stream_opts = nullptr;
av_dict_set(&stream_opts, "timeout", "5000000", 0); // in microseconds.
avformat_open_input(&formatCtx , ..., &stream_opts);`

Additionally you can also set an interrupt callback:

`formatCtx->interrupt_callback.opaque = (void*)this;
 formatCtx->interrupt_callback.callback = &FFMPEGDecoder::check_interrupt;`

`FFMPEGDecoder::check_interrupt(void * t)
{
  return static_cast<FFMPEGDecoder*>(t)->m_abort; // m_abort is set to true when the app is exited
}`

demiantres, 2020-04-04:
> 
> 
> Webcamoid hangs because FFmpeg keep waiting for you to send some data through a UDP port, that's how it woks, not sure if there's a way to set a timeout for it.

Webcamoid hangs on exit although new data is sent to the UDP port.

hipersayanX, 2020-04-04:
It may work now, the timeout seems to be in milliseconds so I've set it to 3 sec.

demiantres, 2020-04-04:
Thanks for the quick bug fix. The app closes properly now.

closed 2020-04-04 by demiantres