webcamoid/webcamoid #261: Stream a video from ffmpeg to the virtual device on MacOS

mmuneebs, 2020-04-03:
#  Summary

Is there a way to stream a video from ffmpeg to the virtual device created by webcamoid?

## Current Behavior

I'm trying to modify the main camera feed and stream it to the virtual device created using Webcamoid UI, using [this script](https://github.com/steven-varga/chromakey/blob/master/chromakey) with some modifications to work on MacOS Catalina (64bit only), using AVFoundation instead of v4l2.
## Expected Behavior

When I try to target the virtual device as ffmpeg target, it outputs bitrates as if it is streaming to the sink fine. But I still see a mangled Webcamoid watermark in the UI as well on video calling apps, instead of the original or modified camera feed.

## Steps to Reproduce

Installed webcamoid using `brew cask install webcamoid`.
Note that starting with MacOS Catalina, it only allows 64bit apps.
I see akvcam as the driver. I do not get any video playback options in the UI, not that that's what I want to do.

## Suggestions and tips

Allow ffmpeg to output a video stream to the created virtual device.

## Your Environment

* Webcamoid information: (version 8.7.1)
* Operating System information: (MacOS Catalina 10.15.4 (19E266))
* Catalina only allows 64bit apps

hipersayanX, 2020-04-04:
You must link your program to the [bridge library](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/ipcbridge.h) and talk to the camera through the API, [here](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/virtualcameraelement.cpp) you have an example.
If you are using a language different from C++ you must write a wrapper for it.

>Allow ffmpeg to output a video stream to the created virtual device.

Some FFmpeg developer must do that, not me.

hipersayanX, 2021-02-16:
>>Allow ffmpeg to output a video stream to the created virtual device.

>Some FFmpeg developer must do that, not me.

This is now possible with the [new virtual camera](https://github.com/webcamoid/akvirtualcamera/issues/3) (still in development).

closed 2021-02-16 by hipersayanX